espressif/esp-zigbee-sdk #641: Unable to send attribute reports

marrog86, 2025-05-08:
Hello, 

I try to implement attribute reporting for ESP_ZB_ZCL_ATTR_WINDOW_COVERING_CURRENT_POSITION_TILT_PERCENTAGE_ID
I am configuring attributes like this: 

`
 
static void create_widnow_cluster( esp_zb_cluster_list_t *esp_zb_cluster_list){
    esp_zb_attribute_list_t *attr_list = esp_zb_zcl_attr_list_create(ESP_ZB_ZCL_CLUSTER_ID_WINDOW_COVERING);
    uint8_t lift_pct, dummy_tilt_pct = 0;

    esp_zb_cluster_add_attr(
        attr_list,
        ESP_ZB_ZCL_CLUSTER_ID_WINDOW_COVERING,
        ESP_ZB_ZCL_ATTR_WINDOW_COVERING_CURRENT_POSITION_LIFT_PERCENTAGE_ID,
        ESP_ZB_ZCL_ATTR_TYPE_U8,
        ESP_ZB_ZCL_ATTR_ACCESS_READ_ONLY | ESP_ZB_ZCL_ATTR_ACCESS_REPORTING,
        &lift_pct
    );//esp_zb_window_covering_cluster_add_attr
    esp_zb_cluster_add_attr(
        attr_list,
        ESP_ZB_ZCL_CLUSTER_ID_WINDOW_COVERING,
        ESP_ZB_ZCL_ATTR_WINDOW_COVERING_CURRENT_POSITION_TILT_PERCENTAGE_ID,
        ESP_ZB_ZCL_ATTR_TYPE_U8,
        ESP_ZB_ZCL_ATTR_ACCESS_READ_ONLY | ESP_ZB_ZCL_ATTR_ACCESS_REPORTING,
        &dummy_tilt_pct
    );
    uint16_t openLimit = 100;
    esp_zb_cluster_add_attr(
        attr_list,
        ESP_ZB_ZCL_CLUSTER_ID_WINDOW_COVERING,
        ESP_ZB_ZCL_ATTR_WINDOW_COVERING_INSTALLED_OPEN_LIMIT_LIFT_ID,
        ESP_ZB_ZCL_ATTR_TYPE_U16,
        ESP_ZB_ZCL_ATTR_ACCESS_READ_ONLY,
        &openLimit
    );
    esp_zb_cluster_add_attr(
        attr_list,
        ESP_ZB_ZCL_CLUSTER_ID_WINDOW_COVERING,
        ESP_ZB_ZCL_ATTR_WINDOW_COVERING_INSTALLED_OPEN_LIMIT_TILT_ID,
        ESP_ZB_ZCL_ATTR_TYPE_U16,
        ESP_ZB_ZCL_ATTR_ACCESS_READ_ONLY,
        &openLimit
    );

    uint16_t closedLimit = 0;
    esp_zb_cluster_add_attr(
        attr_list,
        ESP_ZB_ZCL_CLUSTER_ID_WINDOW_COVERING,
        ESP_ZB_ZCL_ATTR_WINDOW_COVERING_INSTALLED_CLOSED_LIMIT_LIFT_ID,
        ESP_ZB_ZCL_ATTR_TYPE_U16,
        ESP_ZB_ZCL_ATTR_ACCESS_READ_ONLY,
        &closedLimit
    );
    esp_zb_cluster_add_attr(
        attr_list,
        ESP_ZB_ZCL_CLUSTER_ID_WINDOW_COVERING,
        ESP_ZB_ZCL_ATTR_WINDOW_COVERING_INSTALLED_CLOSED_LIMIT_TILT_ID,
        ESP_ZB_ZCL_ATTR_TYPE_U16,
        ESP_ZB_ZCL_ATTR_ACCESS_READ_ONLY,
        &closedLimit
    );
    esp_zb_cluster_list_add_window_covering_cluster(esp_zb_cluster_list, attr_list, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE);
} 
`
Setting the attibute from coorinator wokrs. 

I am updating on the end device the attribute like this: 
`
esp_zb_lock_acquire(portMAX_DELAY);
        esp_zb_zcl_status_t state = esp_zb_zcl_set_attribute_val(
            1,
            ESP_ZB_ZCL_CLUSTER_ID_WINDOW_COVERING,
            ESP_ZB_ZCL_CLUSTER_SERVER_ROLE,
            ESP_ZB_ZCL_ATTR_WINDOW_COVERING_CURRENT_POSITION_TILT_PERCENTAGE_ID,
            &tilt_percentage,
            ZB_TRUE);
        esp_zb_lock_release();

        
        esp_zb_zcl_report_attr_cmd_t report_cmd = {
            .zcl_basic_cmd = {
                .dst_addr_u.addr_short = 0x0000, /*!< Single short address or group address */
                //.dst_addr_u.addr_long
                .dst_endpoint = 1,              /*!< Destination endpoint */
                .src_endpoint = 1,              /*!< Source endpoint */
            },
            .address_mode = ESP_ZB_APS_ADDR_MODE_DST_ADDR_ENDP_NOT_PRESENT,//ESP_ZB_APS_ADDR_MODE_16_ENDP_PRESENT,
            .clusterID = ESP_ZB_ZCL_CLUSTER_ID_WINDOW_COVERING,
            .attributeID = ESP_ZB_ZCL_ATTR_WINDOW_COVERING_CURRENT_POSITION_TILT_PERCENTAGE_ID,
            //.cluster_role = ESP_ZB_ZCL_CLUSTER_SERVER_ROLE,
        };
        
        esp_zb_lock_acquire(portMAX_DELAY);
        state = esp_zb_zcl_report_attr_cmd_req(&report_cmd);
        esp_zb_lock_release();
`

and getting a trap: 
`ZB_TRACE_LOG[0]: common/zb_debug.c:64    Assertion failed /builds/thread_zigbee/esp-zboss/components/zboss_stack/zboss/zcl/zcl_general_commands.c:612
Zigbee stack assertion failed zcl/zcl_general_commands.c:612
`
Sorry for bad formatting, I don't know why code formatting does not work for me. 

Can you please give me a hint how to get attribute reporting to work?
Greets.

xieqinan, 2025-05-09:
Why did you comment out `.cluster_role = ESP_ZB_ZCL_CLUSTER_SERVER_ROLE,`? Please enable it and try again.

marrog86, 2025-05-10:
`sp_zb_light.c:90:14: error: 'esp_zb_zcl_report_attr_cmd_t' {aka 'struct esp_zb_zcl_report_attr_cmd_s'} has no member named 'cluster_role'
   90 |             .cluster_role = ESP_ZB_ZCL_CLUSTER_SERVER_ROLE,
      |              ^~~~~~~~~~~~`

I commented out as i thought that this attribute is outdated. 

The header file is located here:
managed_components/espressif__esp-zigbee-lib/include/zcl/esp_zigbee_zcl_command.h 
but does not contain cluster_role. 

Do I misinterpret this?

xieqinan, 2025-05-12:
> Do I misinterpret this?

No, it is my fault. 

Could you please attempt to set the `report_cmd.direction = ESP_ZB_ZCL_CMD_DIRECTION_TO_CLI` and test again?

marrog86, 2025-05-13:
Yes, this change avoids the trap. 
I see also the update as a log entry in zigbee2mqtt, but I did not get the expected value. 

Tilt is sending the first (itinial) value of esp_zb_cluster_add_attr as update. When I change the Value on Zigbee2Mqtt, it is switched back by the device. So I would say, Reporting is done, but sending always the initial value used in esp_zb_cluster_add_attr. 
I expect that update does not work as esp_zb_zcl_get_attribute gives the original value again after updating it.  

Do you have a hint for me why update is not done properly?
Thanks
`
   esp_zb_lock_acquire(portMAX_DELAY);
    esp_zb_zcl_status_t state = esp_zb_zcl_set_attribute_val(
        1,
        ESP_ZB_ZCL_CLUSTER_ID_WINDOW_COVERING,
        ESP_ZB_ZCL_CLUSTER_SERVER_ROLE,
        ESP_ZB_ZCL_ATTR_WINDOW_COVERING_CURRENT_POSITION_TILT_PERCENTAGE_ID,
        &tilt_percentage,
        ZB_TRUE);
    
    esp_zb_zcl_attr_t * AttrVal = esp_zb_zcl_get_attribute(
                1,
                ESP_ZB_ZCL_CLUSTER_ID_WINDOW_COVERING,
                ESP_ZB_ZCL_CLUSTER_SERVER_ROLE,
                ESP_ZB_ZCL_ATTR_WINDOW_COVERING_CURRENT_POSITION_TILT_PERCENTAGE_ID
            );
    esp_zb_lock_release();
    
    uint8_t test_attr = *(uint8_t *)AttrVal->data_p;
    ESP_LOGI(TAG, "value is now  %d,type %d",test_attr, AttrVal->type );

    esp_zb_zcl_report_attr_cmd_t report_cmd = {
        .zcl_basic_cmd = {
            .dst_addr_u.addr_short = 0x0000, /*!< Single short address or group address */
            //.dst_addr_u.addr_long
            .dst_endpoint = 1,              /*!< Destination endpoint */
            .src_endpoint = 1,              /*!< Source endpoint */
        },
        .address_mode = ESP_ZB_APS_ADDR_MODE_DST_ADDR_ENDP_NOT_PRESENT,//ESP_ZB_APS_ADDR_MODE_16_ENDP_PRESENT,
        .clusterID = ESP_ZB_ZCL_CLUSTER_ID_WINDOW_COVERING,
        .attributeID = ESP_ZB_ZCL_ATTR_WINDOW_COVERING_CURRENT_POSITION_TILT_PERCENTAGE_ID,
        .direction = ESP_ZB_ZCL_CMD_DIRECTION_TO_CLI
        //.cluster_role = ESP_ZB_ZCL_CLUSTER_SERVER_ROLE,
    };
    esp_zb_lock_acquire(portMAX_DELAY);
    esp_zb_zcl_status_t state = esp_zb_zcl_report_attr_cmd_req(&report_cmd);
    esp_zb_lock_release();
    if (state != ESP_ZB_ZCL_STATUS_SUCCESS)
    {
        ESP_LOGE(TAG, "Reporting tilt percentage attribute failed!");
    }`

xieqinan, 2025-05-16:
> ESP_LOGI(TAG, "value is now  %d,type %d",test_attr, AttrVal->type );

If the log confirms that the attribute value has been updated, then calling `esp_zb_zcl_report_attr_cmd_req()` should indeed trigger a report with the new value.  

To verify this, please install a sniffer and capture the Zigbee packets during the report.  

This will help confirm whether the report command is sent correctly.

marrog86, 2025-05-16:
perhaps I wrote to much at once. 

In a nutshell: 
update over zigbee works, but value is wrong. 
update on the device with cl_set_attribute_val is not reflected by following zcl_get_attribute function.

marrog86, 2025-05-16:
Hello, 
found my failure: 
&closedLimit was already a pointer and not a value. Passed value to set_attribute is simply wrong. 
Now it works and updates are visible in zigbee2mqtt. 

Thanks for help.

closed 2025-05-16 by marrog86