espressif/esp-zigbee-sdk #642: Minor spelling: ARRAY_LENTH

miroslav-beranic, 2025-05-09:
### Answers checklist.

- [x] I have read the documentation [ESP Zigbee SDK Programming Guide](https://docs.espressif.com/projects/esp-zigbee-sdk) and tried the [debugging tips](https://docs.espressif.com/projects/esp-zigbee-sdk/en/latest/esp32/developing.html#debugging), the issue is not addressed there.
- [x] I have updated ESP Zigbee libs ([esp-zboss-lib](https://components.espressif.com/components/espressif/esp-zboss-lib) and [esp-zigbee-lib](https://components.espressif.com/components/espressif/esp-zigbee-lib)) to the latest version, with corresponding IDF version, and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.3

### esp-zigbee-lib version.

1.0.0

### esp-zboss-lib version.

1.0.0

### Espressif SoC revision.

ESP32-C6

### What is the expected behavior?

Hi! I've notice a small spelling error in ZB HA Thermostat example, file `examples/esp_zigbee_HA_sample/HA_thermostat/main/esp_zb_thermostat.c`.

It has:
```
#define ARRAY_LENTH(arr) (sizeof(arr) / sizeof(arr[0]))
```
I guess `ARRAY_LENTH` should be `ARRAY_LENGTH`.

Branch `main`.

You can fix it or I can create PR.

### What is the actual behavior?

Definition should be renamed from `ARRAY_LENTH` to `ARRAY_LENGTH`. 

### Steps to reproduce.

1. Open file `examples/esp_zigbee_HA_sample/HA_thermostat/main/esp_zb_thermostat.c`, it is on line 26 and use 3 times.

xieqinan, 2025-05-12:
Fixs: https://github.com/espressif/esp-zigbee-sdk/commit/e0361ff7d6e7c43b93d58d88bc6effab646ccafa, please consider closing this issue.

xieqinan, 2025-05-13:
The [PR](https://github.com/espressif/esp-zigbee-sdk/pull/644) has already been merged. Closing this issue—please feel free to reopen it if the problem persists.

closed 2025-05-13 by xieqinan