espressif/esp-zigbee-sdk #645: tuya window sensor binding

Chris-Mark-Wilson, 2025-05-09:
### Question

Hi Ive tried for 2 days to get a tuya window sensor to bind so I can get access to its state but all i ever get is failed with error 132.
I just want to get hold of the state changed event, and I think i am finding out that tuya are a law unto themselves..
Some pointers? anything to help me figure this out would be very gratefully received thankyou

xieqinan, 2025-05-12:
Could you please provide more details and context about the issue you encountered? It would be helpful if you could share the ESP an Tuya log output and the Wireshark packet capture in a `.zip` file to assist with troubleshooting.

Chris-Mark-Wilson, 2025-05-12:
ok will try thankyou xieqinan, Im learning the library and functions from the examples, have gone back to the thermostat example and bound a sonoff trv, which I can now switch on and off. I will use that same method to try and bind the tuya window sensor using the ias cluster or binary cluster and try and see how that works. Could you please repeat instructions how to get wireshark to read specifically the tuya output? I have an ha setup using home assistant also and that may be creating noise. my esp32c6 is basically set up as a standalone thermostat, to control a trv, using input from a temp  sensor, presence sensor  and the window sensor

xieqinan, 2025-05-12:
>  Could you please repeat instructions how to get wireshark to read specifically the tuya output?

Please refer to: https://docs.espressif.com/projects/esp-zigbee-sdk/en/latest/esp32/developing.html#sniffer-and-wireshark.

xieqinan, 2025-05-12:
> I think i am finding out that tuya are a law unto themselves..

By the way, that’s true. If a Tuya product is declared to support other ecosystems, there should be documentation available to guide you through the setup process. You can follow that to integrate the Tuya device with other ecosystems. If you encounter any non-standard behavior from the Tuya that violates the ZCL specification, it’s best to contact Tuya’s technical support for a quicker resolution.

Chris-Mark-Wilson, 2025-05-12:
Hi, thanks for all your help, appreciated,
i really cant get any output from wireshark using pyspinel with an esp32c6 running ot_rcp. this is the command im using python3 sniffer.py -u /dev/ttyACM1 -c 13 --crc --rssi  | wireshark -k -i -
wireshark opens but nothing, ot_rcp is on ACM1, the app config is 

```/* Zigbee configuration */
#define MAX_CHILDREN                    10          /* the max amount of connected devices */
#define INSTALLCODE_POLICY_ENABLE       false       /* enable the install code policy for security */
#define HA_THERMOSTAT_ENDPOINT          10          /* esp thermostat device endpoint */
#define ESP_ZB_PRIMARY_CHANNEL_MASK     (1l << 13)  /* Zigbee primary channel mask use in the example */

/* Attribute values in ZCL string format
 * The string should be started with the length of its own.
 */
#define MANUFACTURER_NAME               "\x09""ESPRESSIF"
#define MODEL_IDENTIFIER                "\x07"CONFIG_IDF_TARGET

#define ESP_ZB_ZC_CONFIG()                                                              \
    {                                                                                   \
        .esp_zb_role = ESP_ZB_DEVICE_TYPE_COORDINATOR,                                  \
        .install_code_policy = INSTALLCODE_POLICY_ENABLE,                               \
        .nwk_cfg.zczr_cfg = {                                                           \
            .max_children = MAX_CHILDREN,                                               \
        },                                                                              \
    }

#define ESP_ZB_DEFAULT_RADIO_CONFIG()                           \
    {                                                           \
        .radio_mode = ZB_RADIO_MODE_NATIVE,                     \
    }

#define ESP_ZB_DEFAULT_HOST_CONFIG()                            \
    {                                                           \
        .host_connection_mode = ZB_HOST_CONNECTION_MODE_NONE,   \
    }
```

these are the logs when i bind the window sensor but i get no response from state changes, the device flashes  but i get nothing in the action handler 

```I (23) boot: ESP-IDF v5.5-dev-3372-g38628f98b9-dirty 2nd stage bootloader
I (23) boot: compile time May 12 2025 10:50:06
I (24) boot: chip revision: v0.1
I (25) boot: efuse block revision: v0.3
I (29) boot.esp32c6: SPI Speed      : 80MHz
I (32) boot.esp32c6: SPI Mode       : DIO
I (36) boot.esp32c6: SPI Flash Size : 4MB
I (40) boot: Enabling RNG early entropy source...
I (44) boot: Partition Table:
I (47) boot: ## Label            Usage          Type ST Offset   Length
I (53) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (60) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (66) boot:  2 factory          factory app      00 00 00010000 00200000
I (73) boot:  3 zb_storage       Unknown data     01 81 00210000 00004000
I (79) boot:  4 zb_fct           Unknown data     01 81 00214000 00000400
I (86) boot: End of partition table
I (89) esp_image: segment 0: paddr=00010020 vaddr=420b0020 size=54dfch (347644) map
I (162) esp_image: segment 1: paddr=00064e24 vaddr=40800000 size=0b1f4h ( 45556) load
I (172) esp_image: segment 2: paddr=00070020 vaddr=42000020 size=ac118h (704792) map
I (305) esp_image: segment 3: paddr=0011c140 vaddr=4080b1f4 size=08254h ( 33364) load
I (313) esp_image: segment 4: paddr=0012439c vaddr=40813450 size=02478h (  9336) load
I (316) esp_image: segment 5: paddr=0012681c vaddr=50000000 size=0001ch (    28) load
I (321) boot: Loaded app from partition at offset 0x10000
I (322) boot: Disabling RNG early entropy source...
I (339) cpu_start: Unicore app
I (347) cpu_start: Pro cpu start user code
I (347) cpu_start: cpu freq: 160000000 Hz
I (348) app_init: Application information:
I (348) app_init: Project name:     Zigbee-Smart-Stat
I (352) app_init: App version:      d796c4e-dirty
I (357) app_init: Compile time:     May 12 2025 10:49:52
I (362) app_init: ELF file SHA256:  721353440...
[... 59 lines cut ...]
I (17354) ZIGBEE: Binding Window Sensor to coordinator
I (17364) ZIGBEE: Sending bind request - addr: 0xd82c, ep: 1, cluster: 0x0500
I (17364) ZIGBEE: Device saved successfully: WINDOW_1
I (17374) ZIGBEE: Window sensor saved as WINDOW_1
I (17874) ZIGBEE: Window sensor binding callback triggered with status: 0x0
I (17874) ZIGBEE: Bind request details - dst: 0xd82c, src_ep: 1, dst_ep: 10
I (17874) ZIGBEE: Window sensor bound successfully
I (17884) ZIGBEE: Coordinator address: 0x0000
I (17884) ZIGBEE: Configuring IAS Zone reporting for window sensor
W (17894) ZIGBEE: Config report command returned 0 TSN - this may be normal
I (18344) ZB_ACTION_HANDLER: Action handler called with callback ID: 0x1002
I (18344) ZB_ACTION_HANDLER: Report config response received
I (18344) ZB_ACTION_HANDLER: Report configuration accepted by sensor
I (18834) ZIGBEE: Device WINDOW_1 authorized
I (19394) ZIGBEE: Permit join status changed
I (21014) DHT_SENSOR: DHT Readings - Temperature: 24.2°C, Humidity: 56.1%
I (31084) DHT_SENSOR: DHT Readings - Temperature: 24.2°C, Humidity: 56.1%
```
[repo here](https://github.com/Chris-Mark-Wilson/Zigbee-Smart-Stat/tree/dev)

[zigbee binding code and main handlers](https://github.com/Chris-Mark-Wilson/Zigbee-Smart-Stat/blob/dev/main/Zigbee/zigbee.c)

[main.c with initialisation and action/attribute handlers](https://github.com/Chris-Mark-Wilson/Zigbee-Smart-Stat/blob/dev/main/main.c)

not sure if im doing something wrong with pyspinel and the ot_rtp? that seems to start up ok?

Chris-Mark-Wilson, 2025-05-12:
ps these are the logs i gety from an ewelink window sensor...
```
I (27194) ZIGBEE: Device associated with network
I (27194) ZIGBEE: Device update notification received
I (27944) ZIGBEE: New device joining network (short: 0x673e)
I (27944) ZIGBEE: Trying to identify window sensor at addr 0x673e
I (27944) ZIGBEE: Using profile 0x0104, searching for window sensor clusters
I (27944) ZIGBEE: Window sensor match descriptor request sent successfully
I (28524) ZIGBEE: Window sensor callback triggered:
I (28524) ZIGBEE:   Status: 0x0
I (28524) ZIGBEE:   Address: 0x673e
I (28524) ZIGBEE:   Endpoint: 1
I (28534) ZIGBEE: Device identified as Window Sensor
I (28534) ZIGBEE: Found Window Sensor at address 0x673e, endpoint 1
I (28544) ZIGBEE: Binding Window Sensor to coordinator
I (28544) ZIGBEE: Sending bind request - addr: 0x673e, ep: 1, cluster: 0x0500
I (28564) ZIGBEE: Device saved successfully: WINDOW_1
I (28564) ZIGBEE: Window sensor saved as WINDOW_1
I (29774) ZIGBEE: Window sensor binding callback triggered with status: 0x0
I (29774) ZIGBEE: Bind request details - dst: 0x673e, src_ep: 1, dst_ep: 10
I (29774) ZIGBEE: Window sensor bound successfully
I (29784) ZIGBEE: Coordinator address: 0x0000
I (29784) ZIGBEE: Configuring IAS Zone reporting for window sensor
W (29794) ZIGBEE: Config report command returned 0 TSN - this may be normal
I (30314) ZB_ACTION_HANDLER: Action handler called with callback ID: 0x1002
I (30314) ZB_ACTION_HANDLER: Report config response received
I (30314) ZB_ACTION_HANDLER: Report configuration accepted by sensor
I (31084) DHT_SENSOR: DHT Readings - Temperature: 24.2°C, Humidity: 56.1%
I (41154) DHT_SENSOR: DHT Readings - Temperature: 24.2°C, Humidity: 56.1%
I (42204) ZIGBEE: Device WINDOW_1 authorized
I (42994) ZIGBEE: Device 0x673e left network
I (43044) ZIGBEE: Device 0x673e left network
I (51254) DHT_SENSOR: DHT Readings - Temperature: 24.2°C, Humidity: 56.1%
```

Chris-Mark-Wilson, 2025-05-14:
Hi sorry but I cannot get the sniffer to work. I am flashing [ot_rcp](https://github.com/espressif/esp-idf/tree/master/examples/openthread/ot_rcp) to an esp32c6 (ive tried with two different ones) and it just hangs at this point 

````I (23) boot: ESP-IDF v5.5-dev-3372-g38628f98b9 2nd stage bootloader
I (23) boot: compile time May 14 2025 09:56:27
I (24) boot: chip revision: v0.2
I (24) boot: efuse block revision: v0.3
I (28) boot.esp32c6: SPI Speed      : 80MHz
I (32) boot.esp32c6: SPI Mode       : DIO
I (36) boot.esp32c6: SPI Flash Size : 2MB
I (39) boot: Enabling RNG early entropy source...
I (44) boot: Partition Table:
I (47) boot: ## Label            Usage          Type ST Offset   Length
I (53) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (59) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (66) boot:  2 factory          factory app      00 00 00010000 00100000
I (73) boot: End of partition table
I (76) esp_image: segment 0: paddr=00010020 vaddr=42028020 size=022bch (  8892) map
I (85) esp_image: segment 1: paddr=000122e4 vaddr=40800000 size=05d34h ( 23860) load
I (96) esp_image: segment 2: paddr=00018020 vaddr=42000020 size=27590h (161168) map
I (129) esp_image: segment 3: paddr=0003f5b8 vaddr=40805d34 size=073f0h ( 29680) load
I (136) esp_image: segment 4: paddr=000469b0 vaddr=4080d130 size=013a8h (  5032) load
I (141) boot: Loaded app from partition at offset 0x10000
I (141) boot: Disabling RNG early entropy source...

````
I've no idea what I'm doing wrong or not doing as the case may be?

```` ./pyspinel/sniffer.py -c 11 -n 1 --rssi -u /dev/ttyACM1 | wireshark -k -i -
Initializing sniffer...
 ** (wireshark:2377282) 10:30:43.156069 [Capture MESSAGE] -- Capture Start ...

````
So i guess i must have missed a step configuring the sniffer? the way I understood it flashing the ot_rcp firmware and configuring wireshark was the way, the instructions tell to use a different firmware if using the nordic board but im on another esp32c6?

xieqinan, 2025-05-15:
> ** (wireshark:2377282) 10:30:43.156069 [Capture MESSAGE] -- Capture Start ...

The sniffer appears to have started, but please make sure it is operating on the same channel as the device to capture the relevant traffic correctly.

Chris-Mark-Wilson, 2025-05-15:
Sorry I still get nothing on the sniffer, I put extra debug logs in the output
```I (1020) main_task: Returned from app_main()
I (1170) ZIGBEE: Permit join status changed
I (1170) ZIGBEE: Network formation successful
I (1170) debug sniffer: Zigbee operating on channel: 13
I (1790) ZIGBEE: Permit join status changed
I (1790) ZIGBEE: Network steering completed
```

but I do have a development in that if I pair up the device, store it to nvs, all is happy but i dont register any state changes.. Then I reboot the device, clearing all nvs in the process (long button press), so now the coordinator has forgotten about the window sensor, this time when I operate the sensor i get this... 
```I (17250) ZIGBEE: Device Status Indication received from: 0xafe3
I (17250) ZIGBEE: Status code: 0x12
I (17250) ZIGBEE: Unknown command ID: 0x00
W (17250) ZIGBEE: Bad key sequence number 0x12
I (39260) ZIGBEE: Device Status Indication received from: 0xafe3
I (39260) ZIGBEE: Status code: 0x12
I (39260) ZIGBEE: Unknown command ID: 0x00
W (39260) ZIGBEE: Bad key sequence number 0x12
```
from
```      case ESP_ZB_NLME_STATUS_INDICATION:  // 0x32
{
    esp_zb_zdo_signal_nwk_status_indication_params_t *status_params = 
        (esp_zb_zdo_signal_nwk_status_indication_params_t*)esp_zb_app_signal_get_params(p_sg_p);
    
    ESP_LOGI(TAG, "Device Status Indication received from: 0x%04x", status_params->network_addr);
    ESP_LOGI(TAG, "Status code: 0x%02x", status_params->status);
    ESP_LOGI(TAG, "Unknown command ID: 0x%02x", status_params->unknown_command_id);

    if (status_params->status ==  ESP_ZB_NWK_COMMAND_STATUS_BAD_KEY_SEQUENCE_NUMBER ) {
        ESP_LOGW(TAG, "Bad key sequence number 0x%02x", status_params->status); 
     
    }
}
break;
```
which is the address of the sensor, trying to get back in, 

so it kind of seems like the sensor, as long as we know who it is, is happy to just leave the packet on the doorstep and doesnt bother ringing the bell? or I'm not listening on the right doorbell frequency? but if we forget who it is, its clearly still bound to us, or thinks it is and is trying to get in with his old key before we changed the locks? when before, when he had the right key, he just left it somewhere and didnt tell us where and we never found it.... it's a annoying that I cannot seem to get any output from the sniffer...

xieqinan, 2025-05-15:
>  it's a annoying that I cannot seem to get any output from the sniffer...

Please note that before running the command  
`python3 sniffer.py -u /dev/ttyACM1 -c 13 --crc --rssi | wireshark -k -i -`,  
you must ensure that the port connected to the `ot_rcp` is not already opened or being monitored by another process or thread.

> but I do have a development in that if I pair up the device, store it to nvs, all is happy but i dont register any state changes.. Then I reboot the device, clearing all nvs in the process (long [...]

This issue was introduced by your operation and is not related to the original problem—we should stay focused on resolving the current issue. But the behavior you're seeing is expected: if the sensor has been factory reset, its network information is erased, so it may fail to rejoin the previous network due to mismatched settings or other reasons. It's recommended to first understand the basic behavior of the sensor. A clear understanding can help avoid unexpected or confusing problems later on.

Chris-Mark-Wilson, 2025-05-15:
I can feel the problem lies with configuring the device to report on state change as i can get a correct status by 
```c
void read_window_sensor_status(uint16_t addr, uint8_t endpoint) {
    // Define the attribute ID we want to read
    uint16_t attr_id = ESP_ZB_ZCL_ATTR_IAS_ZONE_ZONESTATUS_ID;
    
    esp_zb_zcl_read_attr_cmd_t read_req = {
        .zcl_basic_cmd = {
            .dst_addr_u.addr_short = addr,
            .dst_endpoint = endpoint,
            .src_endpoint = HA_THERMOSTAT_ENDPOINT,
        },
        .address_mode = ESP_ZB_APS_ADDR_MODE_16_ENDP_PRESENT,
        .clusterID = ESP_ZB_ZCL_CLUSTER_ID_IAS_ZONE,
        .attr_number = 1,
        .attr_field = &attr_id,
    };
```

which gives me 
```py
58156) ZB_ACTION_HANDLER: Action handler called with callback ID: 0x1000, message: 0x4082ce58
I (158156) ZB_ACTION_HANDLER: Read attribute response received
I (158156) ZB_ACTION_HANDLER: IAS Zone Status: 0x0000
I (158166) ZB_ACTION_HANDLER: Window is CLOSED
I (158166) ZB_ACTION_HANDLER: Additional status - Alarm2: 0, Tamper: 0, Battery Low: 0
```
so I have that rigged up to a button press. The thing is, I press the button 
```py I (868446) BUTTON: Button pressed
I (868646) BUTTON: Short press detected (200 ms)
I (868646) ZIGBEE: Reading window sensor status
I (868646) ZIGBEE: Read attribute command returned TSN: 10
```
then if I leave it alone i get 
```py I (868446) BUTTON: Button pressed
I (868646) BUTTON: Short press detected (200 ms)
I (868646) ZIGBEE: Reading window sensor status
I (868646) ZIGBEE: Read attribute command returned TSN: 10
I (876336) ZIGBEE: Device Status Indication received from: 0x3048
I (876336) ZIGBEE: Status code: 0x06
I (876336) ZIGBEE: Unknown command ID: 0x00
I (886346) ZIGBEE: Device Status Indication received from: 0x3048
I (886346) ZIGBEE: Status code: 0x06
I (886346) ZIGBEE: Unknown command ID: 0x00
I (896356) ZIGBEE: Device Status Indication received from: 0x3048
I (896356) ZIGBEE: Status code: 0x06
I (896356) ZIGBEE: Unknown command ID: 0x00
I (906366) ZIGBEE: Device Status Indication received from: 0x3048
I (906366) ZIGBEE: Status code: 0x06
I (906366) ZIGBEE: Unknown command ID: 0x00
W (908686) ZIGBEE: Device unavailable: Short address: 0x3048
W (908686) ZIGBEE: Lost connection to device: WINDOW_1
```
however, if i press the button 
```py 
I (989446) BUTTON: Button pressed
I (989626) BUTTON: Short press detected (182 ms)
I (989626) ZIGBEE: Reading window sensor status
I (989626) ZIGBEE: Read attribute command returned TSN: 13
```
and THEN open or close the device, I immediately get 
```py 
I (989446) BUTTON: Button pressed
I (989626) BUTTON: Short press detected (182 ms)
I (989626) ZIGBEE: Reading window sensor status
I (989626) ZIGBEE: Read attribute command returned TSN: 13
I (993666) ZB_ACTION_HANDLER: Action handler called with callback ID: 0x1000, message: 0x4082ce58
I (993666) ZB_ACTION_HANDLER: Read attribute response received
I (993666) ZB_ACTION_HANDLER: IAS Zone Status: 0x0001
I (993676) ZB_ACTION_HANDLER: Window is OPEN
I (993676) ZB_ACTION_HANDLER: Additional status - Alarm2: 0, Tamper: 0, Battery Low: 0
```
so can get the status, sort of, if i immediately move the sensor, but i cant tell it to send the status.. Im using this 
```c
esp_zb_zcl_config_report_record_t report_record = {
    .attributeID = ESP_ZB_ZCL_ATTR_IAS_ZONE_ZONESTATUS_ID,
    .attrType = ESP_ZB_ZCL_ATTR_TYPE_16BITMAP,
    .min_interval = 1,    // Report at least every second if changed
    .max_interval = 10,  // Report at least every 10seconds even if not changed CHANGE BACK WHEN DEBUGGED
    .reportable_change = 0  // Any change should trigger a report for binary sensors
    
};

esp_zb_zcl_config_report_cmd_t report_cmd = {
    .zcl_basic_cmd = {
        .dst_addr_u.addr_short = bind_req->req_dst_addr,
        .dst_endpoint = bind_req->src_endp,
        .src_endpoint = bind_req->dst_endp,
    },
    .address_mode = ESP_ZB_APS_ADDR_MODE_16_ENDP_PRESENT,
    .clusterID = ESP_ZB_ZCL_CLUSTER_ID_IAS_ZONE,
    .direction = ESP_ZB_ZCL_CMD_DIRECTION_TO_SRV,
    .dis_defalut_resp = 0,
    .record_number = 1,
    .record_field = &report_record
};

ESP_LOGI(TAG, "Configuring IAS Zone reporting for window sensor");
uint8_t tsn = esp_zb_zcl_config_report_cmd_req(&report_cmd);
```
but this must be wrong...?

Chris-Mark-Wilson, 2025-05-17:
I Cannot get this sniffer to work at all, I have installed pyspinel, wireshark, i have flashed the esp32c6 with the ot_rcp firmware as is, no modification besides ipd.py set-target esp32c6 then build and flash. nothing else is montoring that port (ttyACM1).
I try running sniffer.py -u /dev/ttyACM1 -b 460800 -c 13 | wireshark -k -i -
i get nothing at all, if i run minicom -D /dev/ttyACM1 -b 460800 i get nothing
I just need to find out what my sensor is sending and cannot get a sniffer to work at all so Im clearly missing something important

Chris-Mark-Wilson, 2025-05-18:
Hi @xieqinan 
ok i have managed to get the sniffer to work using a different esp32c6, configured to use pins rx=17 tx =18, using extcap.

i have saved the packets from startup. clear nvs - reboot, pair tuya window sensor (which will have been paired before it will need to obtain a new address), then operating the sensor. The packets show a request for data from the tuya sensor but no actuial response from the coordinator
[link to packets.zip](https://github.com/Chris-Mark-Wilson/Zigbee-Smart-Stat/blob/dev/packets.zip)

the device will pair, it will send a status update when asked, but i cannot get it to send reports, although it is sending something when the state changes (see logs), it looks like a data request, im guessing its something like 'where do i send my status?' but that is just a guess?

does this throw any light on it please?

Chris-Mark-Wilson, 2025-05-18:
I also managed to get the network key from the coordinator, added thatr to ieee8514 preconfigured keys and got this output 
[decoded-packets](https://github.com/Chris-Mark-Wilson/Zigbee-Smart-Stat/blob/dev/packets-decoded.zip)

Chris-Mark-Wilson, 2025-05-18:
i am definately getting reports being sent from the sensor, i get 
```
431	840.370801	0xde76	0x0000	ZigBee HA	90	ZCL IAS Zone: Zone Status Change Notification, Seq: 167
```
in wireshark everytime it changes status, but nothing in any of my handlers, the signal handler, action handler, maybe i need a new handler?

xieqinan, 2025-05-19:
> i am definately getting reports being sent from the sensor, i get

It is an important indication. Please share your handler. A proper handler might look like the following:

* Register `esp_zb_core_action_handler_register()`
* When the frame is received, the `ESP_ZB_CORE_CMD_IAS_ZONE_ZONE_STATUS_CHANGE_NOT_ID` event will be triggered in the `zb_core_action_handler()` with the type `esp_zb_zcl_ias_zone_status_change_notification_message_t`

By the way, please provide the Wireshark packet in `.pcap` format, which is more intuitive.

Chris-Mark-Wilson, 2025-05-19:
Hi, the signal handler is [here](https://github.com/Chris-Mark-Wilson/Zigbee-Smart-Stat/blob/dev/main/Zigbee/zigbee.c#L548)
the action handler is [here](https://github.com/Chris-Mark-Wilson/Zigbee-Smart-Stat/blob/dev/main/main.c#L99)
i have a further development..
late last night with blearly eyes having gonw round in circles for the the 15th time (ignore enrolling the device for ias, ignore configuring reporting as ias should do automatically, make sure endpoint is configured as client to recieve ias) 

coordinator is on endpoint 10, as is in HA_THERMOSTAT_ENDPOINT 
but the window sensor is sending reports to endpoint 1, from its own endpoint 1... according to wireshark... so I think that may be the issue..
however, the [binding request](https://github.com/Chris-Mark-Wilson/Zigbee-Smart-Stat/blob/dev/main/Zigbee/zigbee.c#L482) should be teeling it the dst endpoint is 10.. so I dont know whats going on here but I think the problem is here somewhere...
full code available from the links, its abit messy, most zigbee stuff is in Zigbee.zigbee.c, the button handler is in Buttons/button.c and main.c contains setup, most tasks (zigbee init etc) plus action handlers etc

Chris-Mark-Wilson, 2025-05-19:
```
403	625.656566	0xde76	0x0000	ZigBee HA	90	ZCL IAS Zone: Zone Status Change Notification, Seq: 165
```

```
Frame 403: 90 bytes on wire (720 bits), 90 bytes captured (720 bits) on interface /dev/ttyACM3:460800, id 0
IEEE 802.15.4 TAP
    Header
    Channel assignment: Page: Default (0), Number: 13
    RSS: -59.00 dBm
    Link Quality Indicator: 10
    FCS type: ITU-T CRC16 (1)
    [Data Length: 54]
IEEE 802.15.4 Data, Src: 0xde76, Dst: 0x0000
    Frame Control Field: 0x8861, Frame Type: Data, Acknowledge Request, PAN ID Compression, Destination Addressing Mode: Short/16-bit, Frame Version: IEEE Std 802.15.4-2003, Source Addressing Mode: Short/16-bit
    Sequence Number: 82
    Destination PAN: 0x9bed
    Destination: 0x0000
    Source: 0xde76
    [Extended Source: TelinkSemico_9c:25:19:97:b3 (a4:c1:38:9c:25:19:97:b3)]
    [Origin: 113]
    FCS: 0x1105 (Correct)
ZigBee Network Layer Data, Dst: 0x0000, Src: 0xde76
    Frame Control Field: 0x2248, Frame Type: Data, Discover Route: Enable, Security, End Device Initiator Data
    Destination: 0x0000
    Source: 0xde76
    Radius: 30
    Sequence Number: 84
    [Extended Source: TelinkSemico_9c:25:19:97:b3 (a4:c1:38:9c:25:19:97:b3)]
    [Origin: 113]
    ZigBee Security Header
ZigBee Application Support Layer Data, Dst Endpt: 1, Src Endpt: 1
    Frame Control Field: Data (0x00)
    Destination Endpoint: 1
    Cluster: Intruder Alarm System Zone (0x0500)
    Profile: Home Automation (0x0104)
    Source Endpoint: 1
    Counter: 129
ZigBee Cluster Library Frame
    Frame Control Field: Cluster-specific (0x09)
        .... ..01 = Frame Type: Cluster-specific (0x1)
        .... .0.. = Manufacturer Specific: False
        .... 1... = Direction: Server to Client
        ...0 .... = Disable Default Response: False
    Sequence Number: 165
    Command: Zone Status Change Notification (0x00)
    ZoneStatus: 0x0001, Alarm 1
    Extended Status: 0x00
    Zone ID: 0x00
    Delay (in quarterseconds): 0

```
this is when the window sensor sends a report, or at least when the state chnages this is what i get in wireshark, but do not recieve any signal in any handlers, looking at Zigbee Application Layer Supoort Layer Data - dst Endpt:1 - shoouldnt this be endpoint 10?