espressif/esp-zigbee-sdk #646: Delaying the joining process and  send the Network Key to a Device by its IEEE address

Dev-Hussam, 2025-05-10:
### Question

Hi, I'm looking for an API's that i can use to delay sending the network key to a joining device, and send the network key to a device using it IEEE address?.

Does the SDK support these features?

xieqinan, 2025-05-12:
> Does the SDK support these features?

Yes, the [esp_zb_apsme_transport_key_request](https://github.com/espressif/esp-zigbee-sdk/blob/main/components/esp-zigbee-lib/include/aps/esp_zigbee_aps.h#L221) and [esp_zb_apsme_switch_key_request()](https://github.com/espressif/esp-zigbee-sdk/blob/main/components/esp-zigbee-lib/include/aps/esp_zigbee_aps.h#L233C11-L233C42) may be helpful.

Dev-Hussam, 2025-05-13:
Thanks for Response,
But what about delaying the Joining process when the coordinator receive Device Association, or Denay the joining before the coordinator send the network key? is this possible?

- I'm using the install code for joining but i need to control the joining process.

xieqinan, 2025-05-13:
> But what about delaying the Joining process when the coordinator receive Device Association, or Denay the joining before the coordinator send the network key? is this possible?

Once `esp_zb_bdb_start_top_level_commissioning(ESP_ZB_BDB_MODE_NETWORK_STEERING);` is called, the device will automatically attempt to find and join a network.  

Is the delay in your project occurring specifically after the **Device Association** step? If not, you can control the join process via enable the `ESP_ZB_BDB_MODE_NETWORK_STEERING` mode. It would be helpful to provide more details about your application so we can better understand the context and assist you more effectively.

Dev-Hussam, 2025-05-13:
i don't have problem in the existing join process,
what i means is when the coordinator open the network and the Device Association occurs on this function (**esp_zb_app_signal_handler**), in this step can i control sending the network key(by delay sending it for few second or deny the joining if possible), so the device not join the network directly?
- in other vendors SDK i read that there is an API i can use to delay the joining process.
Thanks

xieqinan, 2025-05-14:
>  in this step can i control sending the network key(by delay sending it for few second or deny the joining if possible), so the device not join the network directly?

No, the esp-zigbee-sdk(v1.6.4) has not provide the API to implement the above feature.

closed 2025-05-14 by Dev-Hussam