espressif/esp-zigbee-sdk #650: About Zigbee 2.4Ghz signal modulation

Jayliu0712, 2025-05-14:
I used the steps in the following link to continuously send Zigbee 2.4GHz signals in order to optimize the hardware RF circuit through modulation. However, it seems that I need to keep the serial port connected to the host computer all the time. Once the serial port is disconnected, the signal on the spectrum analyzer disappears. Is there a way that after burning the test program, I can disconnect the serial port from the host computer and continuously send the 2.4GHz signal offline?

`https://docs.espressif.com/projects/esp-test-tools/en/latest/esp32h2/development_stage/rf_test_items/zigbee_non_signaling_test.html#non-signaling-test`

Jayliu0712, 2025-05-15:
> `https://docs.espressif.com/projects/esp-test-tools/en/latest/esp32h2/development_stage/rf_test_items/zigbee_non_signaling_test.html#non-signaling-test`

There's another question I'd like to ask. In the link mentioned above, after I burned the bin file and started the transmission test, when the RF circuit was close to the spectrum analyzer, no 2.4GHz signal could be detected. What could be the reasons for this problem?

EspressifTom, 2025-05-15:
Hi Jay,

Disconnect the serial port, but keep the power supply as image, You could get the continuesly signal. 

[image]

Jayliu0712, 2025-05-15:
> Hi Jay,
> 
> Disconnect the serial port, but keep the power supply as image, You could get the continuesly signal.
> 
> [image]

@EspressifTom I've successfully solved this problem.Thanks for your help！

xieqinan, 2025-05-16:
@Jayliu0712 ,

If this problem has been resolved, please consider closing this issue.

closed 2025-05-16 by Jayliu0712