espressif/esp-zigbee-sdk #653: setting up pyspinel and wireshark to sniff zigbee packets

Chris-Mark-Wilson, 2025-05-17:
### Question

Hi I have a problem setting up and getting any response from the pyspinel sniffer with wireshark
I have..
flash the ot_rcp example, as is with no configuration to an esp32c6
i have installed pyspinel and run the install.sh command
I have installed wireshark and configured the ieee 81254 settings as instructed and also the protocol-zigbee settings to the decryption key
i run <path>/sniffer.py -u /ttyACM1 -c 15 -b 460800 | wireshark -k -i -
I get nothing.
If i minicom -D /dev/ttyACM1 -b 460800 i get nothing
I just dont know what im doing wrong?

xieqinan, 2025-05-19:
Based on https://github.com/espressif/esp-zigbee-sdk/issues/645, this issue has been fixed by yourself. Right?

Chris-Mark-Wilson, 2025-05-19:
ah yes sorry, fixed. For further info, esp32c6. In menuconfig set uart pins to manual 17 and 18, install pyspinel with extcap, run wireshark, set properties as instructions but critically **get the network key from the app** which goes in preferences->protocols->zigbee->preconfigured keys, and it changes everytime the device boots so If anyone knows a way to lock it for testing? 
But yes fixed and now working thankyou... lovely when you can see whats actually happening without relying on logging         something that isnt there

xieqinan, 2025-06-13:
Hi @Chris-Mark-Wilson ,

Could you please follow the commands below? They work on my side:

```
cd ~/esp/esp-idf/examples/openthread/ot_rcp
idf.py set-target esp32c6
idf.py build
idf.py -p [PORT] erase-flash flash
sniffer.py -u [PORT] -c 15 -b 460800 | wireshark -k -i -
```

By default, if you start the sniffer before the device joins the network, Wireshark will use the **Global default link key (`Alliance09`)** to decrypt the network key automatically—no manual configuration needed.