espressif/esp-zigbee-sdk #654: Build error in esp_zigbee_all_device_types_app example

wiktorpyk, 2025-05-17:
### Question

I'm getting the following error when building the example

```
/home/wiktorpyk/repos/esp-zigbee-sdk/examples/esp_zigbee_all_device_types_app/main/esp_zigbee_all_device_types_app.h:36:2: error: #error "CLI example for RCP is not ready!"
   36 | #error "CLI example for RCP is not ready!"
      |  ^~~~~
/home/wiktorpyk/repos/esp-zigbee-sdk/examples/esp_zigbee_all_device_types_app/main/esp_zigbee_all_device_types_app.c: In function 'app_main':
/home/wiktorpyk/repos/esp-zigbee-sdk/examples/esp_zigbee_all_device_types_app/main/esp_zigbee_all_device_types_app.c:157:25: error: implicit declaration of function 'ESP_ZB_DEFAULT_RADIO_CONFIG'; did you mean 'ESP_ZB_DEFAULT_HOST_CONFIG'? [-Wimplicit-function-declaration]
  157 |         .radio_config = ESP_ZB_DEFAULT_RADIO_CONFIG(),
      |                         ^~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                         ESP_ZB_DEFAULT_HOST_CONFIG
/home/wiktorpyk/repos/esp-zigbee-sdk/examples/esp_zigbee_all_device_types_app/main/esp_zigbee_all_device_types_app.c:156:39: error: missing braces around initializer [-Werror=missing-braces]
  156 |     esp_zb_platform_config_t config = {
      |                                       ^
  157 |         .radio_config = ESP_ZB_DEFAULT_RADIO_CONFIG(),
      |                         {                            }
cc1: some warnings being treated as errors
ninja: build stopped: subcommand failed.
```

Could someone help me?

lpy4105, 2025-05-19:
Hi @wiktorpyk, 

What target are you trying this example on? I believe you are using the Gateway board.

Currently, only targets with native IEEE 802.15.4 radio (e.g. esp32h2, esp32c6) are supported and tested. For some "gateway" targets (e.g. esp32s3), the example is not supported nor tested.

wiktorpyk, 2025-05-21:
> What target are you trying this example on?

c6

lpy4105, 2025-05-22:
I can't reproduce the issue on my end. The esp32c6 target should be supported. Can you do a fullclean by `idf.py fullclean` and retry?

xieqinan, 2025-08-12:
Hi @wiktorpyk ,

Is the https://github.com/espressif/esp-zigbee-sdk/issues/654#issuecomment-2899797837 helpful? If not, please let me know the problems your encountered.

xieqinan, 2025-09-30:
Hi @wiktorpyk ,

Since there has been no response from you for a long time, I intend to close this issue. I believe you can execute the command line below to fix this issue. If the issue still exists, please feel free to re-open it.
``` bash
cd /home/wiktorpyk/repos/esp-zigbee-sdk/examples/esp_zigbee_all_device_types_app

rm -rf build

idf.py set-target esp32c6

idf.py build
```