espressif/esp-zigbee-sdk #655: The coordinator sends an information response.

Mushan-bin, 2025-05-21:
### Question

If the coordinator wants to send custom cluster instructions to the terminal, can the coordinator itself determine whether the terminal has successfully received the instructions?

xieqinan, 2025-05-22:
> can the coordinator itself determine whether the terminal has successfully received the instructions?

Yes, it can. If the `dis_default_resp` field of the custom command is set to `false` and sent by the coordinator, a default response from the terminal will be received by the coordinator. The sequence number in the response should match the return value of `esp_zb_zcl_custom_cluster_cmd_req()`. That can be used as the way to determine whether the terminal has successfully received the instructions?

xieqinan, 2025-05-22:
> Is it this one?

Yes.

Mushan-bin, 2025-05-28:
Is it only necessary to configure the coordinator? Does the terminal not need to be configured at all? @xieqinan

closed 2025-09-30 by xieqinan