espressif/esp-zigbee-sdk #656: Created more than 20 Zigbee endpoints issue.

mundevx, 2025-05-22:
### Question

When i created 32 endpoints, i got this error after successfully binding.  

Zigbee stack assertion failed common/zb_bufpool_mult.c:1233

abort() was called at PC 0x4201d2b9 on core 0
--- 0x4201d2b9: zb_assert at ??:?

Core  0 register dump:
MEPC    : 0x4080063e  RA      : 0x40806ad4  SP      : 0x4081c1a0  GP      : 0x4080dbe0
--- 0x4080063e: panic_abort at C:/esp/esp-idf/components/esp_system/panic.c:472
0x40806ad4: __ubsan_include at C:/esp/esp-idf/components/esp_system/ubsan.c:313

TP      : 0x408084f4  T0      : 0x37363534  T1      : 0x7271706f  T2      : 0x33323130
--- 0x408084f4: xTaskResumeAll at C:/esp/esp-idf/components/freertos/FreeRTOS-Kernel/tasks.c:2724

S0/FP   : 0x00000004  S1      : 0x4081c204  A0      : 0x4081c1cc  A1      : 0x4081c202
A2      : 0x00000000  A3      : 0x4081c1f9  A4      : 0x00000001  A5      : 0x40814000
A6      : 0x00000000  A7      : 0x76757473  S2      : 0x00000000  S3      : 0x00000083
S4      : 0x00b8b838  S5      : 0x00000000  S6      : 0x000186a0  S7      : 0xfffffffe
S8      : 0x00000000  S9      : 0x00000000  S10     : 0x00000000  S11     : 0x00000091
T3      : 0x6e6d6c6b  T4      : 0x6a696867  T5      : 0x66656463  T6      : 0x62613938
MSTATUS : 0x00001881  MTVEC   : 0x40800001  MCAUSE  : 0x00000007  MTVAL   : 0x00000000  
--- 0x40800001: _vector_table at ??:?

MHARTID : 0x00000000

Stack memory:
4081c1a0: 0x00000000 0x00000000 0x4081c200 0x4080c2a8 0x000186a0 0x00000000 0x00b8b838 0x4080e70c
--- 0x4080c2a8: abort at C:/esp/esp-idf/components/newlib/abort.c:35 (discriminator 3)

4081c1c0: 0x4081c204 0x4080e728 0x4081c200 0x726f6261 0x20292874 0x20736177 0x6c6c6163 0x61206465
4081c1e0: 0x43502074 0x34783020 0x64313032 0x20396232 0x63206e6f 0x2065726f 0x00000030 0x42010000
--- 0x42010000: zb_zcl_device_callback at esp_zigbee_core.c.obj:?

4081c200: 0x00010030 0x31303234 0x39623264 0x00000000 0x0000002f 0x00000038 0x4209810d 0x4201d2bc
--- 0x4201d2bc: zb_aps_multi_pre_link_key_init at zb_init_common.c.obj:?

4081c220: 0x00000000 0x40814000 0x00000000 0x000004d1 0x00000000 0x00000000 0x40821888 0x4201c6f4
--- 0x4201c6f4: zb_get_buf_tail_ptr at zb_bufpool_mult.c.obj:?

4081c240: 0x000186a0 0x00000000 0x00b8b838 0x7fffffff 0x00000045 0x40836385 0x40810180 0x42062756
--- 0x42062756: zb_zcl_cvc_next_step_alarm at zcl_continuous_value_change_commands.c.obj:?

4081c260: 0x42062728 0x00000091 0x40810180 0x42022054 0x00000000 0x00000000 0x4082a930 0x00000045
--- 0x42062728: zb_zcl_cvc_next_step_alarm at zcl_continuous_value_change_commands.c.obj:?

What i have to do now? 
0x42022054: zb_sched_loop_iteration at ??:?

4081c280: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4081c2a0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x42010d04
--- 0x42010d04: esp_zb_stack_main_loop_iteration at ??:?

4081c2c0: 0x00000000 0x00000000 0x00000000 0x42010d1a 0x00000000 0x00000000 0x00000000 0x42007750
--- 0x42010d1a: esp_zb_stack_main_loop at ??:?
0x42007750: esp_zb_task at D:/Projects/NUOS_Zigbee/esp_zigbee_HA_ControllerModule/main/esp_zb_controller_main.c:1670 (discriminator 2)

4081c2e0: 0x00000001 0x00000000 0x00000005 0x00000000 0x00000000 0x00000000 0x00000000 0x408092e8
--- 0x408092e8: vPortTaskWrapper at C:/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/riscv/port.c:205

xieqinan, 2025-05-22:
Have you increase the size of the binding table, which is 16 by default? refer to: https://github.com/espressif/esp-zigbee-sdk/blob/main/components/esp-zigbee-lib/include/esp_zigbee_core.h#L142

mundevx, 2025-05-22:
@xieqinan ,

I already set it. 
    esp_zb_io_buffer_size_set(64); 
    esp_zb_scheduler_queue_size_set(64); 
    esp_zb_aps_src_binding_table_size_set(100);  //this requires
    esp_zb_overall_network_size_set(32); // Coordinator + 31 devices

mundevx, 2025-05-22:
All endpoints are bind successfully!!

I (2781) ZB_CLUSTERS: Light Level 0 Bound successfully!

I (2795) ZB_CLUSTERS: Light State 1 Bound successfully!

I (2898) ZB_CLUSTERS: Light Level 1 Bound successfully!

I (2912) ZB_CLUSTERS: Light State 2 Bound successfully!

I (3030) ZB_CLUSTERS: Light Level 2 Bound successfully!

I (3044) ZB_CLUSTERS: Light State 3 Bound successfully!

I (3147) ZB_CLUSTERS: Light Level 3 Bound successfully!

I (3160) ZB_CLUSTERS: Light State 4 Bound successfully!

I (3277) ZB_CLUSTERS: Light Level 4 Bound successfully!

I (3291) ZB_CLUSTERS: Light State 5 Bound successfully!

I (3399) ZB_CLUSTERS: Light Level 5 Bound successfully!

I (3413) ZB_CLUSTERS: Light State 6 Bound successfully!

I (3530) ZB_CLUSTERS: Light Level 6 Bound successfully!

I (3544) ZB_CLUSTERS: Light State 7 Bound successfully!

I (3647) ZB_CLUSTERS: Light Level 7 Bound successfully!

I (3661) ZB_CLUSTERS: Light State 8 Bound successfully!

I (3777) ZB_CLUSTERS: Light Level 8 Bound successfully!

I (3791) ZB_CLUSTERS: Light State 9 Bound successfully!

I (3898) ZB_CLUSTERS: Light Level 9 Bound successfully!

I (3912) ZB_CLUSTERS: Light State 10 Bound successfully!

I (4031) ZB_CLUSTERS: Light Level 10 Bound successfully!

I (4044) ZB_CLUSTERS: Light State 11 Bound successfully!

I (4154) ZB_CLUSTERS: Light Level 11 Bound successfully!

I (4168) ZB_CLUSTERS: Light State 12 Bound successfully!

I (4283) ZB_CLUSTERS: Light Level 12 Bound successfully!

I (4297) ZB_CLUSTERS: Light State 13 Bound successfully!

I (4405) ZB_CLUSTERS: Light Level 13 Bound successfully!

I (4420) ZB_CLUSTERS: Light State 14 Bound successfully!

I (4534) ZB_CLUSTERS: Light Level 14 Bound successfully!

I (4549) ZB_CLUSTERS: Light State 15 Bound successfully!

I (4656) ZB_CLUSTERS: Light Level 15 Bound successfully!

I (4670) ZB_CLUSTERS: Light State 16 Bound successfully!

I (4783) ZB_CLUSTERS: Light Level 16 Bound successfully!

I (4797) ZB_CLUSTERS: Light State 17 Bound successfully!

I (4905) ZB_CLUSTERS: Light Level 17 Bound successfully!

I (4920) ZB_CLUSTERS: Light State 18 Bound successfully!

I (5035) ZB_CLUSTERS: Light Level 18 Bound successfully!

I (5049) ZB_CLUSTERS: Light State 19 Bound successfully!

I (5153) ZB_CLUSTERS: Light Level 19 Bound successfully!

I (5167) ZB_CLUSTERS: Light State 20 Bound successfully!

I (5283) ZB_CLUSTERS: Light Level 20 Bound successfully!

I (5297) ZB_CLUSTERS: Light State 21 Bound successfully!

I (5406) ZB_CLUSTERS: Light Level 21 Bound successfully!

I (5421) ZB_CLUSTERS: Light State 22 Bound successfully!

I (5540) ZB_CLUSTERS: Light Level 22 Bound successfully!

I (5555) ZB_CLUSTERS: Light State 23 Bound successfully!

I (5659) ZB_CLUSTERS: Light Level 23 Bound successfully!

I (5673) ZB_CLUSTERS: Light State 24 Bound successfully!

I (5791) ZB_CLUSTERS: Light Level 24 Bound successfully!

I (5805) ZB_CLUSTERS: Light State 25 Bound successfully!

I (5912) ZB_CLUSTERS: Light Level 25 Bound successfully!

I (5927) ZB_CLUSTERS: Light State 26 Bound successfully!

I (6041) ZB_CLUSTERS: Light Level 26 Bound successfully!

I (6056) ZB_CLUSTERS: Light State 27 Bound successfully!

I (6158) ZB_CLUSTERS: Light Level 27 Bound successfully!

I (6172) ZB_CLUSTERS: Light State 28 Bound successfully!

I (6282) ZB_CLUSTERS: Light Level 28 Bound successfully!

I (6297) ZB_CLUSTERS: Light State 29 Bound successfully!

I (6402) ZB_CLUSTERS: Light Level 29 Bound successfully!

I (6417) ZB_CLUSTERS: Light State 30 Bound successfully!

I (6542) ZB_CLUSTERS: Light Level 30 Bound successfully!

I (6569) ZB_CLUSTERS: Light State 31 Bound successfully!

I (6687) ZB_CLUSTERS: Light Level 31 Bound successfully!

I (6705) ZB_CLUSTERS: Light State 32 Bound successfully!

But crash occur is in attribute callback, only 11 endpoints when control from Home Assistant are working OK with espressif zigbee device, Above 11, it crashes.

Guru Meditation Error: Core  0 panic'ed (Instruction access fault). Exception was unhandled.

Core  0 register dump:
MEPC    : 0x4082d6a4  RA      : 0x42061ef0  SP      : 0x408210b0  GP      : 0x4080eb00
--- 0x42061ef0: zb_zcl_cvc_next_step_alarm at zcl_continuous_value_change_commands.c.obj:?

TP      : 0x4080e690  T0      : 0x40022494  T1      : 0x93400000  T2      : 0x09c7e1b3
--- 0x40022494: multi_heap_internal_unlock in ROM

S0/FP   : 0x4082da28  S1      : 0x408311b5  A0      : 0x00000000  A1      : 0x4082da28
A2      : 0x00000000  A3      : 0x00000000  A4      : 0x00000001  A5      : 0x4082d6a4  
A6      : 0x00000040  A7      : 0x0000094f  S2      : 0x00000013  S3      : 0x7fffffff
S4      : 0x0e3551b6  S5      : 0x00000000  S6      : 0x000186a0  S7      : 0xfffffffe
S8      : 0x00000000  S9      : 0x00000000  S10     : 0x00000000  S11     : 0x0000002f
T3      : 0x00000000  T4      : 0x000000fe  T5      : 0xffffffff  T6      : 0x000000ff
MSTATUS : 0x00001881  MTVEC   : 0x40800001  MCAUSE  : 0x00000001  MTVAL   : 0x4082d6a4  
--- 0x40800001: _vector_table at ??:?

MHARTID : 0x00000000

Stack memory:
408210b0: 0x42061dae 0x0000002f 0x40811510 0x420216ea 0x00000000 0x00000000 0x40826028 0x00000013
--- 0x42061dae: zb_zcl_cvc_next_step_alarm at zcl_continuous_value_change_commands.c.obj:?
0x420216ea: zb_sched_loop_iteration at ??:?

408210d0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
408210f0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x420103ac
--- 0x420103ac: esp_zb_stack_main_loop_iteration at ??:?

40821110: 0x4267695a 0x00000000 0x00000000 0x420103c2 0x00000000 0x00000000 0x00000000 0x42007adc
--- 0x420103c2: esp_zb_stack_main_loop at ??:?
0x42007adc: esp_zb_task at D:/Projects/esp_zigbee_HA_ControllerModule/main/esp_zb_controller_main.c:1659 (discriminator 2)

40821130: 0x00000001 0x00000000 0x00000005 0x00000000 0x00000000 0x00000000 0x00000000 0x4080a48c
--- 0x4080a48c: vPortTaskWrapper at C:/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/riscv/port.c:205

mundevx, 2025-05-22:
Here is my observation:

This log is for Endpoint 1
I (40796) ZB_MAIN_FILE: Received message B: endpoint(1), cluster(0x8), attribute(0x0), data (0)
I (40805) ZB_MAIN_FILE: Received message B: endpoint(1), cluster(0x6), attribute(0x0), data (1)
I (40941) ZB_MAIN_FILE: Received message B: endpoint(1), cluster(0x8), attribute(0x0), data (0)

This log is for Endpoint 14
I (104247) ZB_MAIN_FILE: Received message B: endpoint(14), cluster(0x8), attribute(0x0), data (0)
I (104256) ZB_MAIN_FILE: Received message B: endpoint(14), cluster(0x6), attribute(0x0), data (1)
Guru Meditation Error: Core  0 panic'ed (Instruction access fault). Exception was unhandled.

---> Here is missing callback of cluster 0x08 for endpoint 14 and not in case of endpoint 1 and exception occurs. Means there is some callbacks queue size is to set. how i can do it?  I already set these parameters, but no success..  
    esp_zb_io_buffer_size_set(200); 
    esp_zb_scheduler_queue_size_set(200); 
    esp_zb_aps_src_binding_table_size_set(100);  //this requires
    esp_zb_aps_dst_binding_table_size_set(100);
    esp_zb_overall_network_size_set(100); // Coordinator + 31 devices

Suggest me some solution...

mundevx, 2025-05-23:
@xieqinan 

I am still getting this error by setting esp_zb_aps_src_binding_table_size_set(100); 

Zigbee stack assertion failed common/zb_bufpool_mult.c:1233

abort() was called at PC 0x4201d2b9 on core 0
--- 0x4201d2b9: zb_assert at ??:?

Core 0 register dump: MEPC    : 0x4080063e  RA      : 0x40806ad4  SP      : 0x4081c2b0  GP      : 0x4080dbd0
--- 0x4080063e: panic_abort at C:/esp/esp-idf/components/esp_system/panic.c:472
0x40806ad4: __ubsan_include at C:/esp/esp-idf/components/esp_system/ubsan.c:313

TP      : 0x408087f4  T0      : 0x37363534  T1      : 0x7271706f  T2      : 0x33323130
--- 0x408087f4: vTaskPlaceOnEventListRestricted at C:/esp/esp-idf/components/freertos/FreeRTOS-Kernel/tasks.c:3968 (discriminator 1)

xieqinan, 2025-08-12:
Hi @mundevx ,

Could you please share more background of your project? Or if this issue had been resolved, please consider closing it.

luar123, 2025-08-30:
Hi @xieqinan, I have the same issue with a similar setup: 13 endpoints with analog input clusters and one dimmable light endpoint (number 14). If I try to control the light (e.q. turn on) I get the same error:
```
[19:08:45][D][zigbee:210][Zigbee_main]: Received message: endpoint(14), cluster(0x8), attribute(0x0), data size(1)
[19:08:45][D][zigbee:210][Zigbee_main]: Received message: endpoint(14), cluster(0x6), attribute(0x0), data size(1)
[19:08:45][D][zigbee:258][Zigbee_main]: Receive Zigbee default response callback
[19:08:45]ZB_TRACE_LOG[0]: common/zb_bufpool_mult.c:1229    corrupted buf 0x4082fa30 0
[19:08:45]ZB_TRACE_LOG[0]: common/zb_bufpool_mult.c:1230    buf_len -1 size 56 max_size 21824 ZB_IO_BUF_SIZE 152
[19:08:45]ZB_TRACE_LOG[0]: common/zb_debug.c:64    Assertion failed /builds/thread_zigbee/esp-zboss/components/zboss_stack/zboss/common/zb_bufpool_mult.c:1233
[19:08:45]Zigbee stack assertion failed common/zb_bufpool_mult.c:1233
```
I tried with default config and additionally with settings:
```
  esp_zb_io_buffer_size_set(128);
  esp_zb_scheduler_queue_size_set(128);
  esp_zb_aps_src_binding_table_size_set(64);
  esp_zb_aps_dst_binding_table_size_set(64);
  esp_zb_zcl_scenes_table_set_size(64);
```
Both fail and even setting `esp_zb_io_buffer_size_set(4096);` doesn't help.

If I put the light on endpoint 1 everything is fine.

xieqinan, 2025-09-01:
Hi @luar123 ,

Could you please share a simple reproducible code snippet and steps? That would help us address the issue more quickly.

luar123, 2025-09-15:
@xieqinan can you reproduce?

xieqinan, 2026-01-20:
Hi @mundevx,

This https://github.com/espressif/esp-zigbee-sdk/issues/656#issuecomment-3294739019 has explain this issue, Do you have any updates or additional information regarding this issue? If not, please consider closing it.