espressif/esp-zigbee-sdk #660: ESP32H2 as a router Not able to connect with ETRX357-LRS Coordinator

MAXJADAV-3599, 2025-05-26:
Hello There,
I am using esp-idf v5.4.1,esp_zigbee_lib 1.6.5 and esp_zboss_lib 1.6.4.
I am trying to connect my esp32h2 device with the coordinator from ETRX357-LRS Zigbee module.

xieqinan, 2025-05-27:
Hi,

Based on the above log and source code, it appears that the `ETRX357-LRS` indeed does not enable APS security key to encrypt the `Transport-Key` command, which is a non-standard behavior. Please refer to Section 4.4.2.3 of the Zigbee R22 specification. If this is confirmed, I'm afraid the ESP32-H2 Zigbee device will not be able to join the network created by the `ETRX357-LRS`.

```
Upon receipt of a secured transport-key command, the APSME shall check the key type sub-field. If the key type field
is set to 0x03 or 0x04 (that is, application link or Trust Center link key) and the receiving device is operating in the
joined and authorized state and the command was not secured using a distributed security link key or a Trust Center
link key, the command shall be discarded. If the device is operating in the joined and authorized state it may accept a
NWK broadcast transport key command with Key type field set to 0x01 (that is, network key) where the message has
no APS encryption. If the key type field is set to 0x01 (that is, network key) and the command was not secured using a
distributed security link key, Trust Center link key, the command shall be discarded.
```

By the way, do you attempt to use the other Third-Party devices to join the `ETRX357-LRS`?

xieqinan, 2025-06-13:
> Yes, I am having a third-party module which is of MGM210PA32 and using EmberZnet (Zigbee stack). By disabling the APS security in it I am able to join with ETRX357-LRS.

OK, the `esp-zigbee-sdk` (v1.6.5) does not yet support disabling the APS layer key, but we plan to cover this scenario in a future version. Once the feature is supported, I’ll provide an update here.

MAXJADAV-3599, 2025-07-03:
Can I expect new version to be available before 15/07/2025?
Because I am already very late to complete this task and my development is stuck!!!
Please make it available in short time....

xieqinan, 2025-07-04:
Hi @MAXJADAV-3599 ,

The new version will be released within this month. Thank you for your patience.