espressif/esp-zigbee-sdk #663: esp_zigbee_gateway fails to link with target ESP32C5 (IDFGH-15382)

amp043, 2025-05-29:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.4.1

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

Expected the build to succeed.

### What is the actual behavior?

With target set to esp32c5 the project fails to build (at the step below)
[1020/1022] Linking CXX executable esp_zigbee_gateway.elf
With target set to esp32c6 the build succeeds.

### Steps to reproduce.

1. cd ~/esp/esp-idf/examples/zigbee/esp_zigbee_gateway
2. idf.py --preview set-target esp32c5
3. idf.py build

### Build or installation Logs.

```plain
apozsgay@fairfax:~/esp/esp-idf/examples/zigbee/esp_zigbee_gateway ((v5.4.1))
$ idf.py build
Executing action: all (aliases: build)
Running ninja in directory /home/apozsgay/esp/esp-idf/examples/zigbee/esp_zigbee_gateway/build
Executing "ninja all"...
[8/1022] Generating ../../partition_table/partition-table.bin
Partition table binary generated. Contents:
*******************************************************************************
# ESP-IDF Partition Table
# Name, Type, SubType, Offset, Size, Flags
nvs,data,nvs,0x9000,24K,
phy_init,data,phy,0xf000,4K,
factory,app,factory,0x10000,1400K,
zb_storage,data,fat,0x16e000,16K,
zb_fct,data,fat,0x172000,1K,
rcp_fw,data,spiffs,0x173000,500K,
*******************************************************************************
[339/1022] Performing configure step for 'bootloader'
-- Found Git: /usr/bin/git (found version "2.25.1")
-- The C compiler identification is GNU 14.2.0
-- The CXX compiler identification is GNU 14.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /home/apozsgay/.espressif/tools/riscv32-esp-elf/esp-14.2.0_20241119/riscv32-esp-elf/bin/riscv32-esp-elf-gcc
-- Check for working C compiler: /home/apozsgay/.espressif/tools/riscv32-esp-elf/esp-14.2.0_20241119/riscv32-esp-elf/bin/riscv32-esp-elf-gcc
-- Check for working C compiler: /home/apozsgay/.espressif/tools/riscv32-esp-elf/esp-14.2.0_20241119/riscv32-esp-elf/bin/riscv32-esp-elf-gcc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
[... 34 lines cut ...]
FAILED: esp_zigbee_gateway.elf
: && /home/apozsgay/.espressif/tools/riscv32-esp-elf/esp-14.2.0_20241119/riscv32-esp-elf/bin/riscv32-esp-elf-g++  -march=rv32imac_zicsr_zifencei  -nostartfiles -march=rv32imac_zicsr_zifencei --specs=nosys.specs    -Wl,--cref -Wl,--defsym=IDF_TARGET_ESP32C5=0 -Wl,--Map=/home/apozsgay/esp/esp-idf/examples/zigbee/esp_zigbee_gateway/build/esp_zigbee_gateway.map -Wl,--no-warn-rwx-segments -Wl,--orphan-handling=warn -fno-rtti -fno-lto -Wl,--gc-sections -Wl,--warn-common -T rom.api.ld -T esp32c5.peripherals.ld -T esp32c5.rom.ld -T esp32c5.rom.api.ld -T esp32c5.rom.rvfp.ld -T esp32c5.rom.wdt.ld -T esp32c5.rom.systimer.ld -T esp32c5.rom.version.ld -T esp32c5.rom.phy.ld -T esp32c5.rom.coexist.ld -T esp32c5.rom.net80211.ld -T esp32c5.rom.pp.ld -T esp32c5.rom.newlib.ld -T esp32c5.rom.newlib-normal.ld -T esp32c5.rom.heap.ld -T memory.ld -T sections.ld CMakeFiles/esp_zigbee_gateway.elf.dir/project_elf_src_esp32c5.c.obj  -o esp_zigbee_gateway.elf -L/home/apozsgay/esp/esp-idf/components/riscv/ld   -L/home/apozsgay/esp/esp-idf/components/soc/esp32c5/ld   -L/home/apozsgay/esp/esp-idf/components/esp_rom/esp32c5/ld   -L/home/apozsgay/esp/esp-idf/examples/zigbee/esp_zigbee_gateway/build/esp-idf/esp_system/ld   -L/home/apozsgay/esp/esp-idf/components/esp_phy/lib/esp32c5   -L/home/apozsgay/esp/esp-idf/components/esp_coex/lib/esp32c5   -L/home/apozsgay/esp/esp-idf/components/esp_wifi/lib/esp32c5 esp-idf/riscv/libriscv.a  esp-idf/esp_driver_gpio/libesp_driver_gpio.a  esp-idf/esp_pm/libesp_pm.a  esp-idf/mbedtls/libmbedtls.a  esp-idf/esp_app_format/libesp_app_format.a  esp-idf/esp_bootloader_format/libesp_bootloader_format.a  esp-idf/app_update/libapp_update.a  esp-idf/esp_partition/libesp_partition.a  esp-idf/efuse/libefuse.a  esp-idf/bootloader_support/libbootloader_support.a  esp-idf/esp_mm/libesp_mm.a  esp-idf/spi_flash/libspi_flash.a  esp-idf/esp_system/libesp_system.a  esp-idf/esp_common/libesp_common.a  esp-idf/esp_rom/libesp_rom.a  esp-idf/hal/libhal.a  esp-idf/log/liblog.a  esp-idf/heap/libheap.a  esp-idf/soc/libsoc.a  esp-idf/esp_security/libesp_security.a  esp-idf/esp_hw_support/libesp_hw_support.a  esp-idf/freertos/libfreertos.a  esp-idf/newlib/libnewlib.a  esp-idf/pthread/libpthread.a  esp-idf/cxx/libcxx.a  esp-idf/esp_timer/libesp_timer.a  esp-idf/esp_driver_gptimer/libesp_driver_gptimer.a  esp-idf/esp_ringbuf/libesp_ringbuf.a  esp-idf/esp_driver_uart/libesp_driver_uart.a  esp-idf/app_trace/libapp_trace.a  esp-idf/esp_event/libesp_event.a  esp-idf/nvs_flash/libnvs_flash.a  esp-idf/esp_driver_pcnt/libesp_driver_pcnt.a  esp-idf/esp_driver_spi/libesp_driver_spi.a  esp-idf/esp_driver_mcpwm/libesp_driver_mcpwm.a  esp-idf/esp_driver_i2s/libesp_driver_i2s.a  esp-idf/sdmmc/libsdmmc.a  esp-idf/esp_driver_sdspi/libesp_driver_sdspi.a  esp-idf/esp_driver_rmt/libesp_driver_rmt.a  esp-idf/esp_driver_tsens/libesp_driver_tsens.a  esp-idf/esp_driver_sdm/libesp_driver_sdm.a  esp-idf/esp_driver_i2c/libesp_driver_i2c.a  esp-idf/esp_driver_ledc/libesp_driver_ledc.a  esp-idf/esp_driver_parlio/libesp_driver_parlio.a  esp-idf/esp_driver_usb_serial_jtag/libesp_driver_usb_serial_jtag.a  esp-idf/driver/libdriver.a  esp-idf/esp_phy/libesp_phy.a  esp-idf/esp_vfs_console/libesp_vfs_console.a  esp-idf/vfs/libvfs.a  esp-idf/lwip/liblwip.a  esp-idf/esp_netif/libesp_netif.a  esp-idf/wpa_supplicant/libwpa_supplicant.a  esp-idf/esp_coex/libesp_coex.a  esp-idf/esp_wifi/libesp_wifi.a  esp-idf/unity/libunity.a  esp-idf/cmock/libcmock.a  esp-idf/console/libconsole.a  esp-idf/http_parser/libhttp_parser.a  esp-idf/esp-tls/libesp-tls.a  esp-idf/esp_adc/libesp_adc.a  esp-idf/esp_driver_cam/libesp_driver_cam.a  esp-idf/esp_eth/libesp_eth.a  esp-idf/esp_gdbstub/libesp_gdbstub.a  esp-idf/esp_hid/libesp_hid.a  esp-idf/tcp_transport/libtcp_transport.a  esp-idf/esp_http_client/libesp_http_client.a  esp-idf/esp_http_server/libesp_http_server.a  esp-idf/esp_https_ota/libesp_https_ota.a  esp-idf/esp_https_server/libesp_https_server.a  esp-idf/esp_lcd/libesp_lcd.a  esp-idf/protobuf-c/libprotobuf-c.a  esp-idf/protocomm/libprotocomm.a  esp-idf/esp_local_ctrl/libesp_local_ctrl.a  esp-idf/espcoredump/libespcoredump.a  esp-idf/wear_levelling/libwear_levelling.a  esp-idf/fatfs/libfatfs.a  esp-idf/ieee802154/libieee802154.a  esp-idf/json/libjson.a  esp-idf/mqtt/libmqtt.a  esp-idf/nvs_sec_provider/libnvs_sec_provider.a  esp-idf/rt/librt.a  esp-idf/spiffs/libspiffs.a  esp-idf/wifi_provisioning/libwifi_provisioning.a  esp-idf/protocol_examples_common/libprotocol_examples_common.a  esp-idf/main/libmain.a  esp-idf/app_trace/libapp_trace.a  esp-idf/app_trace/libapp_trace.a  esp-idf/cmock/libcmock.a  esp-idf/unity/libunity.a  esp-idf/esp_driver_cam/libesp_driver_cam.a  esp-idf/esp_hid/libesp_hid.a  esp-idf/esp_lcd/libesp_lcd.a  esp-idf/esp_local_ctrl/libesp_local_ctrl.a  esp-idf/esp_https_server/libesp_https_server.a  esp-idf/espcoredump/libespcoredump.a  -u esp_system_include_coredump_init  esp-idf/fatfs/libfatfs.a  esp-idf/wear_levelling/libwear_levelling.a  esp-idf/mqtt/libmqtt.a  esp-idf/nvs_sec_provider/libnvs_sec_provider.a  -u nvs_sec_provider_include_impl  esp-idf/rt/librt.a  esp-idf/spiffs/libspiffs.a  esp-idf/wifi_provisioning/libwifi_provisioning.a  esp-idf/protocomm/libprotocomm.a  esp-idf/protobuf-c/libprotobuf-c.a  esp-idf/json/libjson.a  ../managed_components/espressif__esp-zigbee-lib/lib/esp32c5/libesp_zb_api.zczr.a  ../managed_components/espressif__esp-zboss-lib/lib/esp32c5/libzboss_stack.zczr.a  ../managed_components/espressif__esp-zigbee-lib/lib/esp32c5/libzboss_port.native.a  esp-idf/ieee802154/libieee802154.a  esp-idf/protocol_examples_common/libprotocol_examples_common.a  esp-idf/console/libconsole.a  esp-idf/esp_eth/libesp_eth.a  esp-idf/riscv/libriscv.a  esp-idf/esp_driver_gpio/libesp_driver_gpio.a  esp-idf/esp_pm/libesp_pm.a  esp-idf/mbedtls/libmbedtls.a  esp-id [... cut ...] df_py_stdout_output_2973558
```

### Diagnostic report archive.

[idf-diag-6f685298-8f9b-483a-81c5-bab7872261ee.zip](https://github.com/user-attachments/files/20491364/idf-diag-6f685298-8f9b-483a-81c5-bab7872261ee.zip)

### More Information.

I also tried the latest (master) esp-idf:
v5.5-dev-3951-ga74725a31b
However, I still get link errors here (many more than with v5.4.1).
Note that the diagnostic report attached above was generated with v5.5 because v5.4.1 does not support report generation.

xieqinan, 2025-05-29:
Due to the ESP32C5 being in the preview stage, please switch to the `master` branch to use it.

xieqinan, 2025-05-30:
I just tested the command line below and it worked fine. Please follow it and test again.

```sh
cd ~/esp/esp-idf

git pull origin master

git submodule update --init --recursive

./install.sh && . ./export.sh

cd /examples/zigbee/esp_zigbee_gateway

idf.py --preview set-target esp32c5

idf.py fullclean

idf.py build

idf.py -p [PORT] erase-flash flash monitor

```