espressif/esp-zigbee-sdk #664: zigbee example do not compile zcl_utility/src' does not exist. (IDFGH-15319)

grigio, 2025-05-29:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

master

### Operating System used.

Linux

### How did you build your project?

Command line with Make

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

I'd like to compile a zigbee example

### What is the actual behavior?

CMake Error at /home/grigio/esp/master/esp-idf/tools/cmake/component.cmake:286 (message):
  SRC_DIRS entry '../../../common/zcl_utility/src' does not exist.
Call Stack (most recent call first):
  /home/grigio/esp/master/esp-idf/tools/cmake/component.cmake:455 (__component_add_sources)
  main/CMakeLists.txt:1 (idf_component_register)

-- Configuring incomplete, errors occurred!

### Steps to reproduce.

```
Executing task: /usr/bin/cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -B /home/grigio/Code/HA_on_off_light/HA_on_off_switch/build -S /home/grigio/Code/HA_on_off_light/HA_on_off_switch -DSDKCONFIG='/home/grigio/Code/HA_on_off_light/HA_on_off_switch/sdkconfig' 

-- Found Git: /usr/bin/git (found version "2.49.0")
-- Minimal build - ON
-- The C compiler identification is GNU 14.2.0
-- The CXX compiler identification is GNU 14.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /home/grigio/.espressif/tools/riscv32-esp-elf/esp-14.2.0_20241119/riscv32-esp-elf/bin/riscv32-esp-elf-gcc
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /home/grigio/.espressif/tools/riscv32-esp-elf/esp-14.2.0_20241119/riscv32-esp-elf/bin/riscv32-esp-elf-gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /home/grigio/.espressif/tools/riscv32-esp-elf/esp-14.2.0_20241119/riscv32-esp-elf/bin/riscv32-esp-elf-g++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building ESP-IDF components for target esp32c5
NOTICE: Processing 3 dependencies:
NOTICE: [1/3] espressif/esp-zboss-lib (1.6.4)
NOTICE: [2/3] espressif/esp-zigbee-lib (1.6.4)
NOTICE: [3/3] idf (6.0.0)
-- Project sdkconfig file /home/grigio/Code/HA_on_off_light/HA_on_off_switch/sdkconfig
Loading defaults file /home/grigio/Code/HA_on_off_light/HA_on_off_switch/sdkconfig.defaults...
-- Adding linker script /home/grigio/esp/master/esp-idf/components/riscv/ld/rom.api.ld
CMake Deprecation Warning at /home/grigio/esp/master/esp-idf/components/mbedtls/mbedtls/CMakeLists.txt:21 (cmake_minimum_required):
  Compatibility with CMake < 3.10 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value.  Or, use the <min>...<max> syntax
  to tell CMake that the project requires at least <min> but has been updated
  to work with policies introduced by <max> or earlier.

-- Found Python3: /home/grigio/.espressif/python_env/idf6.0_py3.9_env/bin/python (found version "3.9.10") found components: Interpreter
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Success
-- Found Threads: TRUE
-- Performing Test C_COMPILER_SUPPORTS_WFORMAT_SIGNEDNESS
-- Performing Test C_COMPILER_SUPPORTS_WFORMAT_SIGNEDNESS - Success
-- USING O3
-- App "light_switch" version: b5ab689
-- Adding linker script /home/grigio/Code/HA_on_off_light/HA_on_off_switch/build/esp-idf/esp_system/ld/memory.ld
-- Adding linker script /home/grigio/Code/HA_on_off_light/HA_on_off_switch/build/esp-idf/esp_system/ld/sections.ld.in
-- Adding linker script /home/grigio/esp/master/esp-idf/components/esp_rom/esp32c5/ld/esp32c5.rom.ld
-- Adding linker script /home/grigio/esp/master/esp-idf/components/esp_rom/esp32c5/ld/esp32c5.rom.api.ld
-- Adding linker script /home/grigio/esp/master/esp-idf/components/esp_rom/esp32c5/ld/esp32c5.rom.rvfp.ld
-- Adding linker script /home/grigio/esp/master/esp-idf/components/esp_rom/esp32c5/ld/esp32c5.rom.wdt.ld
-- Adding linker script /home/grigio/esp/master/esp-idf/components/esp_rom/esp32c5/ld/esp32c5.rom.systimer.ld
-- Adding linker script /home/grigio/esp/master/esp-idf/components/esp_rom/esp32c5/ld/esp32c5.rom.version.ld
-- Adding linker script /home/grigio/esp/master/esp-idf/components/esp_rom/esp32c5/ld/esp32c5.rom.phy.ld
-- Adding linker script /home/grigio/esp/master/esp-idf/components/esp_rom/esp32c5/ld/esp32c5.rom.coexist.ld
-- Adding linker script /home/grigio/esp/master/esp-idf/components/esp_rom/esp32c5/ld/esp32c5.rom.net80211.ld
-- Adding linker script /home/grigio/esp/master/esp-idf/components/esp_rom/esp32c5/ld/esp32c5.rom.pp.ld
-- Adding linker script /home/grigio/esp/master/esp-idf/components/esp_rom/esp32c5/ld/esp32c5.rom.libc.ld
-- Adding linker script /home/grigio/esp/master/esp-idf/components/esp_rom/esp32c5/ld/esp32c5.rom.libc-suboptimal_for_misaligned_mem.ld
-- Adding linker script /home/grigio/esp/master/esp-idf/components/esp_rom/esp32c5/ld/esp32c5.rom.newlib.ld
-- Adding linker script /home/grigio/esp/master/esp-idf/components/esp_rom/esp32c5/ld/esp32c5.rom.heap.ld
-- Adding linker script /home/grigio/esp/master/esp-idf/components/soc/esp32c5/ld/esp32c5.peripherals.ld
f9765d0316-b5ab689-1.6.4-2025-05-17 07:30:37 UTC
CMake Error at /home/grigio/esp/master/esp-idf/tools/cmake/component.cmake:286 (message):
  SRC_DIRS entry '../../../common/zcl_utility/src' does not exist.
Call Stack (most recent call first):
  /home/grigio/esp/master/esp-idf/tools/cmake/component.cmake:455 (__component_add_sources)
  main/CMakeLists.txt:1 (idf_component_register)

-- Configuring incomplete, errors occurred!

 *  The terminal process "/usr/bin/cmake '-G', 'Ninja', '-DPYTHON_DEPS_CHECKED=1', '-DESP_PLATFORM=1', '-B', '/home/grigio/Code/HA_on_off_light/HA_on_off_switch/build', '-S', '/home/grigio/Code/HA_on_off_light/HA_on_off_switch', '-DSDKCONFIG='/home/grigio/Code/HA_on_off_light/HA_on_off_switch/sdkconfig''" terminated with exit code: 1. 
```

### Build or installation Logs.

```plain

```

xieqinan, 2025-06-11:
Hi @grigio ，

The issue has been fixed in this commit: https://github.com/espressif/esp-idf/commit/1f45f98573f5ece137b78440c354e249971b873d, let's close it.

closed 2025-06-11 by xieqinan

Salim-m14, 2025-10-28:
Hi @xieqinan and @chshu,

On Windows 11, using `$ENV{IDF_PATH}` inside CMakeLists.txt causes a CMake parsing error, even when forward slashes `/` are used in the path.

CMake Error at build/component_requires.temp.cmake:4 (__component_set_property):

We can find inside the component_requires.temp.cmake : 

__component_set_property(___idf_main INCLUDE_DIRS ".;C:\Users\USERNAME\esp\v5.4.3\esp-idf/examples/zigbee/common/zcl_utility/include")

Using `${IDF_PATH}` without **ENV** everything is working fine. I am beginner on git, i don't really no if it's an issue only for me or also other people, i am just letting you know.