espressif/esp-zigbee-sdk #669: Thermostat cluster attributes reporting

luisyague, 2025-06-04:
### Question

Hello,

I am trying to develop a zigbee heater in which I must implement the thermostat cluster. I am being able to send reports on the localTemperature attribute but I am getting a fault when trying to report the running state. I know this attribute is not reportable in ZCL but I wonder if there's any way to report this value. I've already read this issue but I am having no success (https://github.com/espressif/esp-zigbee-sdk/issues/605#issuecomment-2749890439). The versions I am using are: 
- idf-Version = 5.2.2
- esp-zBoss-lib = 1.6.4
- esp-zigbee-lib = 1.6.4
- esp32C6

This is the code snippet I am using: 
```
void CreateThermostatCluster() {
  esp_zb_thermostat_cluster_cfg_t thermostatClusterConfig = {
      .local_temperature = thmClusterAttributes.localTemp,
      .occupied_cooling_setpoint = ESP_ZB_ZCL_THERMOSTAT_OCCUPIED_COOLING_SETPOINT_DEFAULT_VALUE,
      .occupied_heating_setpoint = thmClusterAttributes.occupiedHeatingSetPoint,
      .control_sequence_of_operation = ESP_ZB_ZCL_THERMOSTAT_CONTROL_SEQ_OF_OPERATION_HEATING_ONLY,
      .system_mode = thmClusterAttributes.systemMode,
  };

  thermostatCluster = esp_zb_thermostat_cluster_create(&thermostatClusterConfig);

  // Add attributes to the Thermostat Cluster
  ESP_ERROR_CHECK(esp_zb_thermostat_cluster_add_attr(thermostatCluster, ESP_ZB_ZCL_ATTR_THERMOSTAT_OCCUPANCY_ID, &thmClusterAttributes.occupancy));
  ESP_ERROR_CHECK(esp_zb_thermostat_cluster_add_attr(thermostatCluster, ESP_ZB_ZCL_ATTR_THERMOSTAT_ABS_MIN_HEAT_SETPOINT_LIMIT_ID, &thmClusterAttributes.absMinHeatSetPointLimit));
  ESP_ERROR_CHECK(esp_zb_thermostat_cluster_add_attr(thermostatCluster, ESP_ZB_ZCL_ATTR_THERMOSTAT_ABS_MAX_HEAT_SETPOINT_LIMIT_ID, &thmClusterAttributes.absMaxSetPointLimit));
  ESP_ERROR_CHECK(esp_zb_thermostat_cluster_add_attr(thermostatCluster, ESP_ZB_ZCL_ATTR_THERMOSTAT_LOCAL_TEMPERATURE_CALIBRATION_ID, &thmClusterAttributes.localTempCalibration));
  ESP_ERROR_CHECK(esp_zb_thermostat_cluster_add_attr(thermostatCluster, ESP_ZB_ZCL_ATTR_THERMOSTAT_MIN_HEAT_SETPOINT_LIMIT_ID, &thmClusterAttributes.minHeatSetPoint));
  ESP_ERROR_CHECK(esp_zb_thermostat_cluster_add_attr(thermostatCluster, ESP_ZB_ZCL_ATTR_THERMOSTAT_MAX_HEAT_SETPOINT_LIMIT_ID, &thmClusterAttributes.maxHeatSetPoint));
  ESP_ERROR_CHECK(esp_zb_custom_cluster_add_custom_attr(thermostatCluster, ESP_ZB_ZCL_ATTR_THERMOSTAT_THERMOSTAT_RUNNING_STATE_ID, 
                                                             ESP_ZB_ZCL_ATTR_TYPE_16BITMAP, 
                                                             ESP_ZB_ZCL_ATTR_ACCESS_READ_ONLY | ESP_ZB_ZCL_ATTR_ACCESS_REPORTING, 
                                                             &thmClusterAttributes.thmRunningState));
  ESP_ERROR_CHECK(esp_zb_thermostat_cluster_add_attr(thermostatCluster, ESP_ZB_ZCL_ATTR_THERMOSTAT_SETPOINT_CHANGE_SOURCE_ID, &thmClusterAttributes.setpointChangeSource));
  ESP_ERROR_CHECK(esp_zb_thermostat_cluster_add_attr(thermostatCluster, ESP_ZB_ZCL_ATTR_THERMOSTAT_SETPOINT_CHANGE_SOURCE_TIMESTAMP_ID, &thmClusterAttributes.setpointChangeSourceTimestamp));
  ESP_ERROR_CHECK(esp_zb_thermostat_cluster_add_attr(thermostatCluster, ESP_ZB_ZCL_ATTR_THERMOSTAT_THERMOSTAT_PROGRAMMING_OPERATION_MODE_ID, &thmClusterAttributes.thermostatProgrammingOperationMode));
  ESP_ERROR_CHECK(esp_zb_thermostat_cluster_add_attr(thermostatCluster, ESP_ZB_ZCL_ATTR_THERMOSTAT_NUMBER_OF_WEEKLY_TRANSITIONS_ID, &thmClusterAttributes.numberOfWeeklyTransitions));
  ESP_ERROR_CHECK(esp_zb_thermostat_cluster_add_attr(thermostatCluster, ESP_ZB_ZCL_ATTR_THERMOSTAT_NUMBER_OF_DAILY_TRANSITIONS_ID, &thmClusterAttributes.numberOfDaylyTransitions));
  ESP_ERROR_CHECK(esp_zb_thermostat_cluster_add_attr(thermostatCluster, ESP_ZB_ZCL_ATTR_THERMOSTAT_START_OF_WEEK_ID, &thmClusterAttributes.startOfWeek));
}

void UpdateStatus(void){
  ThmStatus status;
  static uint16_t mTemp = 200;
  static uint16_t activePower = 600;

  ThmGetStatus(&status);

  mTemp++;
  activePower++;
  bool active = !thmClusterAttributes.thmRunningState;

  thmClusterAttributes.localTemp = mTemp * 10;
  // thmClusterAttributes.thmRunningState = status.active;
  thmClusterAttributes.thmRunningState = 0x0000;
  electricalClusterAttributes.activePower = activePower;

  printf("Setting active %d - mTemp %d - activePower %d\n", active, mTemp, activePower);

  /* --- LocalTemperature ------------------------------------------------ */
  esp_zb_zcl_report_attr_cmd_t report_attr_cmd = {0};
  report_attr_cmd.address_mode = ESP_ZB_APS_ADDR_MODE_16_ENDP_PRESENT;
  report_attr_cmd.zcl_basic_cmd.dst_endpoint = 1;
  report_attr_cmd.zcl_basic_cmd.dst_addr_u.addr_short = 0x0000; // Coordinator
  report_attr_cmd.direction = ESP_ZB_ZCL_CMD_DIRECTION_TO_CLI;
  report_attr_cmd.clusterID = ESP_ZB_ZCL_CLUSTER_ID_THERMOSTAT;
  report_attr_cmd.zcl_basic_cmd.src_endpoint = ENDPOINT_ID;
  report_attr_cmd.attributeID = ESP_ZB_ZCL_ATTR_THERMOSTAT_LOCAL_TEMPERATURE_ID;
  report_attr_cmd.profile_id  = ESP_ZB_AF_HA_PROFILE_ID;

  esp_zb_lock_acquire(portMAX_DELAY);
  esp_zb_zcl_set_attribute_val(ENDPOINT_ID, ESP_ZB_ZCL_CLUSTER_ID_THERMOSTAT, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE, ESP_ZB_ZCL_ATTR_THERMOSTAT_LOCAL_TEMPERATURE_ID, &thmClusterAttributes.localTemp, false);
  esp_zb_zcl_set_attribute_val(ENDPOINT_ID, ESP_ZB_ZCL_CLUSTER_ID_THERMOSTAT, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE, ESP_ZB_ZCL_ATTR_THERMOSTAT_THERMOSTAT_RUNNING_STATE_ID, &thmClusterAttributes.thmRunningState, false);
  esp_zb_zcl_set_attribute_val(ENDPOINT_ID, ESP_ZB_ZCL_CLUSTER_ID_ELECTRICAL_MEASUREMENT, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE, ESP_ZB_ZCL_ATTR_ELECTRICAL_MEASUREMENT_ACTIVE_POWER_ID, &electricalClusterAttributes.activePower, false);
  report_attr_cmd.attributeID = ESP_ZB_ZCL_ATTR_THERMOSTAT_LOCAL_TEMPERATURE_ID;
  ESP_ERROR_CHECK(esp_zb_zcl_report_attr_cmd_req(&report_attr_cmd));
  report_attr_cmd.attributeID = ESP_ZB_ZCL_ATTR_THERMOSTAT_THERMOSTAT_RUNNING_STATE_ID;
  ESP_ERROR_CHECK(esp_zb_zcl_report_attr_cmd_req(&report_attr_cmd));
  esp_zb_lock_release();
}
```
And this is the assertion I get: 
```
ZB_TRACE_LOG[0]: common/zb_debug.c:64    Assertion failed /builds/thread_zigbee/esp-zboss/components/esp_zb_sdk/src/esp_zigbee_zcl_command.c:252
Zigbee stack assertion failed /builds/thread_zigbee/esp-zboss/components/esp_zb_sdk/src/esp_zigbee_zcl_command.c:252
```

Thanks a lot in advance for the support.

xieqinan, 2025-06-05:
@luisyague ,
Please share the complete process of building the data model using `esp_zb_device_register()`.

luisyague, 2025-06-05:
Hello @xieqinan , here is the code: 
```
void CreateEndPoint() {
  esp_zb_endpoint_config_t endPointConfig = {
      .endpoint = ENDPOINT_ID,
      .app_profile_id = ESP_ZB_AF_HA_PROFILE_ID,
      .app_device_id = ESP_ZB_HA_THERMOSTAT_DEVICE_ID,
      .app_device_version = 0
  };

  esp_zb_ep_list_t *endPointList = esp_zb_ep_list_create();

  esp_zb_cluster_list_t *clusterList = esp_zb_zcl_cluster_list_create();

  ESP_ERROR_CHECK(esp_zb_cluster_list_add_basic_cluster(clusterList, basicCluster, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE));
  ESP_ERROR_CHECK(esp_zb_cluster_list_add_identify_cluster(clusterList, identifyServerCluster, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE));
  ESP_ERROR_CHECK(esp_zb_cluster_list_add_identify_cluster(clusterList, identifyClientCluster, ESP_ZB_ZCL_CLUSTER_CLIENT_ROLE));
  ESP_ERROR_CHECK(esp_zb_cluster_list_add_groups_cluster(clusterList, groupsCluster, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE));
  ESP_ERROR_CHECK(esp_zb_cluster_list_add_thermostat_cluster(clusterList, thermostatCluster, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE));
  ESP_ERROR_CHECK(esp_zb_cluster_list_add_thermostat_ui_config_cluster(clusterList, thermostatUiCluster, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE));
  ESP_ERROR_CHECK(esp_zb_cluster_list_add_metering_cluster(clusterList, meteringCluster, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE));
  ESP_ERROR_CHECK(esp_zb_cluster_list_add_electrical_meas_cluster(clusterList, electricalCluster, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE));
  ESP_ERROR_CHECK(esp_zb_cluster_list_add_custom_cluster(clusterList, distantConfigCluster, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE));  

  ESP_ERROR_CHECK(esp_zb_ep_list_add_ep(endPointList, clusterList, endPointConfig));

  ESP_ERROR_CHECK(esp_zb_device_register(endPointList));
}
```

luisyague, 2025-08-13:
Hi @xieqinan , I am testing it again (just updating the localTemperature) with home assistant (with a cc2531 dongle + zha addon) with no success. This the code:
```
void CreateThermostatCluster() {
  esp_zb_thermostat_cluster_cfg_t thermostatClusterConfig = {
      .local_temperature = thmClusterAttributes.localTemp,
      .occupied_cooling_setpoint = ESP_ZB_ZCL_THERMOSTAT_OCCUPIED_COOLING_SETPOINT_DEFAULT_VALUE,
      .occupied_heating_setpoint = thmClusterAttributes.occupiedHeatingSetPoint,
      .control_sequence_of_operation = ESP_ZB_ZCL_THERMOSTAT_CONTROL_SEQ_OF_OPERATION_HEATING_ONLY,
      .system_mode = thmClusterAttributes.systemMode,
  };

  thermostatCluster = esp_zb_thermostat_cluster_create(&thermostatClusterConfig);

  // Add attributes to the Thermostat Cluster
  ESP_ERROR_CHECK(esp_zb_thermostat_cluster_add_attr(thermostatCluster, ESP_ZB_ZCL_ATTR_THERMOSTAT_OCCUPANCY_ID, &thmClusterAttributes.occupancy));
  ESP_ERROR_CHECK(esp_zb_thermostat_cluster_add_attr(thermostatCluster, ESP_ZB_ZCL_ATTR_THERMOSTAT_ABS_MIN_HEAT_SETPOINT_LIMIT_ID, &thmClusterAttributes.absMinHeatSetPointLimit));
  ESP_ERROR_CHECK(esp_zb_thermostat_cluster_add_attr(thermostatCluster, ESP_ZB_ZCL_ATTR_THERMOSTAT_ABS_MAX_HEAT_SETPOINT_LIMIT_ID, &thmClusterAttributes.absMaxSetPointLimit));
  ESP_ERROR_CHECK(esp_zb_thermostat_cluster_add_attr(thermostatCluster, ESP_ZB_ZCL_ATTR_THERMOSTAT_LOCAL_TEMPERATURE_CALIBRATION_ID, &thmClusterAttributes.localTempCalibration));
  ESP_ERROR_CHECK(esp_zb_thermostat_cluster_add_attr(thermostatCluster, ESP_ZB_ZCL_ATTR_THERMOSTAT_MIN_HEAT_SETPOINT_LIMIT_ID, &thmClusterAttributes.minHeatSetPoint));
  ESP_ERROR_CHECK(esp_zb_thermostat_cluster_add_attr(thermostatCluster, ESP_ZB_ZCL_ATTR_THERMOSTAT_MAX_HEAT_SETPOINT_LIMIT_ID, &thmClusterAttributes.maxHeatSetPoint));
  ESP_ERROR_CHECK(esp_zb_thermostat_cluster_add_attr(thermostatCluster, ESP_ZB_ZCL_ATTR_THERMOSTAT_THERMOSTAT_RUNNING_STATE_ID, &thmClusterAttributes.thmRunningState));
  ESP_ERROR_CHECK(esp_zb_thermostat_cluster_add_attr(thermostatCluster, ESP_ZB_ZCL_ATTR_THERMOSTAT_SETPOINT_CHANGE_SOURCE_ID, &thmClusterAttributes.setpointChangeSource));
  ESP_ERROR_CHECK(esp_zb_thermostat_cluster_add_attr(thermostatCluster, ESP_ZB_ZCL_ATTR_THERMOSTAT_SETPOINT_CHANGE_SOURCE_TIMESTAMP_ID, &thmClusterAttributes.setpointChangeSourceTimestamp));
  ESP_ERROR_CHECK(esp_zb_thermostat_cluster_add_attr(thermostatCluster, ESP_ZB_ZCL_ATTR_THERMOSTAT_THERMOSTAT_PROGRAMMING_OPERATION_MODE_ID, &thmClusterAttributes.thermostatProgrammingOperationMode));
  ESP_ERROR_CHECK(esp_zb_thermostat_cluster_add_attr(thermostatCluster, ESP_ZB_ZCL_ATTR_THERMOSTAT_NUMBER_OF_WEEKLY_TRANSITIONS_ID, &thmClusterAttributes.numberOfWeeklyTransitions));
  ESP_ERROR_CHECK(esp_zb_thermostat_cluster_add_attr(thermostatCluster, ESP_ZB_ZCL_ATTR_THERMOSTAT_NUMBER_OF_DAILY_TRANSITIONS_ID, &thmClusterAttributes.numberOfDaylyTransitions));
  ESP_ERROR_CHECK(esp_zb_thermostat_cluster_add_attr(thermostatCluster, ESP_ZB_ZCL_ATTR_THERMOSTAT_START_OF_WEEK_ID, &thmClusterAttributes.startOfWeek));
}

void CreateEndPoint() {
  esp_zb_endpoint_config_t endPointConfig = {
      .endpoint = ENDPOINT_ID,
      .app_profile_id = ESP_ZB_AF_HA_PROFILE_ID,
      .app_device_id = ESP_ZB_HA_THERMOSTAT_DEVICE_ID,
      .app_device_version = 0
  };

  esp_zb_ep_list_t *endPointList = esp_zb_ep_list_create();

  esp_zb_cluster_list_t *clusterList = esp_zb_zcl_cluster_list_create();

  ESP_ERROR_CHECK(esp_zb_cluster_list_add_basic_cluster(clusterList, basicCluster, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE));
  ESP_ERROR_CHECK(esp_zb_cluster_list_add_identify_cluster(clusterList, identifyServerCluster, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE));
  ESP_ERROR_CHECK(esp_zb_cluster_list_add_identify_cluster(clusterList, identifyClientCluster, ESP_ZB_ZCL_CLUSTER_CLIENT_ROLE));
  ESP_ERROR_CHECK(esp_zb_cluster_list_add_groups_cluster(clusterList, groupsCluster, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE));
  ESP_ERROR_CHECK(esp_zb_cluster_list_add_thermostat_cluster(clusterList, thermostatCluster, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE));
  ESP_ERROR_CHECK(esp_zb_cluster_list_add_thermostat_ui_config_cluster(clusterList, thermostatUiCluster, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE));
  ESP_ERROR_CHECK(esp_zb_cluster_list_add_metering_cluster(clusterList, meteringCluster, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE));
  ESP_ERROR_CHECK(esp_zb_cluster_list_add_electrical_meas_cluster(clusterList, electricalCluster, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE));
  ESP_ERROR_CHECK(esp_zb_cluster_list_add_custom_cluster(clusterList, distantConfigCluster, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE));  

  ESP_ERROR_CHECK(esp_zb_ep_list_add_ep(endPointList, clusterList, endPointConfig));

  ESP_ERROR_CHECK(esp_zb_device_register(endPointList));
}

void SensorsEventHandler(uint8_t eventId) {
  switch (eventId) {
    case SENSORS_AMBIENT_TEMP_EVENT:
      esp_zb_lock_acquire(portMAX_DELAY);
      thmClusterAttributes.localTemp = SensorsGetAmbientTemp() * 10;
#ifdef ZIGBEE_DEBUG
      printf("ZIGBEE_DEBUG: Updating ambientTemp to %d\n", thmClusterAttributes.localTemp);
#endif
      esp_zb_zcl_report_attr_cmd_t report_attr_cmd = {0};
      report_attr_cmd.address_mode = ESP_ZB_APS_ADDR_MODE_16_ENDP_PRESENT;
      report_attr_cmd.zcl_basic_cmd.src_endpoint = ENDPOINT_ID;
      report_attr_cmd.zcl_basic_cmd.dst_endpoint = ENDPOINT_ID;
      report_attr_cmd.zcl_basic_cmd.dst_addr_u.addr_short = 0x0000;  // Note that the destination address should be changed
[... cut ...]
      report_attr_cmd.attributeID = ESP_ZB_ZCL_ATTR_THERMOSTAT_LOCAL_TEMPERATURE_ID;
      int ret = esp_zb_zcl_set_attribute_val(ENDPOINT_ID, ESP_ZB_ZCL_CLUSTER_ID_THERMOSTAT, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE, ESP_ZB_ZCL_ATTR_THERMOSTAT_LOCAL_TEMPERATURE_ID, &thmClusterAttributes.localTemp, true);
      printf("Ret %x\n", ret);
      ESP_ERROR_CHECK(esp_zb_zcl_report_attr_cmd_req(&report_attr_cmd));
      esp_zb_lock_release();
      break;
    default:
      break;
  }
}
```

Thanks again for the support.

xieqinan, 2025-08-14:
The code looks fine. However, to help debug the issue, I suggest the following steps:

* Check the `home assistant` endpoint ID to ensure it matches `report_attr_cmd.zcl_basic_cmd.dst_endpoint = ENDPOINT_ID`.
* Read the [attribute report documentation](https://docs.espressif.com/projects/esp-zigbee-sdk/en/latest/esp32/user-guide/zcl_general_report.html) which explains how to report attributes using the esp-zigbee-sdk.
* Refer to the [sniffer documentation](https://docs.espressif.com/projects/esp-zigbee-sdk/en/latest/esp32/developing.html#sniffer-and-wireshark) to set up a sniffer for monitoring frame interactions. If possible, please attach the `.cap` packet here for further analysis.