espressif/esp-zigbee-sdk #670: BDB finding and binding target callback (TZ-1855)

renson-thijs-dhaenens, 2025-06-10:
### Question

Is there a callback to know when the finding and binding procedure is done for the target? 
For the initiator, a callback function is passed trough:
`esp_err_t esp_zb_bdb_finding_binding_start_initiator(uint8_t ep_id, esp_zb_bdb_binding_initiator_callback_t cb);
`

For target we have nu such callback function: 

`esp_err_t esp_zb_bdb_finding_binding_start_target(uint8_t ep_id, uint16_t duration);
`

Is there something like the `esp_zb_app_signal_handler` where we can handle the `esp_zb_bdb_commissioning_status_t`?

xieqinan, 2025-06-11:
> Is there a callback to know when the finding and binding procedure is done for the target?

The F&B target maintains the identification procedure for the specified `duration`. After this time elapses, the `ESP_ZB_BDB_SIGNAL_FINDING_AND_BINDING_TARGET_FINISHED` signal is triggered to notify the application that the F&B procedure has completed. During this period, the F&B initiator can discover and bind to the target's clusters.

If this process does not meet your requirements, please share your specific scenario and explain why you need to handle the `esp_zb_bdb_commissioning_status_t` for the F&B target.

renson-thijs-dhaenens, 2025-06-11:
I am implementing a mechanism join and to bind multiple endpoint via find and bind with 1 button press on the coordinator, 1 button press on the end device. 

State machine coordinator after button press: 

1.  Form and open entwork
2. Start find and bind EP1 after network closed
3. Start timer of x seconds
4. Start find and bind EP2 after timer
5. Stop procedure

State machine end device after button press:

1. Join network
2. Start find and bind EP1 after network closed
3. Start timer of x+5 seconds
4. Start find and bind EP2 after timer
5. Stop procedure

Problems I have: 

End device needs to know when the network is closed but there is no signal `esp_zb_app_signal_t` when network is closed on end device.

xieqinan, 2025-06-13:
Understood — you intend to use the `Permit Join` mechanism (i.e., opening/closing the network) to ensure that all end devices have joined once the network is closed. However, in the `esp-zigbee-sdk`, the `Permit Join` command is generally considered irrelevant for end devices, since an end device is attached to its parent and does not act as a relay for other nodes.

Regarding your application and the older devices based on Silicon Labs: are you saying that the end device can receive the `Permit Join` command from the coordinator to sync moment? If so, could you please double-check the destination network address of the `Permit Join` command sent by the coordinator? If it is `0xFFFC` (broadcast to routers and coordinator), I think the end device should not receive or handle it.

That said, if you expect the end device to receive the `Permit Join` command to sync moment when using `esp-zigbee-sdk`, you can refer to the following patch. It provides an example of how to forward the `Permit Join` command to the user application. Please give  a test.

Hope this helps!

``` c
diff --git a/examples/esp_zigbee_HA_sample/HA_on_off_light/main/esp_zb_light.c b/examples/esp_zigbee_HA_sample/HA_on_off_light/main/esp_zb_light.c
index 2853883..ddb2970 100644
--- a/examples/esp_zigbee_HA_sample/HA_on_off_light/main/esp_zb_light.c
+++ b/examples/esp_zigbee_HA_sample/HA_on_off_light/main/esp_zb_light.c
@@ -125,6 +125,25 @@ static esp_err_t zb_action_handler(esp_zb_core_action_callback_id_t callback_id,
     return ret;
 }
 
+bool zb_apsde_data_indication_handler(esp_zb_apsde_data_ind_t ind)
+{
+    uint8_t duration = 0xFF;
+    uint8_t tsn = 0;
+    if (ind.status == 0 && ind.profile_id == 0x0000 && ind.dst_endpoint == 0 && ind.cluster_id == 0x0036) {
+        if (ind.dst_short_addr >= 0xFFF8 || ind.dst_short_addr == esp_zb_get_short_address()) {
+            assert(ind.asdu_length == 3 && ind.asdu);
+            tsn = *ind.asdu;
+            duration = *(ind.asdu + 1);
+            if (duration == 0) {
+                ESP_LOGW(TAG, "Close network, tsn: %d", tsn);
+            } else {
+                ESP_LOGW(TAG, "Open network, duration: %d, tsn: %d", duration, tsn);
+            }
+        }
+    }
+    return false; // Note that if return true, the stask will skip this APSDE-Indication;
+}
+
 static void esp_zb_task(void *pvParameters)
 {
     /* initialize Zigbee stack */
@@ -137,6 +156,7 @@ static void esp_zb_task(void *pvParameters)
         .model_identifier = ESP_MODEL_IDENTIFIER,
     };
 
+    esp_zb_aps_data_indication_handler_register(zb_apsde_data_indication_handler);
     esp_zcl_utility_add_ep_basic_manufacturer_info(esp_zb_on_off_light_ep, HA_ESP_LIGHT_ENDPOINT, &info);
     esp_zb_device_register(esp_zb_on_off_light_ep);
     esp_zb_core_action_handler_register(zb_action_handler);
diff --git a/examples/esp_zigbee_HA_sample/HA_on_off_switch/main/esp_zb_switch.c b/examples/esp_zigbee_HA_sample/HA_on_off_switch/main/esp_zb_switch.c
index b3d61ab..bbc3c88 100644
--- a/examples/esp_zigbee_HA_sample/HA_on_off_switch/main/esp_zb_switch.c
+++ b/examples/esp_zigbee_HA_sample/HA_on_off_switch/main/esp_zb_switch.c
@@ -1,4 +1,4 @@
-/*
+﻿/*
  * SPDX-FileCopyrightText: 2021-2022 Espressif Systems (Shanghai) CO LTD
  *
  * SPDX-License-Identifier: CC0-1.0
@@ -39,17 +39,17 @@ static const char *TAG = "ESP_ZB_ON_OFF_SWITCH";
 
 static void zb_buttons_handler(switch_func_pair_t *button_func_pair)
 {
-    if (button_func_pair->func == SWITCH_ONOFF_TOGGLE_CONTROL) {
-        /* implemented light switch toggle functionality */
-        esp_zb_zcl_on_off_cmd_t cmd_req;
-        cmd_req.zcl_basic_cmd.src_endpoint = HA_ONOFF_SWITCH_ENDPOINT;
-        cmd_req.address_mode = ESP_ZB_APS_ADDR_MODE_DST_ADDR_ENDP_NOT_PRESENT;
-        cmd_req.on_off_cmd_id = ESP_ZB_ZCL_CMD_ON_OFF_TOGGLE_ID;
-        esp_zb_lock_acquire(portMAX_DELAY);
-        esp_zb_zcl_on_off_cmd_req(&cmd_req);
-        esp_zb_lock_release();
-        ESP_EARLY_LOGI(TAG, "Send 'on_off toggle' command");
-    }
+    static uint8_t duration = 0;
+    esp_zb_zdo_permit_joining_req_param_t cmd_req;
+
+    cmd_req.permit_duration = duration;
+    cmd_req.dst_nwk_addr = 0xFFFF;
+    cmd_req.tc_significance = 1;
+    esp_zb_lock_acquire(portMAX_DELAY);
+    esp_zb_zdo_permit_joining_req(&cmd_req, NULL, NULL);
+    esp_zb_lock_release();
+    ESP_LOGI(TAG, "Broadcast permit_join, duration: %d", cmd_req.permit_duration);
+    duration = duration == 0 ? 180 : 0;
 }

```

renson-thijs-dhaenens, 2025-06-13:
@xieqinan, 

Thank you!

This is working indeed, but now I am thinking if there is an easier method do send a broadcast message from the coordinator when the network is closed. 
I added your example in the signal handler:

`    case ESP_ZB_NWK_SIGNAL_PERMIT_JOIN_STATUS:
        if (err_status == ESP_OK) {
            if (*(uint8_t *)esp_zb_app_signal_get_params(p_sg_p)) {
                ESP_LOGI(TAG, "Network(0x%04hx) is open for %d seconds", esp_zb_get_pan_id(), *(uint8_t *)esp_zb_app_signal_get_params(p_sg_p));
                light_driver_set_color_RGB(0,0,255); // Set LED to blue color
            } else {
                ESP_LOGW(TAG, "Network(0x%04hx) closed, devices joining not allowed.", esp_zb_get_pan_id());
                
                static uint8_t duration = 0;
                esp_zb_zdo_permit_joining_req_param_t cmd_req;
            
                cmd_req.permit_duration = duration;
                cmd_req.dst_nwk_addr = 0xFFFF;
                cmd_req.tc_significance = 1;
                esp_zb_lock_acquire(portMAX_DELAY);
                esp_zb_zdo_permit_joining_req(&cmd_req, NULL, NULL);
                esp_zb_lock_release();
                ESP_LOGI(TAG, "Broadcast permit_join, duration: %d", cmd_req.permit_duration);
                duration = duration == 0 ? 180 : 0;
            }
        }
        break;`

But I just need to send a broadcast, so all the other nodes know that the network is closed.

renson-thijs-dhaenens, 2025-06-16:
@xieqinan, 

Yes indeed, we need the new esp zigbee device to also act on the network closed from the Silicon labs devices. 
I will continue testing with your previous suggestion and wil come back if I have any more issues. 

Thank you very much!

xieqinan, 2025-08-12:
Hi @renson-thijs-dhaenens ，

Do you have any updates on this issue? Please let me know if you encounter any problems.

renson-thijs-dhaenens, 2025-08-12:
@xieqinan, this issue can be closed for now. 
The network closed broadcast works.

Thanks!