espressif/esp-zigbee-sdk #671: HELPPPPPPPPPP

darren0852, 2025-06-14:
### Question

#include <stdio.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "nvs_flash.h"
#include "esp_log.h"
#include "esp_zigbee_core.h"
#define TAG "ZB_COORDINATOR"
#define ESP_ZB_PRIMARY_CHANNEL_MASK 0x07FFF800
void esp_zb_app_signal_handler(esp_zb_app_signal_t *signal_struct)
{
    if (!signal_struct || !signal_struct->p_app_signal) {
        ESP_LOGE(TAG, "Invalid Zigbee signal struct!");
        return;
    }
    esp_zb_app_signal_type_t sig_type = *(signal_struct->p_app_signal);
    esp_err_t status = signal_struct->esp_err_status;

    switch (sig_type) {
        case ESP_ZB_ZDO_SIGNAL_SKIP_STARTUP:
            if (status == ESP_OK) {
                ESP_LOGI(TAG, "Zigbee stack started successfully as Coordinator.");
                esp_zb_bdb_start_top_level_commissioning(
                    ESP_ZB_BDB_MODE_NETWORK_FORMATION | ESP_ZB_BDB_MODE_NETWORK_STEERING);
            } else {
                ESP_LOGE(TAG, "Failed to start Zigbee stack. Status: %d", status);
            }
            break;
        case ESP_ZB_BDB_SIGNAL_DEVICE_FIRST_START:
            ESP_LOGI(TAG, "BDB: First start signal.");
            break;
        case ESP_ZB_BDB_SIGNAL_DEVICE_REBOOT:
            ESP_LOGI(TAG, "BDB: Reboot signal.");
            break;
        case ESP_ZB_BDB_SIGNAL_STEERING:
            ESP_LOGI(TAG, "BDB: Steering finished.");
            break;
        default:
            ESP_LOGW(TAG, "Unhandled Zigbee signal: %d", sig_type);
            break;
    }
}
void zb_zcl_green_power_cluster_handler(uint8_t param)
{
    (void)param;
}
void app_main(void)
{
    ESP_LOGI(TAG, "Zigbee Coordinator Starting...");

    ESP_ERROR_CHECK(nvs_flash_init());

    esp_zb_cfg_t zb_nwk_cfg = {
        .esp_zb_role = ESP_ZB_DEVICE_TYPE_COORDINATOR,
        .install_code_policy = false,
        .nwk_cfg.zczr_cfg = {
            .max_children = 10,
        },
    };
    ESP_LOGI(TAG, "Initializing Zigbee stack...");
    esp_zb_init(&zb_nwk_cfg);
    esp_zb_set_primary_network_channel_set(ESP_ZB_PRIMARY_CHANNEL_MASK);
    ESP_LOGI(TAG, "Starting Zigbee stack...");
    esp_zb_start(true);  
    esp_zb_stack_main_loop();
}

SPIWP:0xee
mode:DIO, clock div:2
load:0x40875720,len:0x16b8
load:0x4086c110,len:0xe84
load:0x4086e610,len:0x3054
entry 0x4086c11a
I (26) boot: ESP-IDF v5.4.1 2nd stage bootloader
I (27) boot: compile time Jun 14 2025 16:37:36
I (28) boot: chip revision: v0.2
I (28) boot: efuse block revision: v0.3
I (30) boot.esp32c6: SPI Speed      : 80MHz
I (34) boot.esp32c6: SPI Mode       : DIO
I (38) boot.esp32c6: SPI Flash Size : 8MB
I (42) boot: Enabling RNG early entropy source...
I (46) boot: Partition Table:
I (49) boot: ## Label            Usage          Type ST Offset   Length
I (55) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (62) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (68) boot:  2 zb_storage       Unknown data     01 81 00010000 00006000
I (75) boot:  3 zb_fct           Unknown data     01 81 00016000 00001000
I (81) boot:  4 factory          factory app      00 00 00020000 00200000
I (88) boot: End of partition table
I (91) esp_image: segment 0: paddr=00020020 vaddr=420c0020 size=46a5ch (289372) map
I (204) esp_image: segment 1: paddr=00066a84 vaddr=40800000 size=09594h ( 38292) load
I (221) esp_image: segment 2: paddr=00070020 vaddr=42000020 size=b626ch (746092) map
I (493) esp_image: segment 3: paddr=00126294 vaddr=40809594 size=03534h ( 13620) load
I (500) esp_image: segment 4: paddr=001297d0 vaddr=4080cad0 size=019b8h (  6584) load
I (510) boot: Loaded app from partition at offset 0x20000
I (511) boot: Disabling RNG early entropy source...
I (521) cpu_start: Unicore app
I (530) cpu_start: Pro cpu start user code
I (530) cpu_start: cpu freq: 160000000 Hz
I (530) app_init: Application information:
I (530) app_init: Project name:     TEST126
I (534) app_init: App version:      1
I (537) app_init: Compile time:     Jun 14 2025 16:37:26
I (542) app_init: ELF file SHA256:  310790a9e...
I (546) app_init: ESP-IDF:          v5.4.1
I (550) efuse_init: Min chip rev:     v0.0
I (554) efuse_init: Max chip rev:     v0.99 
I (558) efuse_init: Chip rev:         v0.2
I (562) heap_init: Initializing. RAM available for dynamic allocation:
I (568) heap_init: At 40814030 len 000685E0 (417 KiB): RAM
I (573) heap_init: At 4087C610 len 00002F54 (11 KiB): RAM
I (579) heap_init: At 50000000 len 00003FE8 (15 KiB): RTCRAM
I (585) spi_flash: detected chip: generic
I (588) spi_flash: flash io: dio
I (591) sleep_gpio: Configure to isolate all GPIO pins in sleep state
I (597) sleep_gpio: Enable automatic switching of GPIO sleep configuration
I (603) coexist: coex firmware version: e727207
I (634) coexist: coexist rom version 5b8dcfa
I (635) main_task: Started on CPU0
I (635) main_task: Calling app_main()
I (635) ZB_COORDINATOR: Zigbee Coordinator Starting...
I (644) ZB_COORDINATOR: Initializing Zigbee stack...
☺���������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������ESP-ROM:esp32c6-20220919
Build:Sep 19 2022
rst:0xc (SW_CPU),boot:0xc (SPI_FAST_FLASH_BOOT)
Saved PC:0x4001975a
--- 0x4001975a: software_reset_cpu in ROM

### Additional context.

This is my code and the output. Why would like that?
I'm doing school project try to create a zigbee network.
Is that any mistake in my coding and what should I do.

xieqinan, 2025-06-16:
@darren0852 ,

Why not refer to the[on_off_switch](https://github.com/espressif/esp-zigbee-sdk/tree/main/examples/esp_zigbee_HA_sample/HA_on_off_switch) example to create a Zigbee Coordinator and form a Zigbee network?

xieqinan, 2025-08-12:
Hi @darren0852,

Do you have test the [on_off_switch](https://github.com/espressif/esp-zigbee-sdk/tree/main/examples/esp_zigbee_HA_sample/HA_on_off_switch) example? Please let me konw the problem confused for you.

xieqinan, 2025-09-30:
Hi @darren0852,

I believe you are confused about how to create a Zigbee coordinator in esp-zigbee-sdk. You can refer to the [on_off_switch](https://github.com/espressif/esp-zigbee-sdk/tree/main/examples/esp-zigbee_HA_sample/HA_on_off_switch) example for guidance.

Since there has been no response from you for a long time, I will close this issue.