espressif/esp-zigbee-sdk #673: Zigbee Motion sensors not working with Gateway example code

mundevx, 2025-06-16:
### Question

I am using espressif zigbee sdk gateway example code. I have IAS Zone Motion sensor that works ok with tuya and home assistant. But with gateway example code, IAS Zone motion sensor is added successfully, binding ok, but not getting any notify or data callback in my gateway code.  I also used     esp_zb_core_action_handler_register &  esp_zb_raw_command_handler_register functions.  what to do?

    /* initialize Zigbee stack */
    esp_zb_cfg_t zb_nwk_cfg = ESP_ZB_ZC_CONFIG();
    esp_zb_init(&zb_nwk_cfg);

    esp_zb_set_primary_network_channel_set(ESP_ZB_PRIMARY_CHANNEL_MASK);

    esp_zb_ep_list_t *ep_list = esp_zb_ep_list_create();
    esp_zb_cluster_list_t *cluster_list = esp_zb_zcl_cluster_list_create();
    esp_zb_endpoint_config_t endpoint_config = {
        .endpoint = ESP_ZB_GATEWAY_ENDPOINT,
        .app_profile_id = ESP_ZB_AF_HA_PROFILE_ID,
        .app_device_id = ESP_ZB_HA_REMOTE_CONTROL_DEVICE_ID,
        .app_device_version = 0,
    };

    esp_zb_attribute_list_t *basic_cluser = esp_zb_basic_cluster_create(NULL);
    esp_zb_basic_cluster_add_attr(basic_cluser, ESP_ZB_ZCL_ATTR_BASIC_MANUFACTURER_NAME_ID, ESP_MANUFACTURER_NAME);
    esp_zb_basic_cluster_add_attr(basic_cluser, ESP_ZB_ZCL_ATTR_BASIC_MODEL_IDENTIFIER_ID, ESP_MODEL_IDENTIFIER);
    esp_zb_cluster_list_add_basic_cluster(cluster_list, basic_cluser, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE);
    esp_zb_cluster_list_add_identify_cluster(cluster_list, esp_zb_identify_cluster_create(NULL), ESP_ZB_ZCL_CLUSTER_SERVER_ROLE);

    esp_zb_ep_list_add_gateway_ep(ep_list, cluster_list, endpoint_config);
    esp_zb_device_register(ep_list);
    esp_zb_core_action_handler_register(zb_action_handler);
	// Register the handler
	//esp_zb_raw_command_handler_register(my_raw_command_handler);
    ESP_ERROR_CHECK(esp_zb_start(false));
    esp_zb_stack_main_loop();
    esp_rcp_update_deinit();

    vTaskDelete(NULL);

xieqinan, 2025-06-17:
Does the Zigbee device created using the gateway example work on the same network as TuYa or HA? Please note that the gateway acts as the Zigbee coordinator with a Wi-Fi interface.

xieqinan, 2025-06-18:
Please start a sniffer to capture the frames and confirm whether the destination addresses of the frame sent by the IAS zone sensors and temperature-humidity Zigbee devices match the Gateway's address.

By the way, could you also share the sniffer capture and the Gateway's output logs? This will help with debugging.

mundevx, 2025-08-12:
Yes, close this issue.

closed 2025-09-30 by xieqinan