espressif/esp-zigbee-sdk #674: Reading RMS From Zigbee-Powered CT Not Working Correctly

pdEmAk, 2025-06-17:
Hi, I am trying to get current readings from a CT clamp with in-built zigbee. At the moment, I am unable to correctly read RMS data from the CT. I am not sure if this has something to do with the CT and my ESP32 C6 not correctly communicating what endpoint to transfer on. Cc @xieqinan 

My logs and code are below.

```
I (35746) ZB_COORD: Received Zigbee signal: 0x30, status: ESP_OK
I (35746) ZB_COORD: Device update signal received
I (35756) ZB_COORD: Received Zigbee signal: 0x2F, status: ESP_OK
I (35756) ZB_COORD: Device announcement signal 0x2F received
I (35766) ZB_COORD: New device joined: 0x9a25
I (35766) ZB_COORD: Device 0x9a25 already known, skipping
I (41496) ZB_COORD: Periodic read from device 0: 0x9a25
E (41496) ZB_COORD: Failed to send read attribute command: ERROR (0x1)
E (41496) ZB_COORD: Request details: device=0x9a25, dst_ep=1, src_ep=1, cluster=0x0b04, attr=0x0508
I (43306) ZB_COORD: Received Zigbee signal: 0x12, status: ESP_OK
I (43306) ZB_COORD: Unhandled signal: 0x12, status: ESP_OK
ZB_TRACE_LOG[0]: common/zb_address.c:1138    zb_address_delete ref 57
I (43316) ZB_COORD: Received Zigbee signal: 0x30, status: ESP_OK
I (43316) ZB_COORD: Device update signal received
I (43316) ZB_COORD: Received Zigbee signal: 0x2F, status: ESP_OK
I (43326) ZB_COORD: Device announcement signal 0x2F received
I (43326) ZB_COORD: New device joined: 0x9a25
I (43336) ZB_COORD: Device 0x9a25 already known, skipping
I (47876) ZB_COORD: Received Zigbee signal: 0x32, status: ESP_OK
I (47876) ZB_COORD: Unhandled signal: 0x32, status: ESP_OK
I (50866) ZB_COORD: Received Zigbee signal: 0x12, status: ESP_OK
I (50866) ZB_COORD: Unhandled signal: 0x12, status: ESP_OK
ZB_TRACE_LOG[0]: common/zb_address.c:1138    zb_address_delete ref 57
I (50876) ZB_COORD: Received Zigbee signal: 0x30, status: ESP_OK
I (50876) ZB_COORD: Device update signal received
I (50886) ZB_COORD: Received Zigbee signal: 0x2F, status: ESP_OK
I (50886) ZB_COORD: Device announcement signal 0x2F received
I (50896) ZB_COORD: New device joined: 0x9a25
I (50896) ZB_COORD: Device 0x9a25 already known, skipping
I (52496) ZB_COORD: Periodic read from device 0: 0x9a25
E (52496) ZB_COORD: Failed to send read attribute command: ERROR (0x2)
E (52496) ZB_COORD: Request details: device=0x9a25, dst_ep=1, src_ep=1, cluster=0x0b04, attr=0x0508
I (58436) ZB_COORD: Received Zigbee signal: 0x12, status: ESP_OK
I (58436) ZB_COORD: Unhandled signal: 0x12, status: ESP_OK
ZB_TRACE_LOG[0]: common/zb_address.c:1138    zb_address_delete ref 57
I (58446) ZB_COORD: Received Zigbee signal: 0x30, status: ESP_OK
I (58446) ZB_COORD: Device update signal received
I (58446) ZB_COORD: Received Zigbee signal: 0x2F, status: ESP_OK
I (58456) ZB_COORD: Device announcement signal 0x2F received
I (58456) ZB_COORD: New device joined: 0x9a25
I (58466) ZB_COORD: Device 0x9a25 already known, skipping
```

```
static void read_ct_clamp_data(uint16_t device_addr, uint8_t endpoint)
{
    // Create the attribute ID array
    static uint16_t attr_id = ESP_ZB_ZCL_ATTR_ELECTRICAL_MEASUREMENT_RMS_CURRENT_ID;

    // Create the read attribute command structure
    esp_zb_zcl_read_attr_cmd_t read_req;
    read_req.address_mode = ESP_ZB_APS_ADDR_MODE_16_ENDP_PRESENT;
    read_req.zcl_basic_cmd.dst_addr_u.addr_short = device_addr;
    read_req.zcl_basic_cmd.dst_endpoint = endpoint;
    read_req.zcl_basic_cmd.src_endpoint = HA_ESP_COORDINATOR_ENDPOINT;
    read_req.clusterID = ESP_ZB_ZCL_CLUSTER_ID_ELECTRICAL_MEASUREMENT;
    read_req.manuf_specific = 0;
    read_req.direction = ESP_ZB_ZCL_CMD_DIRECTION_TO_SRV;
    read_req.dis_defalut_resp = 0;
    read_req.manuf_code = 0;
    read_req.attr_number = 1;
    read_req.attr_field = &attr_id;

static void read_ct_clamp_data(uint16_t device_addr, uint8_t endpoint)
{
    // Create the attribute ID array
    static uint16_t attr_id = ESP_ZB_ZCL_ATTR_ELECTRICAL_MEASUREMENT_RMS_CURRENT_ID;

    // Create the read attribute command structure
    esp_zb_zcl_read_attr_cmd_t read_req;
    read_req.address_mode = ESP_ZB_APS_ADDR_MODE_16_ENDP_PRESENT;
    read_req.zcl_basic_cmd.dst_addr_u.addr_short = device_addr;
    read_req.zcl_basic_cmd.dst_endpoint = endpoint;
    read_req.zcl_basic_cmd.src_endpoint = HA_ESP_COORDINATOR_ENDPOINT;
    read_req.clusterID = ESP_ZB_ZCL_CLUSTER_ID_ELECTRICAL_MEASUREMENT;
    read_req.manuf_specific = 0;
    read_req.direction = ESP_ZB_ZCL_CMD_DIRECTION_TO_SRV;
    read_req.dis_defalut_resp = 0;
    read_req.manuf_code = 0;
    read_req.attr_number = 1;
    read_req.attr_field = &attr_id;

    // Send the read attribute command
    esp_err_t err = esp_zb_zcl_read_attr_cmd_req(&read_req);

    if (err != ESP_OK)
    {
        ESP_LOGE(TAG, "Failed to send read attribute command: %s (0x%x)", esp_err_to_name(err), err);
        ESP_LOGE(TAG, "Request details: device=0x%04x, dst_ep=%d, src_ep=%d, cluster=0x%04x, attr=0x%04x",
                 device_addr, endpoint, HA_ESP_COORDINATOR_ENDPOINT,
                 ESP_ZB_ZCL_CLUSTER_ID_ELECTRICAL_MEASUREMENT, attr_id);
    }
    else
    {
        ESP_LOGI(TAG, "Reading RMS current from device 0x%04x, endpoint %d", device_addr, endpoint);
    }
}
```

```
static void handle_zcl_attr_response(esp_zb_zcl_cmd_read_attr_resp_message_t *message)
{
    ESP_LOGI(TAG, "Received ZCL attribute response from 0x%04x",
             message->info.src_address.u.short_addr);

    // Process the attribute response
    if (message->info.status == ESP_ZB_ZCL_STATUS_SUCCESS)
    {
        ESP_LOGI(TAG, "Successfully read attribute from device");

        // Parse the attribute data
        esp_zb_zcl_read_attr_resp_variable_t *variable = message->variables;
        while (variable)
        {
            ESP_LOGI(TAG, "Attribute ID: 0x%04x, Status: 0x%02x",
                     variable->attribute.id, variable->status);

            if (variable->status == ESP_ZB_ZCL_STATUS_SUCCESS)
            {
                // Check if this is a manufacturer name attribute from Basic cluster
                if (message->info.cluster == ESP_ZB_ZCL_CLUSTER_ID_BASIC &&
                    variable->attribute.id == ESP_ZB_ZCL_ATTR_BASIC_MANUFACTURER_NAME_ID)
                {
                    char manufacturer[32] = {0};
                    memcpy(manufacturer, variable->attribute.data.value,
                           variable->attribute.data.size);
                    ESP_LOGI(TAG, "Device 0x%04hx manufacturer: %s",
                             message->info.src_address.u.short_addr, manufacturer);

                    // Check if it's your CT clamp (replace with actual manufacturer name)
                    if (strstr(manufacturer, "Tuya") != NULL)
                    {
                        ESP_LOGI(TAG, "Found CT clamp device!");
                        // Now try reading current data
                        read_ct_clamp_data(message->info.src_address.u.short_addr, message->info.src_endpoint);
                        // read_basic_info(message->info.src_address.u.short_addr, message->info.src_endpoint);
                    }
                }
                else
                {
                    // Handle other attribute types as before
                    switch (variable->attribute.data.type)
                    {
                    case ESP_ZB_ZCL_ATTR_TYPE_U16:
                        ESP_LOGI(TAG, "Attribute value (U16): %u", *(uint16_t *)(variable->attribute.data.value));
                        break;
                    case ESP_ZB_ZCL_ATTR_TYPE_U32:
                        ESP_LOGI(TAG, "Attribute value (U32): %lu", *(uint32_t *)(variable->attribute.data.value));
                        break;
                    case ESP_ZB_ZCL_ATTR_TYPE_CHAR_STRING:
                        ESP_LOGI(TAG, "Attribute value (String): %.*s",
                                 variable->attribute.data.size,
                                 (char *)variable->attribute.data.value);
                        break;
                    default:
                        ESP_LOGI(TAG, "Attribute type: 0x%02x",
                                 variable->attribute.data.type);
                        break;
                    }
                }
            }
            variable = variable->next;
        }
    }
    else
    {
        ESP_LOGW(TAG, "Failed to read attribute, status: 0x%02x", message->info.status);
    }
}
```

xieqinan, 2025-06-18:
I reviewed the `read_ct_clamp_data()` function, and it appears to be implemented correctly. However, please note that the return value of `esp_zb_zcl_read_attr_cmd_req()` is the **TSN** (Transaction Sequence Number), please refer to this [comment in the SDK](https://github.com/espressif/esp-zigbee-sdk/blob/main/components/esp-zigbee-lib/include/zcl/esp_zigbee_zcl_command.h#L2431) for clarification.

Additionally, I noticed that the log `New device joined: 0x9a25` and `common/zb_address.c:1138    zb_address_delete ref 57` are printed multiple times. Could you confirm whether the **CT clamp** device is a **sleepy device** ? If it is not, the repeated log entries might suggest that the CT clamp hasn't actually completed the join process with the Zigbee coordinator.

Please double-check the CT clamp’s documentation to confirm its commissioning method and ensure it is properly joining and staying connected to the Zigbee network.

pdEmAk, 2025-06-18:
> I reviewed the `read_ct_clamp_data()` function, and it appears to be implemented correctly. However, please note that the return value of `esp_zb_zcl_read_attr_cmd_req()` is the **TSN** (Transaction [...]

Hi,

I am not sure if it is a sleepy device. It's a Tuya CT clamp and so far, I have been unable to find any documentation that clearly states how to handle this. I have left a decent time window to allow enough time for the CT to join the network (180s), that doesn't seem to do anything. 

If the read function is correct, what do you reckon the cause of the error is? I tried to implement a scan devices endpoint with the mindset that the device is communicating on a different endpoint but no luck so far.

```
static void scan_device_endpoints(uint8_t param)
{
    uint16_t device_addr = g_pending_scan_device_addr;
    // Using the global variable for device address
    if (device_addr == 0)
    {
        ESP_LOGE(TAG, "Invalid device address for endpoint scan");
        return;
    }

    // Check if device is in our list and update retry count
    int device_idx = find_device_index(device_addr);
    if (device_idx >= 0) {
        ESP_LOGI(TAG, "Scanning endpoints for device 0x%04x (attempt %d/%d)", 
                device_addr, connected_devices[device_idx].retry_count + 1, MAX_RETRY_ATTEMPTS);
    } else {
        ESP_LOGI(TAG, "Scanning endpoints for device 0x%04x", device_addr);
    }

    esp_zb_zdo_active_ep_req_param_t req_param = {
        .addr_of_interest = device_addr};

    // Allocate memory for the device address to pass as user context
    uint16_t *addr_ptr = (uint16_t *)malloc(sizeof(uint16_t));
    if (!addr_ptr) {
        ESP_LOGE(TAG, "Failed to allocate memory for endpoint scan context");
        return;
    }
    *addr_ptr = device_addr;
    
    esp_zb_zdo_active_ep_req(&req_param, active_ep_callback, addr_ptr);
    
    ESP_LOGI(TAG, "Active endpoint request sent to device 0x%04x", device_addr);
}
```

xieqinan, 2025-06-19:
Please ensure that the `CT clamp` device is functioning properly. I suspect that it may be a **legacy device**. To confirm this, please provide a Wireshark packet capture.

If it is indeed a legacy device, you can add the following API to ensure compatibility:

```c
esp_zb_secur_link_key_exchange_required_set(false);
```
By the way, could you please also submit an issue to TuYa to identify the behavior of `CT clamp`? Otherwise, it is hard to detect the root of this issue.

pdEmAk, 2025-06-23:
> Please ensure that the `CT clamp` device is functioning properly. I suspect that it may be a **legacy device**. To confirm this, please provide a Wireshark packet capture.
> 
> If it is indeed a legacy device, you can add the following API to ensure compatibility:
> 
> esp_zb_secur_link_key_exchange_required_set(false);
> By the way, could you please also submit an issue to TuYa to identify the behavior of `CT clamp`? Otherwise, it is hard to detect the root of this issue.

Hi @xieqinan, I still cannot get a correct endpoint discovery for the device. My best bet is to capture network traffic with wireshark as you have rightly pointed out, however, I am not sure how to correctly set up wireshark to sniff zigbee traffic on windows. I have the software installed but I am not sure what "adapter/dongle" to use. What would you recommend? 

The documentation here: https://docs.espressif.com/projects/esp-techpedia/en/latest/esp-friends/tool-recommendation/wireshark-windows.html mentions using `wlanhelper` but the npcap documentation recommends otherwise so I imagine the espressif documentation doesn't cater to the new wireshark version.

xieqinan, 2025-06-24:
> What would you recommend?

Please refer to https://docs.espressif.com/projects/esp-zigbee-sdk/en/latest/esp32/developing.html#sniffer-and-wireshark , it may be helpful.

pdEmAk, 2025-08-14:
> Hi [@pdEmAk](https://github.com/pdEmAk) ,
> 
> Do you have any updates? I believe if you follow the [#674 (comment)](https://github.com/espressif/esp-zigbee-sdk/issues/674#issuecomment-2998596007) to install a sniffer, it can help identify the CT clamp behavior. Please let me know your problem.

hi. I managed to get it working. thank you so much.

xieqinan, 2025-08-15:
> hi. I managed to get it working. thank you so much.

Glad to hear that. Please try analyzing the frame interaction between the Zigbee-Power CT and the Zigbee device using a sniffer to determine the cause of the failure. If that’s difficult, you can attach the `.cap` packet here.

closed 2026-01-20 by xieqinan