espressif/esp-zigbee-sdk #675: How Zigbee gateway get endpoint details

darren0852, 2025-06-19:
### Question

Hi guys,
This is my code of Zigbee gateway and my question is how I get all endpoint details like
1. How many endpoint device in this Zigbee network
2. What each endpoint details

`#include <stdio.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "nvs_flash.h"
#include "esp_log.h"
#include "esp_zigbee_core.h"
#include "esp_zigbee_endpoint.h"
#include "esp_zigbee_attribute.h"
#include "zdo/esp_zigbee_zdo_command.h"

#define TAG "ZB_COORDINATOR"
#define ESP_ZB_PRIMARY_CHANNEL_MASK 0x07FFF800
#define MY_ENDPOINT 1
void zb_zcl_green_power_cluster_handler(uint8_t param)
{
    (void)param;
}
static void bdb_start_top_level_commissioning_cb(uint8_t mode_mask)
{
    if (esp_zb_bdb_start_top_level_commissioning(mode_mask) != ESP_OK) {
    ESP_LOGE(TAG, "Failed to start Zigbee bdb commissioning");
    return;
}

}

void esp_zb_app_signal_handler(esp_zb_app_signal_t *signal_struct)
{
    if (!signal_struct || !signal_struct->p_app_signal) {
        ESP_LOGE(TAG, "Invalid Zigbee signal struct!");
        return;
    }

    uint32_t *p_sg_p       = signal_struct->p_app_signal;
    esp_err_t err_status = signal_struct->esp_err_status;
    esp_zb_app_signal_type_t sig_type = *p_sg_p;
    esp_zb_zdo_signal_device_annce_params_t *dev_annce_params = NULL;

    switch (sig_type) {
        case ESP_ZB_ZDO_SIGNAL_SKIP_STARTUP:
        if (err_status == ESP_OK) {
            ESP_LOGI(TAG, "Zigbee stack started. Forming network...");

            esp_zb_bdb_start_top_level_commissioning(ESP_ZB_BDB_MODE_NETWORK_FORMATION);
        } else {
            ESP_LOGE(TAG, "Zigbee stack failed to start: %d", err_status);
        }
        break;

        case ESP_ZB_BDB_SIGNAL_FORMATION:
            if (err_status == ESP_OK) {
                ESP_LOGI(TAG, "✅ Zigbee network formed. Now allowing devices to join...");
                esp_zb_ieee_addr_t ieee_address;
            esp_zb_get_long_address(ieee_address);
            ESP_LOGI(TAG, "Formed network successfully (Extended PAN ID: %02x:%02x:%02x:%02x:%02x:%02x:%02x:%02x, PAN ID: 0x%04hx, Channel:%d, Short Address: 0x%04hx)",
                     ieee_address[7], ieee_address[6], ieee_address[5], ieee_address[4],
                     ieee_address[3], ieee_address[2], ieee_address[1], ieee_address[0],
                     esp_zb_get_pan_id(), esp_zb_get_current_channel(), esp_zb_get_short_address());
                esp_zb_bdb_start_top_level_commissioning(ESP_ZB_BDB_MODE_NETWORK_STEERING);
            } else {
                ESP_LOGE(TAG, "❌ Failed to form network: %d", err_status);
            }
            break;

        case ESP_ZB_BDB_SIGNAL_STEERING:
            if (err_status == ESP_OK) {
                ESP_LOGI(TAG, "✅ Device join permitted (steering active).");
            } else {
                ESP_LOGE(TAG, "❌ Failed to enable steering: %d", err_status);
            }
            break;

        case ESP_ZB_ZDO_SIGNAL_DEVICE_ANNCE:
                dev_annce_params = (esp_zb_zdo_signal_device_annce_params_t *)esp_zb_app_signal_get_params(p_sg_p);
                ESP_LOGI(TAG, "New device commissioned or rejoined (short: 0x%04hx)", dev_annce_params->device_short_addr);
            break;
        case ESP_ZB_BDB_SIGNAL_DEVICE_FIRST_START:
        case ESP_ZB_BDB_SIGNAL_DEVICE_REBOOT:
        if (err_status == ESP_OK) {
            ESP_LOGI(TAG, "Device started up in%s factory-reset mode", esp_zb_bdb_is_factory_new() ? "" : " non");
            if (esp_zb_bdb_is_factory_new()) {
                ESP_LOGI(TAG, "Start network formation");
                esp_zb_bdb_start_top_level_commissioning(ESP_ZB_BDB_MODE_NETWORK_FORMATION);
            } else {
                esp_zb_bdb_open_network(500);
                ESP_LOGI(TAG, "Device rebooted");
            }
        } else {
            ESP_LOGW(TAG, "%s failed with status: %s, retrying", esp_zb_zdo_signal_to_string(sig_type),
                     esp_err_to_name(err_status));
            esp_zb_scheduler_alarm((esp_zb_callback_t)bdb_start_top_level_commissioning_cb,
                                   ESP_ZB_BDB_MODE_INITIALIZATION, 1000);
        }
        break;
        case ESP_ZB_NWK_SIGNAL_PERMIT_JOIN_STATUS:
            if (err_status == ESP_OK) {
                uint8_t permit_duration = *(uint8_t *)esp_zb_app_signal_get_params(p_sg_p);

                if (permit_duration) {
                    esp_zb_ieee_addr_t ieee_address;
                    esp_zb_get_long_address(ieee_address);

                    ESP_LOGI(TAG, "🔓 Network is OPEN for %d seconds", permit_duration);
                    ESP_LOGI(TAG, "📡 Channel: %d", esp_zb_get_current_channel());
                    ESP_LOGI(TAG, "🆔 PAN ID: 0x%04hx", esp_zb_get_pan_id());
                    ESP_LOGI(TAG, "🆔 Extended PAN ID: %02x:%02x:%02x:%02x:%02x:%02x:%02x:%02x",
                            ieee_address[7], ieee_address[6], ieee_address[5], ieee_address[4],
                            ieee_address[3], ieee_address[2], ieee_address[1], ieee_address[0]);
                } else {
                    ESP_LOGW(TAG, "🚫 Network is CLOSED, devices joining not allowed.");
                }
            }
            break;

        default:
            ESP_LOGI(TAG, "ZDO signal: %s (0x%x), status: %s", esp_zb_zdo_signal_to_string(sig_type), sig_type,
            esp_err_to_name(err_status));
            break;
    }
}
void app_main(void)
{
    ESP_LOGI(TAG, "Zigbee Coordinator Starting...");

    ESP_ERROR_CHECK(nvs_flash_init());

    esp_zb_cfg_t zb_nwk_cfg = {
        .esp_zb_role = ESP_ZB_DEVICE_TYPE_COORDINATOR,
        .install_code_policy = false,
        .nwk_cfg.zczr_cfg = {
            .max_children = 10,
        },
    };

    esp_zb_set_primary_network_channel_set(ESP_ZB_PRIMARY_CHANNEL_MASK);

    ESP_LOGI(TAG, "Starting Zigbee stack...");
    esp_zb_start(true);

    esp_zb_stack_main_loop();
}
`

xieqinan, 2025-06-20:
> How many endpoint device in this Zigbee network

The `esp_zb_zdo_active_ep_req()` can be used to obtain all endpoints of a device.

> What each endpoint details

The `esp_zb_zdo_simple_desc_req()` can be used to obtain the detailed information of endpoint.

xieqinan, 2025-08-12:
Hi @darren0852 ,

Do you have any updates? If the https://github.com/espressif/esp-zigbee-sdk/issues/675#issuecomment-2989637626 is helpful, please consider closing this issue.

closed 2025-09-30 by xieqinan