espressif/esp-zigbee-sdk #676: How to communicate with an end device without security? (V2)

amp043, 2025-06-19:
### Question

This issue is a follow-up to #404 

We have a coordinator that needs to communicate with a sleepy end device that is running in non-secure mode. The end device is compatible with Zigbee 2006, with no security. We have successfully integrated with this device using other Zigbee stacks, and are currently trying to migrate to ESP32 for Zigbee.

We are using the project HA_on_off_switch as a baseline. The end device will successfully join the coordinator, but when registering a raw command callback with `esp_zb_raw_command_handler_register()`, that callback never fires. We have tried using `esp_zb_secur_network_security_enable(false)` with no change to the result.

Our versions are:
IDF version: v5.5-dev-4312-gb6ea70c57b
esp-zigbee-lib version: 1.6.5
esp-zboss-lib version: 1.6.4
SoC: ESP32H2

The console output is here:
[console-log.txt](https://github.com/user-attachments/files/20824209/console-log.txt)

Sniffer capture for the end device joining and sending data packets:
[nonsecure_end_device.zip](https://github.com/user-attachments/files/20824268/nonsecure_end_device.zip)

Data packets (the ones we need to receive) are at packet numbers [22, 26, 29, 33, 35, 39, 43]. Note that these use a proprietary profile (0xC0FB) so Wireshark is unable to decode them.

The modified source file esp_zb_switch.c is pasted in the next section.

### Additional context.

```
/*
 * SPDX-FileCopyrightText: 2021-2022 Espressif Systems (Shanghai) CO LTD
 *
 * SPDX-License-Identifier: CC0-1.0
 *
 * Zigbee HA_on_off_switch Example
 *
 * This example code is in the Public Domain (or CC0 licensed, at your option.)
 *
 * Unless required by applicable law or agreed to in writing, this
 * software is distributed on an "AS IS" BASIS, WITHOUT WARRANTIES OR
 * CONDITIONS OF ANY KIND, either express or implied.
 */

#include "esp_check.h"
#include "esp_err.h"
#include "string.h"
#include "nvs_flash.h"
#include "esp_log.h"
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "ha/esp_zigbee_ha_standard.h"
#include "esp_zb_switch.h"

#include "zboss_api.h"

#if defined ZB_ED_ROLE
#error Define ZB_COORDINATOR_ROLE in idf.py menuconfig to compile light switch source code.
#endif
typedef struct light_bulb_device_params_s {
    esp_zb_ieee_addr_t ieee_addr;
    uint8_t  endpoint;
    uint16_t short_addr;
} light_bulb_device_params_t;

static switch_func_pair_t button_func_pair[] = {
    {GPIO_INPUT_IO_TOGGLE_SWITCH, SWITCH_ONOFF_TOGGLE_CONTROL}
};

static const char *TAG = "ESP_ZB_ON_OFF_SWITCH";

static bool rawCmdHandlerCb(uint8_t bufid) {
    uint8_t buf[zb_buf_len(bufid)];
[... 159 lines cut ...]
        .manufacturer_name = ESP_MANUFACTURER_NAME,
        .model_identifier = ESP_MODEL_IDENTIFIER,
    };

    esp_zcl_utility_add_ep_basic_manufacturer_info(esp_zb_on_off_switch_ep, HA_ONOFF_SWITCH_ENDPOINT, &info);
    esp_zb_device_register(esp_zb_on_off_switch_ep);
    esp_zb_raw_command_handler_register(rawCmdHandlerCb);
    esp_zb_set_primary_network_channel_set(ESP_ZB_PRIMARY_CHANNEL_MASK);
    ESP_ERROR_CHECK(esp_zb_start(false));
    ESP_ERROR_CHECK(esp_zb_secur_network_security_enable(false));
    esp_zb_stack_main_loop();
}

void app_main(void)
{
    esp_zb_platform_config_t config = {
        .radio_config = ESP_ZB_DEFAULT_RADIO_CONFIG(),
        .host_config = ESP_ZB_DEFAULT_HOST_CONFIG(),
    };
    ESP_ERROR_CHECK(nvs_flash_init());
    ESP_ERROR_CHECK(esp_zb_platform_config(&config));

    xTaskCreate(esp_zb_task, "Zigbee_main", 4096, NULL, 5, NULL);
}
```

xieqinan, 2025-06-20:
Hi,

The proprietary profile ID `0xC0FB` will not be handled by the Zigbee stack, as there is no official specification defining its behavior. Therefore, such frames will be dropped at the **AF layer** of the stack.

To handle these frames, you can register an APSDE-indication callback using `esp_zb_aps_data_indication_handler_register()`. This allows you to intercept the frames manually and process them as needed.

Here is an example of how to use it. For more details, refer to the [APS documentation](https://docs.espressif.com/projects/esp-zigbee-sdk/en/latest/esp32/user-guide/aps.html):

```c
static bool zb_apsde_data_indication_handler(esp_zb_apsde_data_ind_t ind)
{
    /* Filter out the frame with 0xC0FB profile ID */
    if (ind.profile_id == 0xC0FB) {
        // Handle the proprietary frame here
        return true;  // Indicates the frame has been handled
    } else {
        return false; // Let the stack handle other profiles
    }
}

void app_main(void)
{
    // Register the APSDE indication handler
    esp_zb_aps_data_indication_handler_register(zb_apsde_data_indication_handler);
}
```

> We have successfully integrated with this device using other Zigbee stacks, and are currently trying to migrate to ESP32 for Zigbee.

I believe other Zigbee stacks also handle frames with proprietary profiles. Could you please share how to integrate this functionality?

amp043, 2025-06-20:
@xieqinan thank you, `esp_zb_aps_data_indication_handler_register()` is working, and now we are able to receive from and send to the non-secure end device.

I have 2 follow-up questions that are related to sleepy end devices in general, (both secure and non-secure).

1. It appears that the MAC polling expiry timeout is set at the default of 7.68ms. We integrate with some sleepy end devices that poll every 10-20 seconds. In our current products we have the timeout set to 30 seconds, which works well for all devices we currently integrate with. I found the definition
`#define ZB_MAC_INDIRECT_POLL_EXPIRE_TMO ZB_MILLISECONDS_TO_SYS_TIMER_INTERVAL(7680U)`
in the file
`esp-zigbee-sdk/examples/esp_zigbee_HA_sample/HA_on_off_switch/managed_components/espressif__esp-zboss-lib/include/zb_config_common.h`
I changed the value to 30000 and recompiled, but it did not have any effect. I think this value is compiled into the library. Is there a way for us to change this?

2. With the Zigbee stacks we are currently using we can register for a callback that fires upon each MAC data_request received from a sleepy end device. The callback provides the short address of the device sending the data_request. We use this to maintain online/offline status for connected sleepy end devices. In many cases, the data_request is the only packet that is sent periodically by the end device. So, it is important that we get this indication. Is there a callback we can register for to get this notification?

xieqinan, 2025-06-23:
> I changed the value to 30000 and recompiled, but it did not have any effect. I think this value is compiled into the library. Is there a way for us to change this?

The API to set the **MAC polling expiry timeout** is not supported in `esp-zigbee-sdk` v1.6.5. However, you can use the following workaround to configure it manually. For example, you can call:

```c
zb_set_transition_persistence_time(30 * 1000 * 1000);
```

before sending the command. This will cause the command to be stored in the stack for 30 seconds.

Please note that this is only a temporary workaround. An official API for this feature will be provided in a future version of the SDK.

```c
diff --git a/examples/esp_zigbee_HA_sample/HA_on_off_switch/main/esp_zb_switch.c b/examples/esp_zigbee_HA_sample/HA_on_off_switch/main/esp_zb_switch.c
index b3d61ab..a853dbe 100644
--- a/examples/esp_zigbee_HA_sample/HA_on_off_switch/main/esp_zb_switch.c
+++ b/examples/esp_zigbee_HA_sample/HA_on_off_switch/main/esp_zb_switch.c
@@ -37,6 +37,30 @@ static switch_func_pair_t button_func_pair[] = {
 
 static const char *TAG = "ESP_ZB_ON_OFF_SWITCH";
 
+#include "zboss_api.h"
+typedef struct esp_zb_mlme_set_request_s {
+    uint8_t pib_attr;
+    uint8_t iface_id;
+    uint8_t pib_length;
+    uint8_t align[8];
+    uint16_t time;
+} ESP_ZB_PACKED_STRUCT esp_zb_mlme_set_request_t;
+
+extern void zb_mlme_set_request(uint8_t param);
+
+void zb_set_transition_persistence_time(uint32_t us)
+{
+    esp_zb_mlme_set_request_t *req = NULL;
+    uint8_t param = zb_buf_get_out();
+
+    req = (esp_zb_mlme_set_request_t *)zb_buf_initial_alloc(param, sizeof(esp_zb_mlme_set_request_t));
+    req->pib_attr = 0x55;
+    req->iface_id = 0;
+    req->pib_length = sizeof(uint16_t);
+    req->time = us / (16 * 960); // Convert to beacon
+    zb_mlme_set_request(param);
+}
+
 static void zb_buttons_handler(switch_func_pair_t *button_func_pair)
 {
     if (button_func_pair->func == SWITCH_ONOFF_TOGGLE_CONTROL) {
@@ -46,6 +70,7 @@ static void zb_buttons_handler(switch_func_pair_t *button_func_pair)
         cmd_req.address_mode = ESP_ZB_APS_ADDR_MODE_DST_ADDR_ENDP_NOT_PRESENT;
         cmd_req.on_off_cmd_id = ESP_ZB_ZCL_CMD_ON_OFF_TOGGLE_ID;
         esp_zb_lock_acquire(portMAX_DELAY);
+        zb_set_transition_persistence_time(10 * 1000 * 1000);
         esp_zb_zcl_on_off_cmd_req(&cmd_req);
         esp_zb_lock_release();
         ESP_EARLY_LOGI(TAG, "Send 'on_off toggle' command");
@@ -167,6 +192,15 @@ void esp_zb_app_signal_handler(esp_zb_app_signal_t *signal_struct)
             }
         }
         break;
+    case ESP_ZB_NLME_STATUS_INDICATION: {
+        esp_zb_zdo_signal_nwk_status_indication_params_t *param = esp_zb_app_signal_get_params(p_sg_p);
+        printf("Signal: %s, status: 0x%x, addr: 0x%x\n", esp_zb_zdo_signal_to_string(sig_type), param->status,
+               param->network_addr);
+    } break;
+    case ESP_ZB_ZDO_DEVICE_UNAVAILABLE: {
+        esp_zb_zdo_device_unavailable_params_t *param = esp_zb_app_signal_get_params(p_sg_p);
+        printf("Signal: %s, addr: 0x%x\n", esp_zb_zdo_signal_to_string(sig_type), param->short_addr);
+    } break;
     default:
         ESP_LOGI(TAG, "ZDO signal: %s (0x%x), status: %s", esp_zb_zdo_signal_to_string(sig_type), sig_type,
                  esp_err_to_name(err_status));
~

```

> Is there a callback we can register for to get this notification?

No, I actually think it’s a good idea to determine whether a sleep device is online or offline. The `End Device Timeout Request` and `End Device Timeout Response` commands include commissioning mechanisms that allow the stack to track the device’s online status.

If a command's destination is unreachable, the stack will trigger an `ESP_ZB_NLME_STATUS_INDICATION` with the status `ESP_ZB_NWK_COMMAND_STATUS_INDIRECT_TRANSACTION_EXPIRY`, and an `ESP_ZB_ZDO_DEVICE_UNAVAILABLE` signal will be sent to the application. You can refer to the related patch above for implementation details.

By the way, it seems this online/offline status may not be critical for many applications. Could you share your specific use case to help us better understand the need to monitor the remote device’s state?

amp043, 2025-06-23:
I tried `zb_set_transition_persistence_time()` as you suggest, and the result is
`Guru Meditation Error: Core  0 panic'ed (Instruction access fault). Exception was unhandled.`
shortly after calling `esp_zb_aps_data_request()`.

Here is my code:
```
typedef struct esp_zb_mlme_set_request_s {
    uint8_t pib_attr;
    uint8_t iface_id;
    uint8_t pib_length;
    uint8_t align[8];
    uint16_t time;
} ESP_ZB_PACKED_STRUCT esp_zb_mlme_set_request_t;

extern void zb_mlme_set_request(uint8_t param);

void zb_set_transition_persistence_time(uint32_t us)
{
    esp_zb_mlme_set_request_t *req = NULL;
    uint8_t param = zb_buf_get_out();

    req = (esp_zb_mlme_set_request_t *)zb_buf_initial_alloc(param, sizeof(esp_zb_mlme_set_request_t));
    req->pib_attr = 0x55;
    req->iface_id = 0;
    req->pib_length = sizeof(uint16_t);
    req->time = us / (16 * 960); // Convert to beacon
    zb_mlme_set_request(param);
}

static uint8_t asdu_buf[16] = { 0x00, 0x01, 0x02, 0x03, 0x04, 0x05, 0x06, 0x07, 0x08, 0x09, 0x0a, 0x0b, 0x0c, 0x0d, 0x0e, 0x0f };

esp_zb_apsde_data_req_t asdu_req = {
    .dst_addr_mode = ESP_ZB_APS_ADDR_MODE_16_ENDP_PRESENT,      /*!< The addressing mode for the destination address used in this primitive and of the APDU to be transferred,  refer to esp_zb_aps_address_mode_t */
    .dst_addr.addr_short = 0x52a7,     /*!< The individual device address or group address of the entity to which the ASDU is being transferred*/
    .dst_endpoint = 5,       /*!< The number of the individual endpoint of the entity to which the ASDU is being transferred or the broadcast endpoint (0xff).*/
    .profile_id = 0xabcd,        /*!< The identifier of the profile for which this frame is intended. */
    .cluster_id = 0x1234,        /*!< The identifier of the object for which this frame is intended. */
    .src_endpoint = 6,       /*!< The individual endpoint of the entity from which the ASDU is being transferred.*/
    .asdu_length = 16,       /*!< The number of octets comprising the ASDU to be transferred */
    .asdu = asdu_buf,              /*!< The set of octets comprising the ASDU to be transferred. */
    .tx_options = 0,         /*!< The transmission options for the ASDU to be transferred, refer to esp_zb_apsde_tx_opt_t */
    .use_alias = false,             /*!< The next higher layer may use the UseAlias parameter to request alias usage by NWK layer for the current frame.*/
    .alias_src_addr = 0,    /*!< The source address to be used for this NSDU. If the use_alias is true */
    .alias_seq_num = 0,          /*!< The sequence number to be used for this NSDU. If the use_alias is true */
    .radius = 4,             /*!< The distance, in hops, that a transmitted frame will be allowed to travel through the network.*/
};

static bool zb_apsde_data_indication_handler(esp_zb_apsde_data_ind_t ind)
{
    /* Filter out the frame with 0xC0FB profile ID */
    printf("prof=%04x\n", ind.profile_id);
    if (ind.profile_id == 0xC0FB) {
        printf("clus=%04x dst=%04x dstep=%u src=%04x srcep=%u len=%lu\n", ind.cluster_id, ind.dst_short_addr, ind.dst_endpoint, ind.src_short_addr, ind.src_endpoint, ind.asdu_length);
        for (int i = 0; i < ind.asdu_length; ++i) {
            printf("%02x ", ind.asdu[i]);
        }
        esp_zb_lock_acquire(portMAX_DELAY);
        zb_set_transition_persistence_time(30 * 1000 * 1000);
        esp_err_t err_status = esp_zb_aps_data_request(&asdu_req);
        esp_zb_lock_release();
        printf("esp_zb_aps_data_request: %s\n", esp_err_to_name(err_status));
        return true;  // Indicates the frame has been handled
    } else {
        return false; // Let the stack handle other profiles
    }
}
```
The code runs fine if I comment out `zb_set_transition_persistence_time()`, however the persistence time is at 7.68sec of course.

Regarding notification of MAC data_request, it has the following benefits for us:
1. Most of the end devices we integrate with are sensors, and in many cases we only receive data from them and never send data. The most power-constrained devices only send data_request every few minutes, so we cannot send to them using our standard 30 second MAC polling expiry time.
2. Sending data to end devices increases energy consumption on the device and reduces battery life.
3. MAC data_request is generally sent on a regular interval, and data packets are sent based on events, which occur randomly and separated by long periods of inactivity. The regular interval of MAC data_request makes it useful as a link status indicator.
4. Some devices may go multiple hours without sending a data packet, but send MAC data_request much more frequently.
5. Link status is a critical parameter in our system. When a sensor is down, an alarm needs to be generated so that an operator can take action.
6. Most end devices we integrate with are third-party products. So, we cannot modify them to force them to send data packets more often, for example.

xieqinan, 2025-06-24:
@amp043 

> I tried zb_set_transition_persistence_time() as you suggest, and the result is
Guru Meditation Error: Core  0 panic'ed (Instruction access fault). Exception was unhandled.
shortly after calling esp_zb_aps_data_request().

Let’s call `memset(req, 0x00, sizeof(esp_zb_mlme_set_request_t));` to ensure the `req` structure is properly cleared before use.

> Regarding notification of MAC data_request, it has the following benefits for us:

OK, that may be reasonable. However, I still don’t recommend managing the `MAC data_request` configuration of the end device at the application level, since the Zigbee stack already handles similar behavior internally, particularly as reflected in the network neighbor table.

Instead, please refer to the solution below — it provides a more convenient and reliable way to monitor the state of the end device.

```c
void update_end_device_status(uint16_t short_addr)
{
    ESP_LOGW(TAG, "0x%04x is online", short_addr);
}

void lookup_nwk_neighbor_table(uint8_t arg)
{
    esp_err_t ret = ESP_OK;
    esp_zb_nwk_info_iterator_t iterator = 0;
    esp_zb_nwk_neighbor_info_t nbr_info = {0};
    uint32_t next_time = 600 * 1000; // minimum interval: 600 seconds
    do {
        ret = esp_zb_nwk_get_next_neighbor(&iterator, &nbr_info);
        if (ret == ESP_OK) {
            ESP_LOGI(TAG, "device short: 0x%04x, relationship: %d", nbr_info.short_addr, nbr_info.relationship);
            if (nbr_info.relationship == ESP_ZB_NWK_RELATIONSHIP_CHILD) {
                ESP_LOGI(TAG, "ED timeout: %d, timeout_counter: %d", nbr_info.device_timeout, nbr_info.timeout_counter);
                update_end_device_status(nbr_info.short_addr);
                if (next_time > nbr_info.timeout_counter) {
                    next_time = nbr_info.timeout_counter;
                }
            }
        }
    } while (ret == ESP_OK);
    ESP_LOGI(TAG, "Next alarm time: %d ms", next_time);
    esp_zb_scheduler_alarm(lookup_nwk_neighbor_table, 0xFF, next_time);
}

void esp_zb_app_signal_handler(esp_zb_app_signal_t *signal_struct)
{
    uint32_t *p_sg_p       = signal_struct->p_app_signal;
    esp_err_t err_status = signal_struct->esp_err_status;
    esp_zb_app_signal_type_t sig_type = *p_sg_p;
    esp_zb_zdo_signal_device_annce_params_t *dev_annce_params = NULL;
    switch (sig_type) {
      /* ... */
    case ESP_ZB_ZDO_SIGNAL_DEVICE_ANNCE:
        esp_zb_scheduler_alarm_cancel(lookup_nwk_neighbor_table, 0xFF);
        lookup_nwk_neighbor_table(0xFF);
        break;
    }
   /* ... */
}
```

xieqinan, 2025-06-26:
>  Do you have an estimate for when an official API will be available for this?

The API will be included in the next version, which is expected to be released next month.