espressif/esp-zigbee-sdk #679: Cannot change access level for controlSequenceOfOperation (and other attributes)

filip-wizzdev, 2025-07-04:
### Question

Hello,
I'm trying to create thermostat endpoint using the code below:

```
esp_zb_thermostat_cfg_t thermostatConfig                      = ESP_ZB_DEFAULT_THERMOSTAT_CONFIG();
thermostatConfig.thermostat_cfg.control_sequence_of_operation = OPERATION_MODE_HEATING_ONLY;
thermostatConfig.thermostat_cfg.occupied_heating_setpoint = m_setTemperature * ZIGBEE_TEMPERATURE_MULTIPLIER;
thermostatConfig.thermostat_cfg.local_temperature         = m_localTemperature * ZIGBEE_TEMPERATURE_MULTIPLIER;
thermostatConfig.thermostat_cfg.system_mode               = static_cast<uint8_t>(m_systemMode);
thermostatConfig.thermostat_cfg.occupied_cooling_setpoint = OCUPIED_COOLING_SETPOINT;
    
esp_zb_ep_list_t* pThermostatEndpoint = createThermostatEndpoint(THERMOSTAT_ENDPOINT_ID, &thermostatConfig);
```

And it seems that after initialising config this way (by using `ESP_ZB_DEFAULT_THERMOSTAT_CONFIG`), you cannot change attribute access, like you would be able to using `esp_zb_cluster_add_attr` and setting `attr_access` to a adequate value. Is it possible to do this in some kind of workaround way or do I need to add all of the attributes by using the `esp_zb_cluster_add_attr` function?

filip-wizzdev, 2025-07-07:
Hi @xieqinan 
So I managed to make the test pass using the example provided in Github repo and following your steps. It seems that this might be an issue related to creating endpoint/cluster in the code of my device. Also, I've used commands for ZED, not ZC, as my device is supposed to be an end device. So instead of using
```
esp> role zc
esp> network type c
esp> bdb_comm start form
```
I used
```
esp> role zed
esp> bdb_comm start steer
```
as well as changed the `CN-NFS-TC-01` test to `CN-NST-TC-01A`, which as far as I understand does the same thing to ZED as  `CN-NFS-TC-01` does to ZC. 

My question now is what does happen exactly when commands
```
esp> zha add 1 thermostat
esp> dm register
```
are run. Which functions are called underneath to create a thermostat type endpoint with required clusters and mandatory attributes?

I suspect that for `zha add 1 thermostat` command functions from [cli_cmd_zha.c](https://github.com/espressif/esp-zigbee-sdk/blob/main/components/esp-zigbee-console/src/cli_cmd_zha.c) file are being called, but I cannot find exactly what set of functions I could use to recreate `zha add 1 thermostat` functionality.

xieqinan, 2025-07-08:
> Also, I've used commands for ZED, not ZC, as my device is supposed to be an end device. So instead of using...

This step isn't directly related to the `TSTAT-TC-01S` test item. Its main purpose is to ensure the DUT is on the same network as the test harness. If you’ve confirmed the network connection is correct, then your setup should be fine.

> `esp> zha add 1 thermostat`

Please double-check this command. It registers a Thermostat device on **endpoint 1**, but based on my testing, the harness sends commands to **endpoint 10**.
Also, this command registers all mandatory attributes for the `basic`, `identify`, and `thermostat` clusters, which is sufficient for testing the `TSTAT-TC-01S` item.

If you want to recreate or customize the data model for your device, please refer to the [console README](https://github.com/espressif/esp-zigbee-sdk/blob/main/components/esp-zigbee-console/README.md#dm) for more details.

xieqinan, 2025-09-30:
Hi @filip-wizzdev,

Do you have any updates about this issue?  If the issue had been resolved, please consider closing it.