espressif/esp-zigbee-sdk #680: Read Attribute Response Fails with Status 0x86 When SPIFFS or LittleFS Is Initialized on Host (ESP32S3)

Dev-Hussam, 2025-07-05:
### Answers checklist.

- [x] I have read the documentation [ESP Zigbee SDK Programming Guide](https://docs.espressif.com/projects/esp-zigbee-sdk) and tried the [debugging tips](https://docs.espressif.com/projects/esp-zigbee-sdk/en/latest/esp32/developing.html#debugging), the issue is not addressed there.
- [x] I have updated ESP Zigbee libs ([esp-zboss-lib](https://components.espressif.com/components/espressif/esp-zboss-lib) and [esp-zigbee-lib](https://components.espressif.com/components/espressif/esp-zigbee-lib)) to the latest version, with corresponding IDF version, and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.3.2 and v5.4.0

### esp-zigbee-lib version.

1.6.5

### esp-zboss-lib version.

1.6.4

### Espressif SoC revision.

ESP32S3

### What is the expected behavior?

Mounting a SPIFFS or LittleFS partition should not interfere with Zigbee attribute reads.

### What is the actual behavior?

Mounting a SPIFFS or LittleFS partition causes Basic cluster read attribute responses to fail with status 0x86

### Steps to reproduce.

1. Build a Zigbee Host-RCP Gateway using ESP32S3 (Host) and ESP32H2 (RCP).
2. Add a SPIFFS (or LittleFS) partition to the host project.
3. Mount the partition in a function (e.g. init_spiffs()), not during startup.
4. Start the Zigbee network and join a device.
5. After the device joins, trigger attribute discovery on the Basic cluster.
6. Observe attribute read responses returning error status 0x86.
7. Comment out the SPIFFS/LittleFS mounting code and re-test — the issue no longer occurs.

### More Information.

 I mount a SPIFFS partition using the following function:
`void init_spiffs()
{

    esp_vfs_spiffs_conf_t conf = {
        .base_path = "/spiffs",
        .partition_label = NULL,
        .max_files = 5,
        .format_if_mount_failed = false,
    };

    esp_err_t ret = esp_vfs_spiffs_register(&conf);
    if (ret != ESP_OK)
    {
        ESP_LOGE(TAG, "Failed to mount SPIFFS (%s)", esp_err_to_name(ret));
        return;
    }

    spiffs_initialized = true; // mark as mounted

    size_t total = 0, used = 0;
    ret = esp_spiffs_info(NULL, &total, &used);
    if (ret != ESP_OK)
    {
        ESP_LOGE(TAG, "Failed to get SPIFFS info (%s)", esp_err_to_name(ret));
    }
    else
    {
        ESP_LOGI(TAG, "SPIFFS mounted: %d/%d KB used", used / 1024, total / 1024);
    }
}`
This function is not called at startup, but later (after joining) from a task unrelated to Zigbee network formation or discovery.

After a device joins the Zigbee network, the host performs discovery using the Basic cluster and sends attribute read requests. However, the read responses return status 0x86 (Unsupported Attribute) for multiple attribute IDs (e.g., 0x4, 0x5, 0x9, 0xC):

`I (11118) DEVICE_DISCOVERY: Reading node attributes(count=4) for device with short address: 0x49c2
I (11188) ATTR_HANDLER: Read attribute response: from address(0x49c2) src endpoint(1) to dst endpoint(1) cluster(0x0)
I (11188) ATTR_HANDLER: Read attribute response: status(0x86), cluster(0x0), attribute(0x5), type(0x0), value(0)
E (11188) ATTR_HANDLER: Read attribute response: error status(0x86)
I (11198) ATTR_HANDLER: Read attribute response: status(0x86), cluster(0x0), attribute(0x4), type(0x0), value(0)
E (11208) ATTR_HANDLER: Read attribute response: error status(0x86)
I (11218) ATTR_HANDLER: Read attribute response: status(0x86), cluster(0x0), attribute(0xc), type(0x0), value(0)
E (11228) ATTR_HANDLER: Read attribute response: error status(0x86)
I (11228) ATTR_HANDLER: Read attribute response: status(0x86), cluster(0x0), attribute(0x9), type(0x0), value(0)
E (11238) ATTR_HANDLER: Read attribute response: error status(0x86)`
This error is reported in the following handler:
`esp_err_t zb_read_attr_resp_handler(const esp_zb_zcl_cmd_read_attr_resp_message_t *message) {
    ...
    esp_zb_zcl_read_attr_resp_variable_t *variable = message->variables;
    while (variable) {
        ESP_LOGI(TAG, "Read attribute response: status(0x%x), cluster(0x%x), attribute(0x%x), type(0x%x), value(%d)",
                    variable->status, message->info.cluster,
                    variable->attribute.id, variable->attribute.data.type,
                    variable->attribute.data.value ? *(uint8_t *)variable->attribute.data.value : 0);
        ...
        variable = variable->next;
    }
    return ESP_OK;
}`

**More Information:**

- Device joins correctly
- `esp_zb_zdo_active_ep_req` and `esp_zb_zdo_simple_desc_req `complete successfully
- Attribute read fails only when FS (SPIFFS or LittleFS) is mounted
- When I comment out the init_spiffs() call, the issue disappears
- Same behavior occurs with LittleFS

**Request:** Please help investigate whether this is an SDK-level issue related to Zigbee stack memory usage, task priorities, or event handling conflicts introduced by FS registration.

lpy4105, 2025-07-07:
Hi @Dev-Hussam, 

Could you please share the snippet that creates the ZCL data model? `0x04, 0x05, 0x09, 0x0C` are all optional attributes of basic cluster, you need to add them manully.

xieqinan, 2025-07-07:
@Dev-Hussam, You can refer to this [function](https://github.com/espressif/esp-zigbee-sdk/blob/main/examples/common/zcl_utility/src/zcl_utility.c#L22) as a guide to add the missing attributes.

Dev-Hussam, 2025-07-07:
Thanks for Replies @xieqinan  @lpy4105,
i don't have problem with adding the attributes because that the Host **can** Read these attributes when i Remove the call for this function(**init_spiffs()**), however this is how i add them on end device: 
`// 1. Add Basic Cluster
    esp_zb_attribute_list_t *basic_cluster = esp_zb_basic_cluster_create(&(light_cfg->basic_cfg));
    ESP_ERROR_CHECK(esp_zb_basic_cluster_add_attr(basic_cluster, ESP_ZB_ZCL_ATTR_BASIC_MANUFACTURER_NAME_ID, MANUFACTURER_NAME));
    ESP_ERROR_CHECK(esp_zb_basic_cluster_add_attr(basic_cluster, ESP_ZB_ZCL_ATTR_BASIC_MODEL_IDENTIFIER_ID, MODEL_IDENTIFIER));
    uint8_t GENERIC_DEVICE_TYPE = CONFIG_ZB_GENERIC_DEVICE_TYPE;
    ESP_ERROR_CHECK(esp_zb_basic_cluster_add_attr(basic_cluster, ESP_ZB_ZCL_ATTR_BASIC_GENERIC_DEVICE_TYPE_ID, &GENERIC_DEVICE_TYPE));
    ESP_ERROR_CHECK(esp_zb_basic_cluster_add_attr(basic_cluster, ESP_ZB_ZCL_ATTR_BASIC_MANUFACTURER_VERSION_DETAILS_ID, FW_VERSION));
    ESP_ERROR_CHECK(esp_zb_cluster_list_add_basic_cluster(cluster_list, basic_cluster, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE));`

i tried to add end device from another Vendor that i already configure and add these attributes to it, but the same error occurs,
- when i comment out the call for **init_spiffs()** on ESP32-S3 HOST, the HOST **can** read them without  error, but when i add this function even if not in startup the error occur.
- I also tried the To Run the ESP32-H2 as SoC Coordinator and call this function: init_spiffs() , This error did not occur.

this is how i send Read Attr Request on HOST:
`uint16_t attributes[] = {
    ESP_ZB_ZCL_ATTR_BASIC_MODEL_IDENTIFIER_ID,
    ESP_ZB_ZCL_ATTR_BASIC_MANUFACTURER_NAME_ID,
    ESP_ZB_ZCL_ATTR_BASIC_MANUFACTURER_VERSION_DETAILS_ID,
    ESP_ZB_ZCL_ATTR_BASIC_GENERIC_DEVICE_TYPE_ID
  };
  Read_node_attributes(short_id, ESP_ZB_ZCL_CLUSTER_ID_BASIC, attributes, 4);

void Read_node_attributes(uint16_t short_id, uint16_t cluster_id, uint16_t attributes[], size_t attr_count) 
{
  ESP_LOGI(TAG, "Reading node attributes(count=%d) for device with short address: 0x%04x", (int)attr_count, short_id);

  esp_zb_zcl_read_attr_cmd_t read_req;
  read_req.zcl_basic_cmd.src_endpoint = ESP_ZB_PRIMARY_ENDPOINT;
  read_req.address_mode = ESP_ZB_APS_ADDR_MODE_16_ENDP_PRESENT;
  read_req.zcl_basic_cmd.dst_endpoint = 1;
  read_req.zcl_basic_cmd.dst_addr_u.addr_short = short_id;
  read_req.clusterID = cluster_id;

  read_req.attr_number = attr_count;
  read_req.attr_field = attributes;

  esp_zb_zcl_read_attr_cmd_req(&read_req);
}`

lpy4105, 2025-07-07:
> the Host can Read these attributes when i Remove the call for this function(init_spiffs())

That's interesting. We will try to reproduce this issue, a mini example code or patch would be help. Meanwhile, could you please help to capture a pcap file of the packet exchanges of the two devices?

Dev-Hussam, 2025-07-07:
thanks for your time,
Here is the PCAP capture from EFR32MG24 device using Network Analyzer in PCAPNG format(remove the .txt): 
[zigbee_capture.pcap.txt](https://github.com/user-attachments/files/21102967/zigbee_capture.pcap.txt)

- i use unique install code for joining. 
- i resend the Read Attribute Req about 5 or 6 times.
- the is EFR32MG24 works as router.
- i also noticed that there is a manufacture code filed in the request that filled with the destination short Id, witch i don't understand way?
- this is a picture from the network analyzer:

[image]

Dev-Hussam, 2025-07-07:
this is the NWK key: BA:8C:16:72:CB:D8:AF:A2:BF:ED:2F:CE:36:B8:08:80
pre-config key : 59:AA:2D:D7:AF:1F:EE:25:06:B9:99:B3:BE:1F:C1:D0