espressif/esp-zigbee-sdk #681: Zigbee stack assertion failed common/zb_bufpool_mult.c:1328

michail-nikolaev, 2025-07-05:
### Answers checklist.

- [x] I have read the documentation [ESP Zigbee SDK Programming Guide](https://docs.espressif.com/projects/esp-zigbee-sdk) and tried the [debugging tips](https://docs.espressif.com/projects/esp-zigbee-sdk/en/latest/esp32/developing.html#debugging), the issue is not addressed there.
- [x] I have updated ESP Zigbee libs ([esp-zboss-lib](https://components.espressif.com/components/espressif/esp-zboss-lib) and [esp-zigbee-lib](https://components.espressif.com/components/espressif/esp-zigbee-lib)) to the latest version, with corresponding IDF version, and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

5.4.1

### esp-zigbee-lib version.

1.6.3

### esp-zboss-lib version.

1.6.3

### Espressif SoC revision.

ESP32C6

### What is the expected behavior?

We should ignored manufacture-specific commands by default instead of panic.

### What is the actual behavior?

```
Zigbee stack assertion failed common/zb_bufpool_mult.c:1328

abort() was called at PC 0x420260d7 on core 0
Core  0 register dump:
MEPC    : 0x40801cd6  RA      : 0x40807d72  SP      : 0x4083fa10  GP      : 0x408146c4
TP      : 0x4083fca0  T0      : 0x37363534  T1      : 0x7271706f  T2      : 0x33323130
S0/FP   : 0x4083fa3c  S1      : 0x4083fa3c  A0      : 0x4083fa48  A1      : 0x4083fa2a
A2      : 0x00000000  A3      : 0x4083fa75  A4      : 0x00000001  A5      : 0x40822000
A6      : 0x00000000  A7      : 0x76757473  S2      : 0x0000001a  S3      : 0x40839109
S4      : 0x02ec8ecb  S5      : 0x00000000  S6      : 0x000186a0  S7      : 0xfffffffe
S8      : 0x00000000  S9      : 0x00000000  S10     : 0x00000012  S11     : 0x00000000
T3      : 0x6e6d6c6b  T4      : 0x6a696867  T5      : 0x66656463  T6      : 0x62613938
MSTATUS : 0x00001881  MTVEC   : 0x40800001  MCAUSE  : 0x00000002  MTVAL   : 0x00000000
MHARTID : 0x00000000
```

### Steps to reproduce.

Receive Hue-specific ADD SCENE COMMAND:

```
Frame 14: 74 bytes on wire (592 bits), 74 bytes captured (592 bits) on interface ZigBee/0/11, id 0
    Section number: 1
    Interface id: 0 (ZigBee/0/11)
    Encapsulation type: IEEE 802.15.4 Wireless PAN with FCS not present (127)
    Arrival Time: Jul  5, 2025 15:27:38.873193000 W. Europe Daylight Time
    UTC Arrival Time: Jul  5, 2025 13:27:38.873193000 UTC
    Epoch Arrival Time: 1751722058.873193000
    [Time shift for this packet: 0.000000000 seconds]
    [Time delta from previous captured frame: 0.004050000 seconds]
    [Time delta from previous displayed frame: 0.004050000 seconds]
    [Time since reference or first frame: 2.326193000 seconds]
    Frame Number: 14
    Frame Length: 74 bytes (592 bits)
    Capture Length: 74 bytes (592 bits)
    [Frame is marked: False]
    [Frame is ignored: False]
    [Protocols in frame: wpan:zbee_nwk:zbee_aps:zbee_zcl:data]
IEEE 802.15.4 Data, Src: 0x1bcf, Dst: Broadcast
    Frame Control Field: 0x8841, Frame Type: Data, PAN ID Compression, Destination Addressing Mode: Short/16-bit, Frame Version: IEEE Std 802.15.4-2003, Source Addressing Mode: Short/16-bit
    Sequence Number: 189
    Destination PAN: 0x4ff5
    Destination: 0xffff
    Source: 0x1bcf
    [Extended Source: PhilipsLight_01:0e:72:83:24 (00:17:88:01:0e:72:83:24)]
    [Origin: 5]
ZigBee Network Layer Data, Dst: Broadcast, Src: 0x3f26
    Frame Control Field: 0x0208, Frame Type: Data, Discover Route: Suppress, Security Data
    Destination: 0xfffd
    Source: 0x3f26
    Radius: 29
    Sequence Number: 169
    [Extended Source: PhilipsLight_01:05:58:4e:73 (00:17:88:01:05:58:4e:73)]
    [Origin: 3]
    ZigBee Security Header
ZigBee Application Support Layer Data, Group: 0x508d, Src Endpt: 64
    Frame Control Field: Data (0x0c)
    Group: 0x508d
    Cluster: Scenes (0x0005)
    Profile: Home Automation (0x0104)
    Source Endpoint: 64
    Counter: 110
ZigBee Cluster Library Frame, Mfr: Philips (0x100b)
    Frame Control Field: Cluster-specific (0x15)
    Manufacturer Code: Philips (0x100b)
    Sequence Number: 38
    Command: Unknown (0x00)
Data (25 bytes)
    Data: 1b50020c13500000001de498b9a3cc0c9b8563bb6cc1ae5db0
    [Length: 25]
```

Workaround:
```C
[[maybe_unused]]
static bool zb_raw_command_handler(uint8_t bufid) {
    uint8_t buf[zb_buf_len(bufid)];
    zb_zcl_parsed_hdr_t *cmd_info = ZB_BUF_GET_PARAM(bufid, zb_zcl_parsed_hdr_t);
    memcpy(buf, zb_buf_begin(bufid), sizeof(buf));
    if (cmd_info->cluster_id == ESP_ZB_ZCL_CLUSTER_ID_SCENES) {
        if (cmd_info->cmd_id == ESP_ZB_ZCL_CMD_SCENES_ADD_SCENE && cmd_info->is_manuf_specific && cmd_info->manuf_specific == 0x100b) {
            log_v("Received manuf specific add scene - response with ERROR to avoid assertion");
            zb_zcl_send_default_handler(bufid, cmd_info, ZB_ZCL_STATUS_FAIL);
            return true;
        }
```1

### More Information.

[dumps.zip](https://github.com/user-attachments/files/21076701/dumps.zip)

Keys:
```
"81:45:86:86:5D:C6:C8:B1:C8:CB:C4:2E:5D:65:D3:B9","Normal",""
"5A:69:67:42:65:65:41:6C:6C:69:61:6E:63:65:30:39","Normal",""
"81:42:86:86:5D:C1:C8:B2:C8:CB:C5:2E:5D:65:D1:B8","Normal",""
"9F:55:95:F1:02:57:C8:A4:69:CB:F4:2B:C9:3F:EE:31","Normal",""
"2c:23:ea:72:5e:b3:01:05:40:82:64:d8:cc:d1:0e19","Normal",""
```

xieqinan, 2025-07-07:
@michail-nikolaev ,

I have confirmed this issue: the ZCL layer of the ZBOSS stack does not handle manufacturer-specific commands because the payload structure differs from standard ZCL commands.

Therefore, when a device receives a manufacturer-specific command, you need to register a callback using `esp_zb_raw_command_handler_register()` to intercept and process the command — just like the workaround you provided.

xieqinan, 2025-08-29:
Hi @michail-nikolaev ,

The issue has been fixed in [esp-zigbee-sdk v1.6.7](https://github.com/espressif/esp-zigbee-sdk/commit/4cab73fed37cb8fe4150d0b4fa180031ca30c16a#diff-68d24fa81558aae3d8c59e2aa57a4fa719ea3b04d7fa14beff45f16f00858f50R20), please remove the workaround and update the new version to test again.

Please feel free to reopen this issue, if the problem still confuses you.

closed 2025-08-29 by xieqinan