espressif/esp-zigbee-sdk #682: Creating thermostat cluster using esp_zb_thermostat_cluster_create adds unsupported attributes

filip-wizzdev, 2025-07-09:
### Question

Hi,
I'm trying to create thermostat endpoint with necessary clusters. One of the clusters that is necessary to be supported is the thermostat cluster. Thermostat cluster has 3 mandatory attributes:

- LocalTemperature (0x0000),
- ControlSequenceOfOperation (0x001b),
- SystemMode (0x001c).

It also has two attributes, one of which needs to be supported (in ZCL library documentation it is exaplained that at least one of the two needs to be supported):

- OccupiedCoolingSetpoint (0x0011),
- OccupiedHeatingSetpoint (0x0012).

I was trying to create thermostat cluster based on the `ESP_ZB_DEFAULT_THERMOSTAT_CONFIG()` and using it as an argument to the `esp_zb_thermostat_cluster_create` function.

```
esp_zb_thermostat_cfg_t thermostatConfig = ESP_ZB_DEFAULT_THERMOSTAT_CONFIG();
esp_zb_thermostat_cluster_create(thermostatConfig->thermostat_cfg);
```
Doing is that way causes both of the optional attributes to be added. In my project I only want to use the _OccupiedHeatingSetpoint_ attribute (my device is a heating only device). At this point I tried redefining the `ESP_ZB_DEFAULT_THERMOSTAT_CONFIG()` config, which has _OccupiedCoolingSetpoint_ already added, like this:

```
esp_zb_thermostat_cfg_t thermostatConfig = {
        .basic_cfg =
            {
                .zcl_version  = ESP_ZB_ZCL_BASIC_ZCL_VERSION_DEFAULT_VALUE,
                .power_source = ESP_ZB_ZCL_BASIC_POWER_SOURCE_DEFAULT_VALUE,
            },
        .identify_cfg =
            {
                .identify_time = ESP_ZB_ZCL_IDENTIFY_IDENTIFY_TIME_DEFAULT_VALUE,
            },
        .thermostat_cfg =
            {
                .local_temperature             = ESP_ZB_ZCL_THERMOSTAT_LOCAL_TEMPERATURE_DEFAULT_VALUE,
                .occupied_heating_setpoint     = ESP_ZB_ZCL_THERMOSTAT_OCCUPIED_HEATING_SETPOINT_DEFAULT_VALUE,
                .control_sequence_of_operation = ESP_ZB_ZCL_THERMOSTAT_CONTROL_SEQ_OF_OPERATION_DEFAULT_VALUE,
                .system_mode                   = ESP_ZB_ZCL_THERMOSTAT_CONTROL_SYSTEM_MODE_DEFAULT_VALUE,
            },
    };
```

This didn't solve anything because somewhere in the `esp_zb_thermostat_cluster_create` _OccupiedCoolingSetpoint_ is added anyway. I tried to make a workaround solution, creating thermostat cluster from scratch using `esp_zb_cluster_add_attr` and purposefully omitting the _OccupiedCoolingSetpoint_ attribute:

```
esp_zb_attribute_list_t* pThermostatCluster = esp_zb_zcl_attr_list_create(0x201);

int16_t heatingMinTemperature = HEATING_MIN * ZIGBEE_TEMPERATURE_MULTIPLIER;
    int16_t heatingMaxTemperature = HEATING_MAX * ZIGBEE_TEMPERATURE_MULTIPLIER;

    int16_t controlSequenceOfOperation = 0x02;

    esp_zb_cluster_add_attr(
        pThermostatCluster,
        0x201,
        ESP_ZB_ZCL_ATTR_THERMOSTAT_LOCAL_TEMPERATURE_ID,
        ESP_ZB_ZCL_ATTR_TYPE_S16,
        ESP_ZB_ZCL_ATTR_ACCESS_READ_ONLY,
        &controlSequenceOfOperation);

    esp_zb_cluster_add_attr(
        pThermostatCluster,
        0x201,
        ESP_ZB_ZCL_ATTR_THERMOSTAT_OCCUPIED_HEATING_SETPOINT_ID,
        ESP_ZB_ZCL_ATTR_TYPE_S16,
        ESP_ZB_ZCL_ATTR_ACCESS_READ_WRITE,
        &occupiedHeatingSetpoint);

    esp_zb_cluster_add_attr(
        pThermostatCluster,
        0x201,
        ESP_ZB_ZCL_ATTR_THERMOSTAT_SYSTEM_MODE_ID,
        ESP_ZB_ZCL_ATTR_TYPE_8BIT_ENUM,
        ESP_ZB_ZCL_ATTR_ACCESS_READ_WRITE,
        &systemMode);

    esp_zb_cluster_add_attr(
        pThermostatCluster,
        0x201,
        ESP_ZB_ZCL_ATTR_THERMOSTAT_CONTROL_SEQUENCE_OF_OPERATION_ID,
        ESP_ZB_ZCL_ATTR_TYPE_8BIT_ENUM,
        ESP_ZB_ZCL_ATTR_ACCESS_READ_WRITE,
        &controlSequenceOfOperation);

```

This seemed to fix this problem (tests related to the _OccupiedCoolingSetpoint_ were passing), but caused an issue with the Server Command _Setpoint Raise/Lower_ for some reason (tests related to this server command stopped passing). This issue didn't arise when I was using `esp_zb_thermostat_cluster_create` command, but as I described earlier, I cannot use this function in my case. This workaround also caused some issue in our device when it reveiced _Setpoint Raise/Lower_  Command. 

```
Guru Meditation Error: Core  0 panic'ed (Load access fault). Exception was unhandled.
Core  0 register dump:
MEPC    : 0x420ab56a  RA      : 0x420ab53c  SP      : 0x4082e410  GP      : 0x408180b0
--- Stack dump detected
--- 0x420ab56a: zb_zcl_thermostat_setpoint_raise_lower_handler at ??:?
--- 0x420ab53c: zb_zcl_thermostat_setpoint_raise_lower_handler at ??:?
TP      : 0x407e41f0  T0      : 0x40030dca  T1      : 0x0000000f  T2      : 0xc7488861
--- 0x40030dca: memset in ROM
S0/FP   : 0x00000001  S1      : 0x00000000  A0      : 0x4082f372  A1      : 0x00000011
A2      : 0x0000ffff  A3      : 0x00000898  A4      : 0x40834064  A5      : 0x00000000
A6      : 0x00000008  A7      : 0x00000008  S2      : 0x00000000  S3      : 0x40834004
S4      : 0x00000001  S5      : 0x00000000  S6      : 0x000186a0  S7      : 0xfffffffe
S8      : 0x00000000  S9      : 0x00000000  S10     : 0x0000000f  S11     : 0x00000000
T3      : 0x00000001  T4      : 0x7fffffff  T5      : 0x0000000f  T6      : 0x000000ff
MSTATUS : 0x00001881  MTVEC   : 0x40800001  MCAUSE  : 0x00000005  MTVAL   : 0x00000007
--- 0x40800001: _vector_table at ??:?
MHARTID : 0x00000000
--- Backtrace:
0x420ab56a in zb_zcl_thermostat_setpoint_raise_lower_handler ()
#0  0x420ab56a in zb_zcl_thermostat_setpoint_raise_lower_handler ()
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
```

filip-wizzdev, 2025-07-09:
@xieqinan 

My question is based on the Cluster Library Specification in the newest version (latest revision December 2019) from Zigbee Alliance. 
In the table below _OccupiedHeatingSetpoint_ and _OccupiedCoolingSetpoint_ are marked with _M*_, which indicates that you need to implement only one of these attributes, if your device doesn't support the other functionality.
[image]

Answering your question about which specific step fails: we're wondering if the Zigbee certification laboratory will stop certification of our device. The device is a radiator and adding handling attributes related to cooling might be questionable for them.

If adding only one of these attributes isn't supported currently in the **esp-zigbee-sdk** how long will it take to add this functionality?

xieqinan, 2025-07-10:
@filip-wizzdev ,

> My question is based on the Cluster Library Specification in the newest version (latest revision December 2019) from Zigbee Alliance.

Yes, I noticed that description as well. However, I believe the phrase **“would only”** indicates that a radiator device is only *required* to implement the `OccupiedHeatingSetpoint` attribute, but it **does not forbid** the inclusion of the `OccupiedCoolingSetpoint` attribute.

So, could you please test your implementation with both attributes included?

If the Zigbee certification lab explicitly disallows the presence of `OccupiedCoolingSetpoint` in a radiator device, please let us know. We will respond quickly and provide a proper solution to meet the certification requirements.