espressif/esp-zigbee-sdk #683: Interaction beetween Setpoint Raise/Lower command and setpoint limit

Antoni-Wizzdev, 2025-07-11:
### Question

I'm currently preparing device for certification and run into an issue regarding thermostat cluster while using ZUTH with test harness. Inside test TSTAT-TC-02S steps 14d and 14e attempts to use Setpoint Raise/Lower command to lower the OccupiedHeatingSetpoint by 3 degrees and if the value lands below value of MinHeatSetpointLimit it expects the OccupiedHeatingSetpoint to be set equal to MinHeatlSetpointLimit. Is Setpoint Raise/Lower command supposed to work like that according to the standard? I attempted to find some information about this but could not find anything pointing to this being the intended functionality. I'm asking because the current ZCL sdk I'm using instead of setting the setpoint to minimum returns information about invalid value. I'm wondering if the error lies on the side of sdk or the test script. Bellow is a screenshots of the ZUTH output and relevant part of the test script.

[image]

`print_step_info('14d', 'TH CLIENT unicast a setpoint raise/lower command frame of the thermostat cluster to DUT SERVER with:'
                                    '\n- mode field set to Heat (0x00)'
                                    '\n- amount field set to 0xE2 (-30 units = -3 degrees)')
           
            status, response = zcl_command(Zcl, dut,
                            zcl_tstat.SetpointRaiseOrLower(Mode=0x00, Amount=-30),
                            default_rsp,
                            zclStatus.SUCCESS)
           
            autoDUT.prompt_wait(5)

            ##########################
            # mode fields supported
            ##########################
            # 0x00 Heat (adjust Heat Setpoint)
            # 0x01 Cool (adjust Cool Setpoint)
            # 0x02 Both (adjust Heat Setpoint and Cool Setpoint)

            # Adjust the heat set point Mode 0
            # The amount needs to be in range, it cannot be any arbitrary value

            OccupiedHeatingSetpoint,min_value,max_value = self.get_attribute_limits("OccupiedHeatingSetpoint",PICS_ITEMS)
           

            # check if the expected value after the adjustment will be less than the maximum allowed
            if ( OccupiedHeatingSetpoint - 300 ) < min_value:
                expected_reading = min_value
            else:
                expected_reading = OccupiedHeatingSetpoint-300 # -30 needs to be scalated

            print_step_info('14e', 'TH CLIENT unicast a ZCL read attributes command frame to DUT SERVER to read the OccupiedHeatingSetpoint attribute.')
            # attribute_objs = [{KEY_ATTR_OBJ: zcl_tstat.OccupiedHeatingSetpoint, KEY_ATTR_VAL: 0x06A4}]
            attribute_objs = [{KEY_ATTR_OBJ: zcl_tstat.OccupiedHeatingSetpoint, KEY_ATTR_VAL: expected_reading}]
            step_14e = self.read_attribute_tracker(Zcl, dut, tstat_cluster_id, attribute_objs)`

xieqinan, 2025-07-14:
@Antoni-Wizzdev 

>  Is Setpoint Raise/Lower command supposed to work like that according to the standard?

No. Where can I find the standard? The SDK only checks the `current_value` and `amount_value`; if they fall within the valid range, the raise/lower command will be accepted. Otherwise, the command will respond with `INVALID_VALUE (0x87)`.

So, if you adjust the `MinHeatSetpointLimit` to a proper value, can **Step 14** be passed?

Antoni-Wizzdev, 2025-07-14:
Step 14 only one of a large group of tests that test all possible uses of  Setpoint Raise/Lower command. And as far as i understand some setpoint limits are changed to what appears to be a random valid value in the previous steps of the same test so modifying them on my end doesn't change the test outcome.

xieqinan, 2025-07-15:
@Antoni-Wizzdev ，

Yes, we have different versions, which is causing some discrepancies. I've added a patch here that can serve as a workaround to handle the Thermostat Raise/Lower command in the application. You can use it as a base to adapt for ZUTH certification.

[thermostat-raise-lower-cmd.zip](https://github.com/user-attachments/files/21228136/thermostat-raise-lower-cmd.zip)

> if the value lands below value of MinHeatSetpointLimit it expects the OccupiedHeatingSetpoint to be set equal to MinHeatlSetpointLimit.

Meanwhile, we will verify with the standard to determine whether this behavior is necessary.

Antoni-Wizzdev, 2025-07-15:
Thank you for the patch. Additionaly i'm not sure if it will be of any help but i received answer from member of CSA on their forums to the same question here is what he said. 

> [Jose Alvarez](https://community.csa-iot.org/users/5400657)
13 hours ago
Hi Filip,
You can check the Thermostat test plan in Causeway https://groups.csa-iot.org/wg/zwg-bdb/document/34773
Also check the Zigbee Cluster Library at section 6.3.2.3 https://groups.csa-iot.org/wg/members-all/document/23019
The Test Case is aligned with the spec requirements. Make sure that the value ranges in your DUT are set properly.
If you still find that the Test Script is not behaving correctly, please file a CCB to the Zigbee Working group describing your setup, PICS and default attribute values. The group will then evaluate the issue and provide a solution.
https://zigbeecertifiedproducts.knack.com/zigbee-alliance-ccb-tool#home/
regards,

xieqinan, 2026-01-20:
Based on the email, your product had passed the ZUTH certification. Let's close this issue. If you have any other questions, you can submit a new issue here.