espressif/esp-zigbee-sdk #685: ZED fails to join distributed network

Antoni-Wizzdev, 2025-07-16:
### Question

Im working on zigbee certification using ZUTH and during test DN-DNS-TC-05(version BDB3.0.1:0:20220620) test harness creates distributed network and my end device manages to pass Association step but is kicked soon after that. When that happenes device receives ESP_ZB_ZDO_SIGNAL_LEAVE. I used nRF Sniffer for 802.15.4 and wireshark to capture the communication you can find it in the attached zip. In addition to that i attach screenshot o ZUTH output. 

[image]

[DN-DNS-TC-05.zip](https://github.com/user-attachments/files/21252629/DN-DNS-TC-05.zip)

xieqinan, 2025-07-16:
Please attempt to add the code below before the `esp_zb_start()` and test again.
```
esp_zb_enable_joining_to_distributed(true);
esp_zb_joining_to_distributed_network_enabled();
```

Antoni-Wizzdev, 2025-07-16:
Thank you this appears to have fixed the issue.