espressif/esp-zigbee-sdk #692: How to Identify and Register Device After `ESP_ZB_ZDO_SIGNAL_DEVICE_ANNCE` in Zigbee Gateway Example

iamiha, 2025-08-03:
### Question

Hi Espressif Team,

I'm working with the Zigbee Gateway example from the `esp-zigbee-sdk`, and I’m handling the `ESP_ZB_ZDO_SIGNAL_DEVICE_ANNCE` signal to detect when a new Zigbee device joins the network.

In the signal handler, I can extract and log the short address and IEEE address:

```c
esp_zb_zdo_signal_device_annce_params_t *dev_annce_params =
    (esp_zb_zdo_signal_device_annce_params_t *)esp_zb_app_signal_get_params(p_sg_p);

ESP_LOGI(TAG, "New device commissioned or rejoined (short: 0x%04hx)", dev_annce_params->device_short_addr);
```

---

### What I Want to Achieve:

1. **Identify the type of device** that has joined (e.g., light, sensor, switch).
2. **Register the device under the gateway.**
   For example, if it's an on/off light, I want to add it using:

```c
esp_zb_zdo_match_desc_req_param_t cmd_req;
cmd_req.dst_nwk_addr = dev_annce_params->device_short_addr;
cmd_req.addr_of_interest = dev_annce_params->device_short_addr;
esp_zb_zdo_find_on_off_light(&cmd_req, user_find_cb, NULL);
```

But for this, I first need to determine what type of device it is.

---

### What I’ve Tried:

* After receiving the device announce signal, I send a **Simple Descriptor Request** to the same short address.
* In the `zigbee_bridge_simple_desc_cb()` callback, I extract:

  * `app_profile_id`
  * `app_device_id`
  * Input/output clusters

---

### What I'm Observing:

* The `app_profile_id` is correctly `0x0104` (Home Automation).
* However, the `app_device_id` is often not one of the known `ESP_ZB_HA_*_DEVICE_ID` values.
* I cannot reliably determine the type of device based on the current information.

---

### My Questions:

1. What is the **recommended flow** for identifying the type of a Zigbee device after receiving `ESP_ZB_ZDO_SIGNAL_DEVICE_ANNCE`?

---

### My Goal:

* Automatically detect and identify a device when it joins.
* Log or send the device's profile/device ID and cluster info to a backend.
* Register the device in the gateway app (e.g., create an on/off light endpoint if it's a light).

Could you please guide me on how to achieve this?

Thanks!

xieqinan, 2025-08-04:
@iamiha ,

> Automatically detect and identify a device when it joins.

I think it’s reasonable to send a ZDO Simple Descriptor Request to the joined node to obtain its information. The ZDO Node Descriptor and Power Descriptor can also help identify relevant details of a joined node. The `app_device_id` is developer-configurable and can be set to any value. If the `app_device_id` is not valid according to your application’s list, you can call `esp_zb_zdo_device_leave_req()` to remove it to ensure safe.

> Register the device in the gateway app (e.g., create an on/off light endpoint if it's a light).

The current version (v1.6.6) does not support adding new endpoints after calling `esp_zb_device_register()`. However, if your device is acting as a gateway, you can use `esp_zb_ep_list_add_gateway_ep()` to enable the Zigbee gateway feature, allowing it to handle all supported cluster requests.

xieqinan, 2025-09-30:
Hi @iamiha,

Do you have any updates about this issue?

closed 2025-11-12 by iamiha