espressif/esp-zigbee-sdk #693: Getting error ESP_ZIGBEE_CORE : Unknown device type!

AumDhabalia, 2025-08-04:
This code is written from the programming guide's ch 5.3 Zigbee ZCL Custom Cluster and still I am having the same error "**ESP_ZIGBEE_CORE : Unknown device type!**". Need help in solving this issue. 

[main.txt](https://github.com/user-attachments/files/21572385/main.txt)

[image]
[image]

xieqinan, 2025-08-04:
After reviewing your code, it seems you intend to create a coordinator device, correct? If so, please check your `sdkconfig` to ensure that `CONFIG_ZB_ENABLED=y` and `CONFIG_ZB_ZCZR=y` are enabled. If they are disabled, please enable them via `menuconfig`.

xieqinan, 2025-08-05:
Please try using `esp_zb_stack_main_loop();` instead of `esp_zb_stack_main_loop_iteration()`, which only runs the loop once.

Additionally, you can refer to the [HA switch example](https://github.com/espressif/esp-zigbee-sdk/tree/main/examples/esp_zigbee_HA_sample/HA_on_off_switch) to structure your application. It can help you better understand the framework of the esp-zigbee-sdk project.

AumDhabalia, 2025-08-05:
Thank you for the help. The new issue is the connectivity range is around 40cm max. but once connected to network, the range where data transmission and reception from end device to coordinator is around 5 to 6m. I want to increase the connectivity range. Already tried scan_duration function. Do you have any solution other than changing hardware? I am using pcb trace esp32c6 module.

AumDhabalia, 2025-08-07:
Thank you for the help. Do you happen to know any simulator in which I can perform simulations for checking whether my program is performing as per requirement or not? Especially for Zigbee since, I donot have the resources for conducting practical experiment.

xieqinan, 2025-08-08:
>  Do you happen to know any simulator in which I can perform simulations for checking whether my program is performing as per requirement or not? 

I don't think there is. However, a [sniffer](https://docs.espressif.com/projects/esp-zigbee-sdk/en/latest/esp32h2/developing.html#sniffer-and-wireshark) can help analyze frame interactions between Zigbee devices, which can be useful for debugging some problems.

AumDhabalia, 2025-08-13:
I want a recommendation on using xiao esp32c6 and esp32c6 devkitc. In Xiao esp32c6 there is only one USB type c port. I want to know whether it works with Serial monitor apps like PuTTy, TeraTerm, etc. fine or external uart to usb is required?

AumDhabalia, 2025-08-13:
Last question.
Xiao ESP32C6 has only one type c port. So will it directly connect to Putty, TeraTerm or usb to uart bridge is required?

xieqinan, 2025-08-14:
> Xiao ESP32C6 has only one type c port. So will it directly connect to Putty, TeraTerm or usb to uart bridge is required?

The Xiao ESP32-C6 is a secondary development by Seeed Studio based on Espressif’s ESP32-C6 chip, so I do not have related development experience. I think you can get a more accurate answer from Seeed Studio.

closed 2025-08-14 by AumDhabalia