espressif/esp-zigbee-sdk #694: esp_zb_aps_src_binding_table_size_set not setting binding table size.

mundevx, 2025-08-08:
### Question

## IDF Component Manager Manifest File
dependencies:
  espressif/esp-zboss-lib: "~1.6.3"
  espressif/esp-zigbee-lib: "~1.6.3"
  espressif/led_strip: "~2.0.0"
  ## Required IDF version
  idf:
    version: ">=5.0.0"

And zigbee stack code : 
    
    esp_zb_cfg_t zb_nwk_cfg = ESP_ZB_ZR_CONFIG();
    const uint16_t table_size = 32;
    esp_err_t  err = esp_zb_aps_src_binding_table_size_set(table_size);
    if (err != ESP_OK) {
        printf("Failed to set APS src binding table size\n");
    }
    esp_zb_init(&zb_nwk_cfg);

esp_zb_aps_src_binding_table_size_set  is not working, showing limited to default 16 only.
+-------+----------+----------+-----------------+-------------------------------------+
| Index | Group ID | Scene ID | Transition Time |   [Cluster ID | Extension Field]   |
+-------+----------+----------+-----------------+-------------------------------------+
|   00  |  0x0037  |   0x37   |      0000       | [ 0x0006 | 0x1 ] |
+-------+----------+----------+-----------------+-------------------------------------+
|   03  |  0x002f  |   0x2f   |      0000       | [ 0x0006 | 0x0 ] |
+-------+----------+----------+-----------------+-------------------------------------+
|   07  |  0x0032  |   0x32   |      0000       | [ 0x0006 | 0x1 ] |
+-------+----------+----------+-----------------+-------------------------------------+
|   10  |  0x0033  |   0x33   |      0000       | [ 0x0006 | 0x1 ] |
+-------+----------+----------+-----------------+-------------------------------------+
|   12  |  0x0035  |   0x35   |      0000       | [ 0x0006 | 0x0 ] |
+-------+----------+----------+-----------------+-------------------------------------+
|   15  |  0x003b  |   0x3b   |      0000       | [ 0x0006 | 0x1 ] |
+-------+----------+----------+-----------------+-------------------------------------+
+-------+----------+----------+-----------------+-------------------------------------+
| Index | Group ID | Scene ID | Transition Time |   [Cluster ID | Extension Field]   |
+-------+----------+----------+-----------------+-------------------------------------+
|   04  |  0x002f  |   0x2f   |      0000       | [ 0x0006 | 0x0 ] |
+-------+----------+----------+-----------------+-------------------------------------+
|   08  |  0x0032  |   0x32   |      0000       | [ 0x0006 | 0x1 ] |
+-------+----------+----------+-----------------+-------------------------------------+
|   11  |  0x0033  |   0x33   |      0000       | [ 0x0006 | 0x1 ] |
+-------+----------+----------+-----------------+-------------------------------------+
|   13  |  0x0035  |   0x35   |      0000       | [ 0x0006 | 0x0 ] |
+-------+----------+----------+-----------------+-------------------------------------+
+-------+----------+----------+-----------------+-------------------------------------+
| Index | Group ID | Scene ID | Transition Time |   [Cluster ID | Extension Field]   |
+-------+----------+----------+-----------------+-------------------------------------+
|   01  |  0x0031  |   0x31   |      0000       | [ 0x0006 | 0x1 ] |
+-------+----------+----------+-----------------+-------------------------------------+
|   05  |  0x002f  |   0x2f   |      0000       | [ 0x0006 | 0x0 ] |
+-------+----------+----------+-----------------+-------------------------------------+
|   09  |  0x0032  |   0x32   |      0000       | [ 0x0006 | 0x1 ] |
+-------+----------+----------+-----------------+-------------------------------------+
|   14  |  0x0035  |   0x35   |      0000       | [ 0x0006 | 0x0 ] |
+-------+----------+----------+-----------------+-------------------------------------+
+-------+----------+----------+-----------------+-------------------------------------+
| Index | Group ID | Scene ID | Transition Time |   [Cluster ID | Extension Field]   |
+-------+----------+----------+-----------------+-------------------------------------+
|   02  |  0x0035  |   0x35   |      0000       | [ 0x0006 | 0x0 ] |
+-------+----------+----------+-----------------+-------------------------------------+
|   06  |  0x002f  |   0x2f   |      0000       | [ 0x0006 | 0x0 ] |
+-------+----------+----------+-----------------+-------------------------------------+

on Adding new scene, i am getting error on scene store callback. 
E (74444) ESP_ZB_SCENE_COMMANDS: zb_get_scene_store_resp_handler(646): Received message: error status(1)

And app shows exceed limit.

why esp_zb_aps_src_binding_table_size_set is not working?

mundevx, 2025-08-08:
Our Espressif Zigbee-based products are currently in the production and deployment phase. We require an urgent resolution to the ongoing issue to ensure uninterrupted operation and timely delivery. Your prompt support and guidance on this matter would be highly appreciated.

mundevx, 2025-08-08:
Thank you for clarifying the behavior of esp_zb_aps_src_binding_table_size_set() and confirming that the ZCL scene table size is hardcoded in v1.6.3.

Given our current production and deployment timelines, we cannot wait until the end-of-August release. Therefore, I will reach out to the Espressif technical support team immediately to request the temporary library version you mentioned.

Please let me know if there are any specific details, logs, or project files you would require to expedite this process.

xieqinan, 2025-08-08:
Thanks for your understanding. If your contact information has been provided to our technical support team, I will send you the available libraries to fix this issue ASAP.

mundevx, 2025-08-12:
Hi @xieqinan , any update please.

xieqinan, 2025-08-12:
> any update please.

I have not received any feedback from our business or technical support. Have you contacted them?

xieqinan, 2025-08-12:
> I have contacted them in writing and am currently awaiting their response to my email.

OK, I will also contact them to reduce your wait time. Thank you for your patience.

mundevx, 2025-08-13:
> > I have contacted them in writing and am currently awaiting their response to my email.
> 
> OK, I will also contact them to reduce your wait time. Thank you for your patience.

@xieqinan , Thank you very much from heart.

mundevx, 2025-08-13:
@xieqinan 

Thanks for the library. It's compiled successfully, but some functions are not working, I am not able to check if the function (esp_zb_zcl_scenes_table_init) is setting my desired value or not as I am getting a crash on calling esp_zb_zcl_scenes_add_scene_cmd_req.  In the previous library, esp_zb_zcl_scenes_add_scene_cmd_req was working ok. 

This is the crash:
Zigbee stack assertion failed common/zb_nvram.c:2406

abort() was called at PC 0x4201cd67 on core 0
--- 0x4201cd67: zb_assert at ??:?
Core  0 register dump:
MEPC    : 0x4080063e  RA      : 0x408066bc  SP      : 0x4081c580  GP      : 0x4080d0c0
--- 0x4080063e: panic_abort at C:/esp/esp-idf/components/esp_system/panic.c:472
--- 0x408066bc: __ubsan_include at C:/esp/esp-idf/components/esp_system/ubsan.c:313
TP      : 0x40809e8c  T0      : 0x37363534  T1      : 0x7271706f  T2      : 0x33323130
--- 0x40809e8c: mmu_ll_get_page_size at C:/esp/esp-idf/components/hal/esp32h2/include/hal/mmu_ll.h:71
--- (inlined by) mmu_ll_check_valid_paddr_region at C:/esp/esp-idf/components/hal/esp32h2/include/hal/mmu_ll.h:128
--- (inlined by) mmu_hal_map_region at C:/esp/esp-idf/components/hal/mmu_hal.c:79
S0/FP   : 0x00000004  S1      : 0x4081c5e4  A0      : 0x4081c5ac  A1      : 0x4081c5e2
A2      : 0x00000000  A3      : 0x4081c5d9  A4      : 0x00000001  A5      : 0x40813000
A6      : 0x00000000  A7      : 0x76757473  S2      : 0x00000009  S3      : 0x0000001f
S4      : 0x00000000  S5      : 0xffffffe5  S6      : 0x4081c6f7  S7      : 0x40823ef4
S8      : 0x40813000  S9      : 0x00000002  S10     : 0x4081e105  S11     : 0x00000000
T3      : 0x6e6d6c6b  T4      : 0x6a696867  T5      : 0x66656463  T6      : 0x62613938
MSTATUS : 0x00001881  MTVEC   : 0x40800001  MCAUSE  : 0x00000007  MTVAL   : 0x00000000  
--- 0x40800001: _vector_table at ??:?
MHARTID : 0x00000000

Stack memory:
4081c580: 0x4081e105 0x00000002 0x4081c5e0 0x4080b7b0 0x4081c6f7 0xffffffe5 0x00000000 0x4080da6c
--- 0x4080b7b0: abort at C:/esp/esp-idf/components/newlib/abort.c:35
4081c5a0: 0x4081c5e4 0x4080da88 0x4081c5e0 0x726f6261 0x20292874 0x20736177 0x6c6c6163 0x61206465
4081c5c0: 0x43502074 0x34783020 0x63313032 0x20373664 0x63206e6f 0x2065726f 0x00000030 0x42010000
--- 0x42010000: zb_zcl_report_attr_callback at ??:?
4081c5e0: 0x1af10030 0x31303234 0x37366463 0x00000000 0x0000002f 0x0000fff0 0x42097cc1 0x4201cd6a
--- 0x4201cd6a: zb_aps_multi_pre_link_key_init at zb_init_common.c.obj:?
4081c600: 0x52bb9554 0x17c7c044 0xf76dccfe 0x00000966 0x00000009 0x00000020 0x40810e88 0x42020322
--- 0x42020322: zb_nvram_write_dataset at ??:?
4081c620: 0x00000000 0x00000000 0x00000000 0x40000000 0x0000437d 0x00090020 0x0000ffff 0x00000000
--- 0x40000000: _start in ROM
4081c640: 0x4081ca30 0x40823ef4 0x4081c6f8 0x42011cf6 0x25d5c561 0x4081d4c4 0x4081c730 0x00000001
--- 0x42011cf6: zcl_scenes_add_scene_handler at esp_zigbee_zcl_scenes.c.obj:?
4081c660: 0x0000004a 0x00000000 0x00000000 0xfffffffe 0x000186a0 0x00000000 0x4081c6f8 0x00000001
4081c680: 0x42011b8a 0x4081e154 0x0000001f 0x4200fd32 0xa83bb67f 0x4081c718 0x00000000 0x4206ede0
--- 0x42011b8a: zcl_scenes_add_scene_handler at esp_zigbee_zcl_scenes.c.obj:?
--- 0x4200fd32: zb_zcl_device_callback at ??:?
--- 0x4206ede0: signal_event_fd_from_isr at C:/esp/esp-idf/components/vfs/vfs_eventfd.c:234
4081c6a0: 0x6dd74278 0xa121a316 0xe9f0f395 0x00000000 0x4202fab0 0x4202fb76 0x4081c730 0x00000001
--- 0x4202fab0: zb_nwk_neighbor_get at ??:?
--- 0x4202fb76: zb_nwk_neighbor_get_by_short at ??:?
4081c6c0: 0x00000000 0x4081c754 0x0000001f 0x42042bea 0x4081c77c 0x4080ee88 0x0000000d 0x00000000
--- 0x42042bea: zb_zcl_scenes_process_add_scene_command at zcl_scenes.c.obj:?
4081c6e0: 0x4080ee88 0x00000004 0x00005e18 0x4081c748 0x00000000 0x0081c748 0x001d001d 0x42060000
--- 0x42060000: zb_nwk_broadcasting_clear_brrt at ??:?
4081c700: 0x00000000 0x40810e88 0x025a7028 0x7fffffff 0x00000002 0x00000000 0x0000001f 0x4204300a
--- 0x4204300a: zb_zcl_process_scenes_specific_commands at ??:?
4081c720: 0x408219e0 0x4081c7a8 0x00000000 0x4206ede0 0x4080ee88 0x00000000 0x00000000 0x00000000
--- 0x4206ede0: signal_event_fd_from_isr at C:/esp/esp-idf/components/vfs/vfs_eventfd.c:234
4081c740: 0x4081c7c0 0x4081c7c8 0x00000000 0x4206ede0 0x00000000 0x005e1800 0x00000000 0x00180000
--- 0x4206ede0: signal_event_fd_from_isr at C:/esp/esp-idf/components/vfs/vfs_eventfd.c:234
4081c760: 0x05020001 0x00010400 0x00001200 0x7f000000 0x0000001f 0x4081e170 0x0000001f 0x7fffffff
4081c780: 0x0000001f 0x4081e170 0x00000000 0x4203c896 0x00000001 0x0000001f 0x40823b5c 0x42037d40
--- 0x4203c896: zb_zcl_handle at ??:?
--- 0x42037d40: ep_process_zcl_cmd at zcl_common.c.obj:?
4081c7a0: 0x4081e020 0x00000000 0x025a7028 0x4081e170 0x0000001f 0x005e1800 0x00000000 0x00180000
4081c7c0: 0x05020001 0x00010400 0x00001200 0x42000000 0x0000001a 0x00000000 0x0000001f 0x420385d4
--- 0x420385d4: zb_zcl_process_device_command at ??:?
4081c7e0: 0x000186a0 0x005e1800 0x00000000 0x00180000 0x05020001 0x00010400 0x00001200 0x42000000
4081c800: 0x00000000 0x42038566 0x4080ee88 0x4202193c 0x00000000 0x00000000 0x408229d0 0x0000001f
--- 0x42038566: zb_zcl_process_device_command at ??:?
--- 0x4202193c: zb_sched_loop_iteration at ??:?
4081c820: 0x4206ec7e 0x4206e9f8 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
--- 0x4206ec7e: event_read at C:/esp/esp-idf/components/vfs/vfs_eventfd.c:284
--- 0x4206e9f8: event_start_select at C:/esp/esp-idf/components/vfs/vfs_eventfd.c:99
[... cut ...]
4081c8e0: 0x4080e2e4 0x4080e2e4 0x4081c8d8 0x4080e2dc 0x00000014 0x40823afc 0x40823afc 0x4081c8d8
4081c900: 0x00000000 0x00000005 0x4081a8d4 0x6267695a 0x6d5f6565 0x006e6961 0x00a66ade 0x00000000
4081c920: 0x4081c8d0 0x00000005 0x00000001 0x00000000 0x00000000 0x00000000 0x40813c94 0x40813cfc
4081c940: 0x40813d64 0x00000000 0x00000000 0x00000001 0x00000000 0x00000000 0x00000000 0x42076240
--- 0x42076240: _cleanup_r at ??:?
4081c960: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000

mundevx, 2025-08-13:
@xieqinan , Ok