espressif/esp-zigbee-sdk #696: Will this work with a generic zigbee network?

LorenzoZappa, 2025-08-09:
### Question

I am planning to buy a ESP32-C5-DevKitC and I want to use it as a smart device and connect it to my zigbee network. I have seen that there is a zigbee example client available on the repo, I was thinking of using this.
My question is if the ESP32-C5-DevKitC will work in an existent zigbee network (not another ESP32-C5-DevKitC as a server).
Sorry but I am very new to zigbee specifications.

srett, 2025-08-10:
Yes

xieqinan, 2025-08-11:
@LorenzoZappa ,

> My question is if the ESP32-C5-DevKitC will work in an existent zigbee network (not another ESP32-C5-DevKitC as a server).
Sorry but I am very new to zigbee specifications.

Yes, please give it a try. If you encounter any Zigbee-related issues, you can submit them here for discussion.

xieqinan, 2025-08-15:
> will the ESP32-C5-DevKitC also work as a repeater?

Yes, you can set the ESP32-C5-DevKitC as router to extend the network, please refer to: [HA dimmable light](https://github.com/espressif/esp-zigbee-sdk/blob/main/examples/esp_zigbee_HA_sample/HA_color_dimmable_light/main/esp_zb_light.h#L31) example.