espressif/esp-zigbee-sdk #697: ESP32-H2 and ZigBee Sensor registration

Usama050, 2025-08-11:
Hello! Everyone, I hope you all guys are doing well. I have question related the sensor registration, like I have ZigBee Door Sensor(IAS Cluster), Motion Sensor(IAS Cluster), Temperature and Humidity sensor, Smart Button, smart relay switch. Now I want to register these sensor in my code so only these sensor can be connected with my ESP32-H2 and whenever any sensor from the list is connected it show the log if other than these will connect it show the error message, the main question is that how can I register my sensor, is there any example according to my question. I will ne grateful for any kind of help. Thank you and Have a good day!

xieqinan, 2025-08-11:
The [zdo\_leave](https://github.com/espressif/esp-zigbee-sdk/blob/main/components/esp-zigbee-lib/include/zdo/esp_zigbee_zdo_command.h#L652C6-L652C33) command may be helpful. The steps are as follows:

1. List the IEEE addresses of your sensors in the Zigbee Coordinator, named `allowed_sensors_address_list`.
2. When a new device joins the network, an `ESP_ZB_ZDO_SIGNAL_DEVICE_ANNCE` signal will be triggered with the joiner’s IEEE address. Compare this address with the `allowed_sensors_address_list`. If it is not in the list, call [zdo\_leave](https://github.com/espressif/esp-zigbee-sdk/blob/main/components/esp-zigbee-lib/include/zdo/esp_zigbee_zdo_command.h#L652C6-L652C33) to remove it; otherwise, accept it.

Usama050, 2025-08-12:
Hello! @xieqinan I understand your point but the problem is that I have a ZigBee door sensor and my boss he also has a ZigBee door sensor and I am sure the IEEE address for both sensor is different, is there another way to registered them?