espressif/esp-zigbee-sdk #704: deep sleep example enters deep sleep at the wrong time

oddlama, 2025-08-20:
### Answers checklist.

- [x] I have read the documentation [ESP Zigbee SDK Programming Guide](https://docs.espressif.com/projects/esp-zigbee-sdk) and tried the [debugging tips](https://docs.espressif.com/projects/esp-zigbee-sdk/en/latest/esp32/developing.html#debugging), the issue is not addressed there.
- [x] I have updated ESP Zigbee libs ([esp-zboss-lib](https://components.espressif.com/components/espressif/esp-zboss-lib) and [esp-zigbee-lib](https://components.espressif.com/components/espressif/esp-zigbee-lib)) to the latest version, with corresponding IDF version, and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.3.2

### esp-zigbee-lib version.

1.6.5

### esp-zboss-lib version.

1.6.4

### Espressif SoC revision.

ESP32-C6

### What is the expected behavior?

This is a follow up to https://github.com/espressif/esp-zigbee-sdk/issues/465#issuecomment-3199023670, I think we talked past one another @xieqinan 😅

Let me summarize what the actual problem is about:

Consider a deep sleep temperature sensor. When the sensor wakes up from deep sleep, it must first measure the temperature and then report it so the home automation software (like Home Assistant) can display the current temperature.

The current deep sleep example just starts a 5 second timer and goes into deep sleep unconditionally after the timer expires. That is not good since there is no guarantee that the home automation software had received the new temperature values. I can extend that to 60 seconds to make sure there is enough time for the home automation software to request the value, but that is purely speculative and increases power consumption by a large factor.

Ideally, I want to force the zigbee stack to report the value _now_ have it send me a message or signal once all values were actually successfully reported. Only then I can go into deep sleep again, otherwise I risk losing temperature data.

So my question is how can I ask the zigbee stack whether everything has been sent? How do I know that I can enter deep sleep safely without risking that the home automation software has not received the data?

### What is the actual behavior?

The device goes into deep sleep unconditionally after 5 seconds. There seems to be no documented way on how to do it better.

### Steps to reproduce.

N/A

xieqinan, 2025-08-28:
It seems to be a usage issue.

> So my question is how can I ask the zigbee stack whether everything has been sent? How do I know that I can enter deep sleep safely without risking that the home automation software has not received the data?

Based on your application, I think we can use the `default response` mechanism to address this issue. When the device starts up, do not call `zb_deep_sleep_start()` immediately. Instead, allow time for the temperature sensor to measure the temperature and report it to the HA.

If an `ESP_ZB_CORE_CMD_DEFAULT_RESP_CB_ID` signal with `resp_to_cmd = 0x0a` and `status_code == 0` (indicating that the ZCL report command has been received by the receiver) is registered in the `esp_zb_core_action_handler_register()`, you can then call `zb_deep_sleep_start()` with `before_deep_sleep_time_sec = 0`, so the device can enter sleep immediately.

Please give this solution a try.

xieqinan, 2026-01-20:
Hi @oddlama,

Do you have any updates? Please let me know your progress so that I can provide appropriate assistance.