espressif/esp-zigbee-sdk #709: Query open network state on wakeup from deep sleep

mcott, 2025-08-28:
### Question

When waking up from deep sleep, is there a way to query the current open/closed network state?

We have implemented the esp_zb_aps_data_indication_handler_register function; however, when the coordinator opens the network while the device is in deep sleep, the handler is not triggered.

### Additional context.

We indicate the open network state on all end devices, as extra check for the installer that these devices are part of the network.

xieqinan, 2025-08-28:
> When waking up from deep sleep, is there a way to query the current open/closed network state?

I don’t think so. Could you please share more details about the purpose? I believe there are other ways to verify whether a device is part of the network.

mcott, 2025-08-28:
The purpose is to give led feedback about the open/closed state of the network with all the devices that are in the network.
We want to give this led feedback with all devices because the coordinator is typically not visible from the end devices.

Because some devices are battery powered, they could be in deep sleep while the coordinator opened the network. That way the sleeping devices missed the "esp_zb_aps_data_indication_handler_register" callback that we use to know the open/closed state.
The idea is that a battery powered device wakes up from deep sleep, requests the network state and updates the led feedback accordingly.

mcott, 2025-08-29:
We use the open/close state of the network as a trigger point. We understand that the sleeping end devices are missing the open/close message from the coordinator, but the device should know that network state for our application.

mcott, 2025-09-01:
Our deep sleep device is a button control with some led feedback for the main device (coordinator) it is connected to.
For our installers it would be useful that they open the network on the coordinator and then preform a button press sequence that leads to led feedback. We can give that feedback when the device is not sleeping, but because it is battery powered we want to sleep as much as possible. To enable giving the correct led feedback, waking up from deep sleep, we need to be able to query the network state on the deepsleep device.

So the main purpose is to give feedback to the installer that the button control is part of the network and the installer can add more button control devices to the network.