espressif/esp-zigbee-sdk #710: Zigbee Gateway does not deliver Default Response from end device on custom cluster with array data type

syam-ari, 2025-08-29:
### Answers checklist.

- [x] I have read the documentation [ESP Zigbee SDK Programming Guide](https://docs.espressif.com/projects/esp-zigbee-sdk) and tried the [debugging tips](https://docs.espressif.com/projects/esp-zigbee-sdk/en/latest/esp32/developing.html#debugging), the issue is not addressed there.
- [x] I have updated ESP Zigbee libs ([esp-zboss-lib](https://components.espressif.com/components/espressif/esp-zboss-lib) and [esp-zigbee-lib](https://components.espressif.com/components/espressif/esp-zigbee-lib)) to the latest version, with corresponding IDF version, and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

ESP-IDF v5.3.2

### esp-zigbee-lib version.

1.6.5

### esp-zboss-lib version.

1.6.4

### Espressif SoC revision.

ESP32S3 + H2 

### What is the expected behavior?

ESP Zigbee Gateway sends a custom ZCL command (cluster ID: <custom_cluster_id>, payload: array of 16 bytes).

Remote (nRF52840) responds with a Default Response (cmd 0x0B), status = ZB_ZCL_STATUS_UNSUP_CLUSTER_COMMAND.

Verified in sniffer trace: Default Response is transmitted and acknowledged at MAC layer.

On ESP side:

 callback should be triggered

Application should receives the Default Response.

### What is the actual behavior?

ESP Zigbee Gateway sends a custom ZCL command (cluster ID: <custom_cluster_id>, payload: array of 16 bytes).

Remote (nRF52840) responds with a Default Response (cmd 0x0B), status = success.

Verified in sniffer trace: Default Response is transmitted and acknowledged at MAC layer.

On ESP side:

No callback triggered.

Application never receives the Default Response.

### Steps to reproduce.

1.Define a custom cluster(0xFC00) on ESP side with an array data type attribute.

2. Send a custom command with array payload (16-byte array).

3. On remote (nRF), send Default Response with status success

4. Observe with sniffer → Default Response is visible.
5. On ESP Gateway → callback not triggered.

xieqinan, 2025-08-29:
Hi,

If a ZCL `default response` is received by the esp-zigbee-sdk stack, it will trigger a callback registered with `esp_zb_core_action_handler_register()` using the callback ID `ESP_ZB_CORE_CMD_DEFAULT_RESP_CB_ID`. Could you please check whether this callback has been registered in your application?

xieqinan, 2025-09-02:
OK, understood. The cluster ID `0xFC00` is reserved as a manufacturer-specific cluster in ZBOSS. Please refer to [this reference](https://github.com/espressif/esp-zboss-lib/blob/dcffc6ae499e6e6335f5856d7ff73c6f51b41039/include/zcl/zb_zcl_common.h#L328).

Could you please use a different custom cluster ID in your application?

syam-ari, 2025-09-02:
Sure, Thanks for your support.

closed 2025-09-02 by syam-ari