espressif/esp-zigbee-sdk #714: Assertion failed in zb_address.c when adding more than 30 Router devices

Dev-Hussam, 2025-09-06:
### Answers checklist.

- [x] I have read the documentation [ESP Zigbee SDK Programming Guide](https://docs.espressif.com/projects/esp-zigbee-sdk) and tried the [debugging tips](https://docs.espressif.com/projects/esp-zigbee-sdk/en/latest/esp32/developing.html#debugging), the issue is not addressed there.
- [x] I have updated ESP Zigbee libs ([esp-zboss-lib](https://components.espressif.com/components/espressif/esp-zboss-lib) and [esp-zigbee-lib](https://components.espressif.com/components/espressif/esp-zigbee-lib)) to the latest version, with corresponding IDF version, and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

5.5.0

### esp-zigbee-lib version.

1.6.7

### esp-zboss-lib version.

1.6.4

### Espressif SoC revision.

ESP32S3(HOST)- ESP32H2(RCP)

### What is the expected behavior?

The gateway should successfully commission all 50 devices without crashing or entering a reboot loop. It should handle the allocation of resources for each new device gracefully.

### What is the actual behavior?

when subsequent devices (e.g., the 31st) attempt to join, the Host MCU crashes and enters a continuous reboot loop with the following error:

`DEVICE_DISCOVERY: Live Memory Usage: Free: 91284 bytes, Min Ever Free: 86212 bytes
zigbee stack assertion failed common/zb_address.c:395

abort() was called at PC 0x4203eac8 on core 0
0x4203eac8: zb_assert at ???:?

Backtrace:
0x40375da1:0x3fcbcf10 0x4037ce21:0x3fcbcf30 0x40384c6a:0x3fcbcf50 0x4203eac8:0x3fcbcf80 0x4203eacd:0x3fcbcfa0 0x4203d551:0x3fcbcfd0 0x4203d5da:0x3fcbd00
0x4204de62:0x3fcbd170 0x420443a4:0x3fcbdcd0 0x42065c5b:0x3fcbce00 0x42031718:0x3fcbce20 0x42031723:0x3fcbce40 0x42012981:0x3fcbce60
--- 0x40375da1: panic_abort at C:/Users/aswar/esp/v5.5/esp-idf/components/esp_system/panic.c:460
--- 0x4037ce21: esp_system_abort at C:/Users/aswar/esp/v5.5/esp-idf/components/esp_system/port/esp_system_chip.c:87
--- 0x40384c6a: abort at C:/Users/aswar/esp/v5.5/esp-idf/components/newlib/src/abort.c:38
--- 0x4203eac8: zb_assert at ???:?
--- 0x4203d4c4: zb_ieee_addr_compress at ???:?
--- 0x4203d551: zb_address_by_ieee at ???:?
--- 0x4203d5da: zb_address_check_mem_for_new_addr at ???:?
--- 0x4204de62: zb_mcps_data_indication_cont at ???:?
--- 0x420443a4: zb_sched_loop_iteration at ???:?
--- 0x42065c5b: zboss_main_loop_iteration at ???:?
--- 0x42031718: esp_zb_stack_main_loop_iteration at ???:?
--- 0x42031723: esp_zb_stack_main_loop at ???:?
--- 0x42012981: esp_zb_task at D:/ESP-IDF_ZB_Examples/ZB_Gateways/HOST_RCP_Gateway_2/components/host_app/src/app.c:415

ELF file SHA256: 0c121f7d7

Rebooting...
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x2b (SPI_FAST_FLASH_BOOT)
Saved PC:0x40375d04
--- 0x40375d04: esp_restart_noos at C:/Users/aswar/esp/v5.5/esp-idf/components/esp_system/port/soc/esp32s3/system_internal.c:162`

### Steps to reproduce.

1. Build and flash the host_rcp_gateway example onto an ESP32-S3 Host and ESP32-H2 RCP.

2. Set up a network with a large number (>30) of Zigbee router devices.

3. Permit joining on the gateway.

4. Power on all router devices to initiate commissioning simultaneously.

5. Observe the Host MCU crash after approximately 30 devices have successfully joined.

### More Information.

I have developed a Zigbee Gateway using an ESP32-S3 (Host) and ESP32-H2 (RCP). I am attempting to commission approximately 50 Zigbee router devices simultaneously.

The commissioning process involves each device attempting to join using a unique install code. When the gateway receives a device association request, it fetches the corresponding install code from a cloud service and adds it to the Zigbee stack's key table.

The system works correctly for the first 30 devices. However, when subsequent devices (e.g., the 31st) attempt to join, the Host MCU crashes and enters a continuous reboot loop with the following error.

Dev-Hussam, 2025-09-08:
duplicated issue

closed 2025-09-08 by Dev-Hussam

xieqinan, 2025-09-08:
@Dev-Hussam ,

Could you please share the all IEEE address of the joined routers? The issue seems to be caused by an insufficient manufacturer table.