espressif/esp-zigbee-sdk #716: The commands are not sent in a timely manner, but are sent in a piled-up manner

ecsply, 2025-09-07:
### Question

I am trying to send commands to the terminal, however, whether using on/off cluster or custom cluster to send commands, the commands are not sent when calling the function, but wait for more than ten function calls before sending them to the terminal (discovered through packet capture)
```c
void light_timer_callback(void *arg) {
  if (state == 0x00) {
    send_command_onoff(sAddr, 0x42, 0x01);
    state = 0x01;
  } else {
    send_command_onoff(sAddr, 0x42, 0x00);
    state = 0x00;
  }
  ESP_ERROR_CHECK(esp_timer_start_once(light_timer, 5000000));
}

void send_command_onoff(uint16_t sAddr, uint8_t devType, uint8_t onoff) {
    esp_zb_zcl_on_off_cmd_t cmd_req;
    uint8_t dst_endpoint = devType;

    cmd_req.on_off_cmd_id = onoff ? ESP_ZB_ZCL_CMD_ON_OFF_ON_ID : ESP_ZB_ZCL_CMD_ON_OFF_OFF_ID;
    cmd_req.address_mode = ESP_ZB_APS_ADDR_MODE_16_ENDP_PRESENT;

    cmd_req.zcl_basic_cmd.dst_addr_u.addr_short = sAddr;
    cmd_req.zcl_basic_cmd.dst_endpoint = dst_endpoint;
    cmd_req.zcl_basic_cmd.src_endpoint = 0xC0;
    esp_zb_lock_acquire(portMAX_DELAY);
    esp_err_t ret = esp_zb_zcl_on_off_cmd_req(&cmd_req);
    esp_zb_lock_release();
    Serial.println(onoff ? "on" : "off");
}
```
Terminal:
[image]

Am I missing some code? Or is there a mandatory sending function?
Hope someone can answer, thank you very much!

### Additional context.

[image]
The packet capture process of Wireshark is shown in the above figure, and the sequence numbers of the commands sent are also messy.

ecsply, 2025-09-08:
Thank you so much for inspiring me with your code.
At first, I followed the example code you gave me and used a timer to send commands, but it still seemed like they were stacked and sent together. Subsequently, I modified the code as follows and removed the timed behavior:
```c
case ESP_ZB_CORE_CMD_DEFAULT_RESP_CB_ID: {
      const esp_zb_zcl_cmd_default_resp_message_t *resp =
          (const esp_zb_zcl_cmd_default_resp_message_t *)message;
      Serial.printf("Default Response: cluster=0x%04X cmd=0x%02X status=%d\n",
                    resp->info.cluster, resp->info.command, resp->status_code);
      vTaskDelay(pdMS_TO_TICKS(100));
      if (state == 0x00) {
        send_command_onoff(sAddr, 0x42, 0x01);
        state = 0x01;
      } else {
        send_command_onoff(sAddr, 0x42, 0x00);
        state = 0x00;
      }
      break;
    }
```
It can work now! But it's not the way I want.
I found through comparing the differences and packet capture analysis that sending commands through a timer always results in `ESP_ZB_CLME_STATUS-INDICATION` and `ESP_ZB_ZDO-VNet UNAVAILABLE`
**There is a problem with the terminal!**
Even if the terminal is connected to the coordinator, it seems invisible to the coordinator. When I have the terminal report to the coordinator at regular intervals, the terminal can receive commands normally.

I have another question: If I report to the coordinator irregularly at the terminal, the commands I send will still pile up and be sent (because the Zigbee protocol stack cannot find the terminal, and then it can be found again after a while? ）. How does the Zigbee protocol stack handle sending commands? How can I improve?

Finally, thank you again for your help. I have been troubled by this issue for a long time. Thank you!

xieqinan, 2025-09-09:
>  I followed the example code you gave me and used a timer to send commands, but it still seemed like they were stacked and sent together. 

No, the method of using a timer to send commands was given based on your [description](https://github.com/espressif/esp-zigbee-sdk/issues/716#issue-3391794983). If I misunderstood your meaning, please correct me.

If the issue is still confused for you, please also share the patch of your application so I can reproduce it.

> I have another question: If I report to the coordinator irregularly at the terminal, the commands I send will still pile up and be sent (because the Zigbee protocol stack cannot find the terminal, [...]

I believe you need to provide more details about your application. I have never encountered this mentioned case. It is important to share more background on your application so we can stay on the same page.

ecsply, 2025-09-09:
I think I have figured out the process of sending commands. Indeed, the problem comes from the end device, not the coordinator (I am using a coordinator, not a switch). 

**Device version:
Coordinator: ESP32-C6
EndDevice: CC2530 Z-stack 3.0**

There are two reasons for this phenomenon:

> 1. The polling interval for EndDevice data requests is set too large.
> 2. The find and bind modes were not used.

AI told me that after calling the command function, the coordinator does not directly send it, but waits for the EndDevice's `Data Request` request before sending the command. Due to the configuration file in the EndDevice `DPOLL_RATE（The number of milliseconds to wait between data request polls to the coordinator.）` The default time is 60 seconds, which causes the EndDevice to only ask the coordinator after 1 minute: "Hello, I'm awake. Do you have my package (waiting for the data to be sent to me)?" That's why many commands were sent together.

In addition, since the find and bind modes are not used, the EndDevice still needs to rely on sending a Data Request to ultimately obtain the command.

Attached below is my code. For ease of debugging, I ran it in Arduino. If you want to use it, you need to modify the file suffix to `.ino`(Sorry, files with the ino suffix cannot be uploaded). Uh, I tried to name the extension .c, but the upload failed and only txt files can be uploaded.

[zigbee.txt](https://github.com/user-attachments/files/22227058/zigbee.txt)

Thank you again! Wishing you a happy life!

closed 2025-09-09 by ecsply