espressif/esp-zigbee-sdk #725: Binding with destination address 16bit group type

lsroka76, 2025-09-25:
### Question

Current SDK states as follows (ZDO Command API):

`typedef struct [esp_zb_zdo_bind_req_param_s](https://docs.espressif.com/projects/esp-zigbee-sdk/en/latest/esp32/api-reference/zdo/esp_zigbee_zdo_command.html#_CPPv427esp_zb_zdo_bind_req_param_s) esp_zb_zdo_bind_req_param_t[](https://docs.espressif.com/projects/esp-zigbee-sdk/en/latest/esp32/api-reference/zdo/esp_zigbee_zdo_command.html#_CPPv427esp_zb_zdo_bind_req_param_t)
The Zigbee ZDO bind command struct.

Note

Current implementation of the API ONLY supports 64 bit extended address’s address mode. Other address mode will support later.

Note

Be aware of the one req_dst_addr is address that command send to, while dst_address is the destination of the binding entry.

Note

NOW the dst_addr_mode is default by ZB_BIND_DST_ADDR_MODE_64_BIT_EXTENDED. Later SDK will support other address mode.`

Does that comment still holds? In other words now there is no option for so called groupcast - binding with destination address type 16bit group id i and group number in address structure?

xieqinan, 2025-09-26:
> Does that comment still holds? In other words now there is no option for so called groupcast - binding with destination address type 16bit group id i and group number in address structure?

Yes, the group address type is supported in current SDK (v1.6.7), could you please give a try?

xieqinan, 2025-09-28:
Hi,

I do not fully understand the requirements of your application based on the information provided above. Could you please share more background details about your needs? In the meantime, you can also refer to [ZCL Groups](https://docs.espressif.com/projects/esp-zigbee-sdk/en/latest/esp32/user-guide/zcl_groups.html), which may be helpful.

lsroka76, 2025-10-16:
Problem solved with add group request  - thanks for link and directions.