espressif/esp-zigbee-sdk #726: Unable to control individual endpoints on 2-gang switch (both switches turn on/off together)

iamiha, 2025-09-29:
### Question

I am using the Zigbee Gateway example from the **ESP Zigbee SDK**.
I added an **On/Off command** for a device (2-gang switch).

**Problem:**

* When I send an On/Off command to control one switch, it **turns on/off both switches together**.
* Physically, the switch works correctly.
* With **Zigbee2MQTT**, it also controls the endpoints individually as expected.

**Expected behavior:**

* I should be able to send an On/Off command to **control one endpoint (one gang) at a time**.

**Steps to reproduce:**

1. Add a 2-gang switch device to the Zigbee Gateway example.
2. Send an On/Off command targeting a single endpoint.
3. Observe that both endpoints are toggled instead of just the selected one.

**What I’ve tried:**

* Verified that physical device operation is correct.
* Verified endpoint control works properly in Zigbee2MQTT.
* In the SDK, the command seems to affect both endpoints.

**Code snippet for control:**

```c
uint8_t gateway_ep = ESP_ZB_GATEWAY_ENDPOINT;

esp_err_t esp_gateway_control_light_on_off(bool on_off, uint16_t sh_addr)
{
    esp_zb_zcl_on_off_cmd_t cmd_req;
    uint8_t cmd_id = (on_off) ? (ESP_ZB_ZCL_CMD_ON_OFF_ON_ID) : (ESP_ZB_ZCL_CMD_ON_OFF_OFF_ID);
    ESP_LOGI(TAG, "gateway_control_light_on_off: device short_addr=0x%04X ", sh_addr);

    cmd_req.zcl_basic_cmd.dst_addr_u.addr_short = sh_addr;
    cmd_req.zcl_basic_cmd.dst_endpoint = 2;   // <-- hardcoded endpoint for testing
    cmd_req.zcl_basic_cmd.src_endpoint = gateway_ep;
    cmd_req.address_mode = ESP_ZB_APS_ADDR_MODE_16_ENDP_PRESENT;
    cmd_req.on_off_cmd_id = cmd_id;

    ESP_LOGI(TAG, "send 'on_off' command: %s", on_off ? "true" : "false");

    esp_zb_lock_acquire(portMAX_DELAY);
    esp_zb_zcl_on_off_cmd_req(&cmd_req);
    esp_zb_lock_release();
    return ESP_OK;
}

/* Task to control On/Off after 5 minutes */
static void delayed_on_off_task(void *pvParam)
{
    uint16_t short_addr = (uint32_t)pvParam;  // cast back from void* to short addr
    ESP_LOGI(TAG, "Waiting 5 minutes before controlling device 0x%04X", short_addr);

    vTaskDelay(pdMS_TO_TICKS(5 * 60 * 1000));  // 5 minutes

    esp_gateway_control_light_on_off(true, short_addr);   // Turn ON
    vTaskDelay(pdMS_TO_TICKS(2000));  // short delay (2s)
    esp_gateway_control_light_on_off(false, short_addr);  // Turn OFF

    ESP_LOGI(TAG, "Completed ON/OFF sequence for device 0x%04X", short_addr);
    vTaskDelete(NULL);  // delete self
}

case ESP_ZB_ZDO_SIGNAL_DEVICE_ANNCE:
    dev_annce_params = (esp_zb_zdo_signal_device_annce_params_t *)esp_zb_app_signal_get_params(p_sg_p);
    ESP_LOGI(TAG, "New device commissioned or rejoined (short: 0x%04hx)", dev_annce_params->device_short_addr);

    xTaskCreate(delayed_on_off_task, "delay_onoff", 4096,
            (void *)(uintptr_t)dev_annce_params->device_short_addr, 5, NULL);

    break;
```

**Request:**
Can you please help me with how to correctly address and control **only one endpoint** of the 2-gang switch using the ESP Zigbee SDK?

iamiha, 2025-09-30:
[zigbee_sniff.csv](https://github.com/user-attachments/files/22611753/zigbee_sniff.csv)

[zb_sniff.txt](https://github.com/user-attachments/files/22611773/zb_sniff.txt)

Attached are the zigbee sniffer logs in CSV and txt format.

xieqinan, 2025-09-30:
Hi,

Could you please provide the sinffer log in `.pcap` format? It is hard for me to parse the log in CSV and txt format.

iamiha, 2025-09-30:
I’m not very familiar with reading the sniffer data yet. The sniffer is running, but the captured packets are not decrypted. I did configure it according to the [doc](https://docs.espressif.com/projects/esp-zigbee-sdk/en/latest/esp32/developing.html#sniffer-and-wireshark) and set the ZigbeeAlliance09 key, but I still don’t know how to properly read the details.

Could you explain how you verified that the On/Off command was sent correctly? Also, could you share some guidance on how to analyze the sniffer log? Right now, I only see encrypted data and I’m not sure how to confirm whether the command is being transmitted as expected.